CTSRD-CHERI/cheribsd #15: CheriABI - kernel support

rwatson, 2015-07-26:
Supporting a userspace execution environment in which all pointers are capabilities requires a system-call interface that understands capabilities, and as necessary, maps them to/from pointers for use by the kernel. In some cases, we need to retain the full capability in the ABI in order to return pointers to userspace with tags (i.e., opaque pointer-aware types in kqueue). It would also be nice (very good indeed) to respect capability bounds when using a userspace-provided capability for access to user memory.

This issue is for kernel-side support for CheriABI.

brooksdavis, 2015-08-10:
This is basically done, except that the passed capability isn't used to derive the new capability when mmap is given an argument and we can't produce binaries to test the kernel support.

One interesting question is: should DDC constrain the result of mmap?  I think the answer is NO because  this is pure capability code and mmap is how you allocate address space in a modern Unix-like OS.  Thus, in a **very** pure capability world where DDC is basically empty you'd want mmap to be the way you produce new address space.

davidchisnall, 2015-08-10:
The exception here is rtld, which uses mmap to map regions that are inside of DDC and PCC.  We probably need to think quite carefully about the relationship between allocating virtual address space and allocating memory permissions.  For cheriabi, it's less critical, but it would be nice if sandboxes could use mmap and have some containment enforced by the OS (though given the cost of mmap, a ccall to the system class probably wouldn't be too painful).

rwatson, 2015-08-10:
My take is that CheriABI, as we're currently developing it, is a fully privileged interface. Much of the time userspace will provide it with capabilities when requesting memory access (e.g., read(2), write(2)) but some operations may return more privileged capabilities (e.g., mmap(2) when a pointer argument isn't provided). We will need a separate, Capsicum-like subset of CheriABI suitable for use in sandboxes. It could be that in the Capsicum-like subset, mmap(2) without a pointer (capability) argument is not permitted.

brooksdavis, 2015-09-14:
As of b221a1b, the basic kernel interface works (i.e. we can start processes with arguments and environment passed on and the write() system call works as expected.)  Significant testing is still required, particularly for mmap.

The kqueue interface also needs considerable further thought.  The current one may work in some limited circumstances, but probably does not work in general.

rwatson, 2017-03-19:
Closing as now implemented pretty thoroughly!

closed 2017-03-19 by rwatson