CTSRD-CHERI/cheribsd #16: CheriABI - userspace support - static binaries

rwatson, 2015-07-26:
This issue is with regards to being able to compile statically linked binaries against CheriABI -- a capability-aware system-call ABI -- such that entire userspace applications can be capability-based, rather than just sandboxed enclaves.

It will depend on #15, kernel support for CheriABI.

brooksdavis, 2015-09-14:
Basic userspace support is complete as of 7e7d309.  A through audit of libc function implementations and considerable testing is required, but a simple printf() based helloworld works.

closed 2015-09-14 by brooksdavis

davidchisnall, 2015-09-15:
Yay!  Do we have instructions for building cheriabi binaries?  Does the SDK contain everything that's needed?

brooksdavis, 2015-09-15:
As long as you've got /usr/libcheri in the SDK sysroot it should just work to add -mabi sandbox to the compile line and then run "brandelf -c (128|256) <binary>" on the resulting binary.  I will introduce a partial flag day in the not too distant future when I get a proper elf machine type, but should be able to keep the kernel working with old binaries without too much trouble.

brooksdavis, 2015-09-15:
One final thing that isn't automatic is that you need to link with -lc -lmalloc_simple to get a plausibly working malloc since libc currently doesn't have one.

davidchisnall, 2015-09-16:
I can't get this to work.  With the latest SDK, I get:

	$ ~/sdk/bin/clang -mabi=sandbox  hello_world.c -static -lc -lmalloc_simple
	/home/dc552/sdk/bin/cheri-unknown-freebsd-ld: /home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/libc.a(wsetup.o): linking 32-bit code with 64-bit code
	/home/dc552/sdk/bin/cheri-unknown-freebsd-ld: failed to merge target specific data of file /home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/libc.a(wsetup.o)
	clang-3.7: error: linker command failed with exit code 1 (use -v to see invocation)

The linker invocation is:

	"/home/dc552/sdk/bin/cheri-unknown-freebsd-ld" "--sysroot=/home/dc552/CHERISDK/sdk/sysroot" "-Bstatic" "-o" "a.out" "/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/crt1.o" "/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/crti.o" "/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/crtbeginT.o" "-L/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri" "/tmp/hello_world-4294bc.o" "-lc" "-lmalloc_simple" "-lgcc" "-lgcc_eh" "-lc" "-lgcc" "-lgcc_eh" "/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/crtend.o" "/home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/crtn.o"

All of the files (including all of the files in the .o files in libc.a and libmalloc.a) appear to be correct.  Extracting them, I get lines like this:

	$ file wsetup.o
	wsetup.o: ELF 64-bit MSB relocatable, MIPS, MIPS-IV version 1 (FreeBSD), not stripped

Interestingly, if I link libc, I get this:

	$ sdk/bin/cheri-unknown-freebsd-ld -r -o libc.o /home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/libc.a
	$ file libc.o
	libc.o: ELF 64-bit MSB relocatable, MIPS, MIPS-I version 1 (FreeBSD), not stripped

It looks as if the MIPS IV flag is being lost somewhere in partial linking, being replaced with MIPS I, which the linker knows is 32-bit (in spite of the 64-bit ELF header) and complains about.  This doesn't seem to be a universal thing though.  `libc.so.7` is 'MIPS-IV version 1', so `ld` is presumably capable of producing files that have MIPS IV headers (are we compiling all of libc with clang?  If not, then it's also correctly merging MIPS III and MIPS IV headers and promoting the result to requiring MIPS IV).

Having spent a couple of hours chasing various dead ends, I'd find it quite helpful to know *exactly* how you are successfully building CHERIABI binaries.

brooksdavis, 2015-09-16:
bin/helloworld contains a hello world program that builds and runs.  I've never seen anything like the errors you're seeing.  I'll try to get something working outside the tree later today.

davidchisnall, 2015-09-16:
In tree, it appears that we pass `--no-warn-mismatch` to the linker, which kills this warning.  I am now able to build something that looks like a working binary with:

	$ ~/sdk/bin/clang -msoft-float -mabi=sandbox -Wl,--no-warn-mismatch hello_world.c -lc -lmalloc_simple

davidchisnall, 2015-09-17:
I have now modified the clang driver so that it will automatically pass `--no-warn-mismatch` when linking sandbox ABI binaries and so that it will run `brandelf` on the result.  Currently it always runs `brandelf -c 256` - I'll modify it to run it with the correct argument soon.

brooksdavis, 2015-09-17:
That's very useful.  I'll update the olden and base system builds appropriately.

On a semi-related note, my current setup doesn't support the apparently obsolete .init and .fini sections via crti.o and crtn.o (both files are complied from empty .c files).  I have a patch to not require them at all in the compile, but haven't committed it as I'm unsure if we can really do away with that support.