CTSRD-CHERI/cheribsd #17: KTrace CCall/CReturn support

rwatson, 2015-07-26:
It would be very useful if the kernel's ktrace(1) facility could be used to track CHERI object-capability invocation/return. This would require dropping out of the CCall/CReturn fast paths to insert log entries describing the transition -- e.g., object code and data pointers, method number, and perhaps other argument information. This might be done via a software-defined exception -- but it will be a bit tricky to unwind "this is a slow path" from "this is an exception, do signal delivery".

rwatson, 2015-07-26:
It would also be nice to have a DTrace probe -- that's an incremental change once KTrace support is there (or vice versa).

brooksdavis, 2015-08-07:
Added CCall/CReturn tracing in af0821ececbc4102ee383b9cc9122c33eb91410a and capability exception tracing in 3a025e837cf8f397035cb19f00ad963849a08e6d.  Created #25 to track DTrace support.

commit 1e2b2d8a (2019-04-17): Do not declare a getentropy() wrapper for newer glibc versions

The getentropy() function first appeared in glibc 2.25. Fixes #17
closed 2015-08-07 by brooksdavis