CTSRD-CHERI/cheribsd #24: mechanism to restrict method calls in a system object

brooksdavis, 2015-07-30:
The system class provides many different methods. It was possible to restrict the methods that could be called in a given system object using sandbox_object_new_flags(), but the underlying checks used to enforce those flags were removed in 87611244e when cheri_system_enter() was removed.  A new mechanism needs to be designed and implemented.

brooksdavis, 2015-07-30:
One possible solution would be to provide an interface to filter the vtable of the object when it is created in sandbox_object_load().  This could be handled by tagging methods with flags at declaration checked against sbo_flags or by some general mechanism where a set of allowed or denied method names was passed in a data structure.

jrtc27, 2022-01-21:
libcheri was removed in #1056

closed 2022-01-21 by jrtc27