CTSRD-CHERI/cheribsd #29: Signal delivery is broken in cheriabi

davidchisnall, 2015-09-17:
sigaction takes a struct that contains a code capability.  Cheriabi doesn't appear to add a `cheri_` compat version to extract the code capability.  We probably also need some changes to the signal delivery code to install a code capability, as well as a code pointer, though I believe that we have some of this currently to allow signals to run outside of a sandbox when raised in the sandbox.

brooksdavis, 2015-09-17:
When I did the first pass of CheriABI function pointers were were integers.  I'll take care of this when I did the reaudit of syscalls on for (u)intptr_t.

davidchisnall, 2015-09-22:
Note that this is a blocker for the C tests that I'm currently working on, so any progress that can be made would help.

brooksdavis, 2015-09-23:
As of cb15cfdc9 sigaction installs and runs handlers, but doesn't call sa_action ones correctly or set `$c12` properly.

davidchisnall, 2015-09-23:
The `sa_action` part is the most important, as I need to be able to poke the register file to skip the faulting instruction in the tests.  If `$c12` is not set correctly, does this mean that `$t9` is?  If so, I can work around that with a little asm trampoline that sets up `$c12` by setting it to `$pcc` with the offset set to `$t9` and then jumps to the C handler.

brooksdavis, 2015-09-23:
It looks like `$t9` matches `$c12` so fixing the signal code property is required.

commit 9988083c (2015-09-25): Get signals working.

Add a CheriABI compatible signal trampoline, sendsig(), and sigreturn().

Simplify mcontext handling in the CheriABI case by not using a separate
location for the saved CHERI registers.  The size and layout of struct
mcontext would differ from mips64 due to mc_tls.

Resolves #29.
closed 2015-09-25 by brooksdavis