CTSRD-CHERI/cheribsd #32: strcmp crashes in cheriabi

davidchisnall, 2015-09-18:
char str[] = "123456789";
	char str2[] = "123456789";
	strcmp(str, str2);

This crashes.  My initial thought was that it was caused by the can-always-read-last-aligned-word assumption, but the fault is a read fault at 0x3, so not actually a capability address.  This may be a compiler bug...

davidchisnall, 2015-09-18:
This (and possibly #31) appears to be caused by `libc.a` in `usr/libcheri` being compiled with the wrong ABI.

	$ ar -x /home/dc552/CHERISDK/sdk/sysroot/usr/libcheri/libc.a
	$ ~/sdk/bin/llvm-objdump --triple cheri-unknown-freebsd -d -g strcmp.o
	
	strcmp.o:       file format ELF64-mips
	
	Disassembly of section .text:
	strcmp:
	       0:       90 8c 00 00     lbu     $12, 0($4)
	       4:       90 ad 00 00     lbu     $13, 0($5)
	...

This is MIPS code, not CHERI code.

brooksdavis, 2015-09-18:
It's not from invoking the compiler wrong, it's from compiling a wrong file.  I'll switch as many as I can over to C implementations for now.

Should we be rejecting assembly containing conventional loads and stores with -mabi=sandbox?

commit 3e1b4a68 (2015-09-18): Use C versions of several functions in the CheriABI case.

Resolves #32.  Some should probably be rewritten in assembly and #33 has
been opened to track them.
closed 2015-09-18 by brooksdavis