CTSRD-CHERI/cheribsd #34: printf %p prints 32-bit value

davidchisnall, 2015-09-22:
Printing a capability with base `0x7fffffe274` and offset 0 prints `0xffff03fe` with `%p`.

brooksdavis, 2015-09-22:
Hmm, printing a completely different pointer also produced `0xffff03fe`.

brooksdavis, 2015-09-22:
It's not immediately clear to me why `(uintmax_t)(uintptr_t)GETARG(void *)` is producing a constant `0xffff03fe` rather than the base plus the offset.

brooksdavis, 2015-09-22:
Since %p is implementation defined in the standard I'm somewhat tempted to make it produce something more interesting than the virtual address of the pointer, but that might be too surprising.

closed 2015-09-22 by brooksdavis

davidchisnall, 2015-09-23:
The `void*` -> `uintptr_t` cast will not change the representation, only the interpretation.  I'm not sure what `uintmax_t` is defined to in cheriabi.  If it's a primitive integer type, then this will get the offset of the value.  If it's `__uintptr_t` then it will not change the representation and the next implicit cast to a primitive type will get the offset.

commit b221d0b7 (2015-09-22): Use cgetbase+cgetoffset to print pointers via %p.

%p is implementation defined some something more robust is probably in
order, but this at least doesn't result in a constant 0xffff03fe.

This resolves #34.