CTSRD-CHERI/cheribsd #35: sig_atomic_t in CheriABI

rwatson, 2015-09-25:
In trying to compile csh(1), I found that it depended on a definition of sig_atomic_t, which appears not to be defined in CheriABI. @davidchisnall  thinks it might want to be a 64-bit integer type, as it is an integer type, whereas I take the view that we would like software that uses sig_atomic_t to hold pointers to work. Which raises the pragmatic question: how much software expects sig_atomic_t to be able to contain a pointer? If any non-trivial number, then perhaps it should be intcap_t?

brooksdavis, 2015-09-25:
sys/mips/include/signal.h defines sig_atomic_t unconditionally as:
{{{
typedef int sig_atomic_t;
}}}
All other 32-bit platforms do the same and other 64-bit platforms use `long`.  A quick check shows that linux defines it to `int` so there might be some BSD code that depends on being able to store a pointer, but no portable code does.

brooksdavis, 2015-09-25:
The issue @rwatson saw was a red-herring.  I've fixed the underlying build system issue in 38a395c.

closed 2015-09-25 by brooksdavis