CTSRD-CHERI/cheribsd #36: No posix_memalign(2) in CheriABI

rwatson, 2015-09-25:
In trying to build cp(1) with CheriABI, I encountered the fact that posix_memalign(2) is missing in CheriABI. This appears to be used by the POSIX.1e ACL code in libc.

brooksdavis, 2015-09-25:
`posix_memalign` is an interface to `malloc`'s storage pool.  Annoyingly the allocated memory needs to be something you can pass to `free` making allocation of very well aligned things a design requirement of `malloc`.  Fortunately, the only use in the system thus far is in `acl_init` where it can be replaced with mmap.

davidchisnall, 2015-09-29:
`valloc` is a wrapper around `posix_memalign`.  Currently, `valloc` does not appear to be in `stdlib.h`, but if it is added then it using it fails with a `posix_memalign` missing linker error.

davidchisnall, 2015-09-29:
I wonder if we also want a `malloc` interface that guarantees that the offset of the returned capability is zero.  Using one of the aligned-allocation interfaces may give this as a side effect, but there are probably some cases where you actually want it as the desired goal.

rwatson, 2015-09-29:
I think I'd actually take the stronger view: malloc() and friends should always return allocations with zero offsets. The only exception, in my current thinking, is mmap(2) (and any variations), which should always return zero offsets if the pointer isn't specified, but might return need to return an non-zero offset if the pointer is specified but isn't sufficiently aligned. We should add a new MAP_PRECISE  (or maybe MAP_ZEROOFFSET) flag that causes mmap(2) to fail if it can't provide a precise capability for the initial allocation. Maybe this will prove too strong a view in practice, but I think it's one we should aim for for now.

davidchisnall, 2015-09-29:
`malloc` currently returns capabilities with a 32 byte offset.  Unfortunately, this means that it is rounding up to a power of two, so if I ask for 1MB, I get a 2MB allocation, if I ask for 8MB, I get a 16MB allocation, and so on (detected by the `malloc` test in the C test suite).

rwatson, 2015-09-29:
Hopefully we will be able to migrate to jemalloc soon. The two dependencies I'm aware of currently are ISA-level: subtraction support, and capability-relative integer atomics.

davidchisnall, 2015-09-29:
Subtraction support shouldn't be needed at the ISA level - we implement a work-around in the compiler that isn't safe in the presence of GC, but should work fine for the internals of `malloc`.  We have capability-relative 64-bit integer arithmetic, what does jemalloc need?

brooksdavis, 2015-09-29:
jemalloc requires tls (which I haven't tested) and one 32-bit atomic which could be emulated, but Jon is working on it the instruction so I'm inclined to wait.

I could make a simple change to give allocations zero offsets at the cost of more expensive `free` calls.  If that's useful in the short term I'll do that.

I can't see any value in worrying about space efficiency.

brooksdavis, 2015-10-01:
It's now possible to compile a libc with jemalloc, but I'm waiting for hardware that supports the cllw/cscw instructions.

brooksdavis, 2017-03-20:
We've switched to jemalloc for CheriABI code so this is done.

closed 2017-03-20 by brooksdavis