CTSRD-CHERI/cheribsd #37: errno difficulties in CheriABI: errno possibly broken, and progname not printed correctly by err(3).

rwatson, 2015-09-26:
Compiling 'realpath' for CheriABI, it appears to work, but error handling is wonky when a file doesn't exist:

root@beri1:/tmp # ./realpath foo
/tmp/foo
root@beri1:/tmp # ./realpath bar
: bar: No error: 0
root@beri1:/tmp # realpath bar
realpath: bar: No such file or directory

It looks likely that errno is not being set (or perhaps not being accesses) correctly, and also that the getprogname() mechanism used in err(3) is not working.

brooksdavis, 2015-09-28:
I spend a fair bit of time trying to debug `__cerror` early last week.  According to streamtrace it seems to be writing the correct value back to the address of `errno`, but it's not being seen in after the function returns an error.

brooksdavis, 2015-09-28:
I've fixed the `getprogname()` issue in 056e95bfe944770f21df6673979514334af8b0f9 (we were resetting `__progname` to a default value due to out of order initialization.)

rwatson, 2015-09-29:
I can confirm that getprogname() is now working more happily:

root@beri1:/tmp # ./realpath foo
realpath: foo: No error: 0

commit bf47faa8 (2015-09-30): Write errno as an int.

Fixes #37.
closed 2015-09-30 by brooksdavis