CTSRD-CHERI/cheribsd #46: /bin/date experiences intermittent crashes with CheriABI

rwatson, 2015-10-25:
If I compile /bin/date with CheriABI:

```
--- a/bin/date/Makefile
+++ b/bin/date/Makefile
@@ -3,6 +3,9 @@
 
 .include <src.opts.mk>
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  date
 SRCS=  date.c netdate.c vary.c
```

Then I experience intermittent crashes with the resulting binary:
```
> ./date
Sat Mar 14 17:44:00 UTC 2015
> ./date
Signal 34
```

This may have to do with CFI in CheriABI:

```
FreeBSD/mips (beri1) (ttyj0)

login: cpuid = 0
cpuid = 0
CHERI cause: ExcCode: 0x11 RegNum: C12 (permit execute violation)
C00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
C01: v:1 s:0 p:7fff81fd b:000000012015b680 l:0000000000000900 o:300 t:0
C02: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12012f8e1 t:0
C03: v:1 s:0 p:7fff81fd b:000000012015b680 l:0000000000000900 o:300 t:0
C04: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12012f8e1 t:0
C05: v:1 s:0 p:7fff81fd b:0000007fffffda80 l:0000000000000400 o:400 t:0
C06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12012a0e0 t:0
C07: v:1 s:0 p:7fff81fd b:0000007fffffc09c l:0000000000000004 o:0 t:0
C08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120159bc0 t:0
C09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffcc44 t:0
C10: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffcc40 t:0
C11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
C12: v:1 s:0 p:7fff81fd b:000000012015b680 l:0000000000000900 o:120052d38 t:0
C13: v:1 s:0 p:000081d5 b:0000007fffffd680 l:0000000000000020 o:0 t:0
C14: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffcc48 t:0
C15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
C16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
C17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002a2c t:0
C18: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002060 t:0
C19: v:1 s:0 p:7fff81fd b:0000007fffffd960 l:0000000000000060 o:0 t:0
C20: v:1 s:0 p:7fff81fd b:0000007fffffda80 l:0000000000000400 o:0 t:0
C21: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120005790 t:0
C22: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002060 t:0
C23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
C24: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201254f0 t:0
C26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
C31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12002643c t:0
Mar 14 17:44:05 beri1 kernel: USER_CHERI_EXCEPTION: pid 948 tid 100045 (date), uid 999: CP2 fault (type 0x32)
Mar 14 17:44:05 beri1 kernel: Trapframe Register Dump:
Mar 14 17:44:05 beri1 kernel: zero: 0   at: 0x120052d38 v0: 0x12015b560 v1: 0x1c
Mar 14 17:44:05 beri1 kernel: a0: 0x7fffffd678  a1: 0x2 a2: 0x35        a3: 0x7fffffb540
Mar 14 17:44:05 beri1 kernel: a4: 0x7fffffbf58  a5: 0x7fffffbf50        a6: 0x7fffffbf18       a6: 0x7fffffbf10
Mar 14 17:44:05 beri1 kernel: t0: 0x14  t1: 0x1e        t2: 0x7fffffbb00        t3: 0x7fffffba80
Mar 14 17:44:05 beri1 kernel: t8: 0x2   t9: 0x1200263d0 s0: 0x400       s1: 0x20
Mar 14 17:44:05 beri1 kernel: s2: 0x20  s3: 0   s4: 0x8 s5: 0
Mar 14 17:44:05 beri1 kernel: s6: 0     s7: 0xffffffffffffffff  k0: 0   k1: 0
Mar 14 17:44:05 beri1 kernel: gp: 0x12017d840   sp: 0x7fffffd5e0        s8: 0x7fffffd5e0       ra: 0x7fffffd7f8
Mar 14 17:44:05 beri1 kernel: sr: 0x4000fcf3    mullo: 0x6666666666666667       mulhi: 0       badvaddr: 0x1200263d0
Mar 14 17:44:05 beri1 kernel: cause: 0x148      pc: 0x12002643c
```

Code dump from printf() in /bin/date -- crash occurs in the CJALR at 0x12002643c:

```
00000001200263d0 <printf>:
   1200263d0:   67bdff60        daddiu  sp,sp,-160
   1200263d4:   67a40098        daddiu  a0,sp,152
   1200263d8:   ebcb2003        csd     s8,a0,0(c11)
   1200263dc:   fa2be860        csc     c17,sp,96(c11)
   1200263e0:   03a0f02d        move    s8,sp
   1200263e4:   49b96002        cgetoffset      t9,c12
   1200263e8:   3c010015        lui     at,0x15
   1200263ec:   0039082d        daddu   at,at,t9
   1200263f0:   64217470        daddiu  at,at,29808
   1200263f4:   67c20020        daddiu  v0,s8,32
   1200263f8:   49a15881        csetoffset      c1,c11,v0
   1200263fc:   64020020        daddiu  v0,zero,32
   120026400:   48210880        csetbounds      c1,c1,v0
   120026404:   f9a10000        csc     c13,zero,0(c1)
   120026408:   6422afc0        daddiu  v0,at,-20544
   12002640c:   48810087        cfromptr        c1,c0,v0
   120026410:   c8410003        cld     v0,zero,0(c1)
   120026414:   64218238        daddiu  at,at,-32200
   120026418:   48810047        cfromptr        c1,c0,at
   12002641c:   c8210003        cld     at,zero,0(c1)
   120026420:   48810087        cfromptr        c1,c0,v0
   120026424:   d8210000        clc     c1,zero,0(c1)  
   120026428:   49a21800        cmove   c2,c3
   12002642c:   48001807        cgetpcc c3
   120026430:   49ac1841        csetoffset      c12,c3,at
   120026434:   49a30800        cmove   c3,c1
   120026438:   49a41000        cmove   c4,c2
   12002643c:   48f16000        cjalr   c17,c12                <--- Crash here
   120026440:   49a56800        cmove   c5,c13
   120026444:   03c0e82d        move    sp,s8
   120026448:   da2be860        clc     c17,sp,96(c11)
   12002644c:   67a10098        daddiu  at,sp,152
   120026450:   cbcb0803        cld     s8,at,0(c11)
   120026454:   49008800        cjr     c17
   120026458:   67bd00a0        daddiu  sp,sp,160
   12002645c:   00000000        nop
```

The execute bit in $c12 is indeed not set:

C12: v:1 s:0 p:7fff81fd b:000000012015b680 l:0000000000000900 o:120052d38 t:0

The value in $c12 is a little surprising, given both a base and an offset that resemble virtual addresses. Presumably only one of these should resemble a virtual address. This looks a bit like it might be a processor bug, as $c3 is loaded from $pcc only a few instructions earlier, and although CSetOffset is now an instruction of ill repute, this is 256-bit CHERI and so this should in fact be fine. A second set of eyes on the code generation here would be useful; I think @davidchisnall or @michael-roe had reported seeing incorrect values come out of $pcc in recent testing?

davidchisnall, 2015-10-25:
Hmm, this looks like a failure that I was seeing that went away when I updated the bitfile.  I believe that it's a hardware bug (though it's possible that it's a bug in the context switch implementation).  It's not possible for `$c12` to not have the execute permission set, on a correct implementation:

	12002642c:   48001807        cgetpcc c3

`$3` now contains `$pcc`, which must have the execute permission set (or we'd be seeing exceptions in instruction fetch).

	120026430:   49ac1841        csetoffset      c12,c3,at

`$c12` now has a capability that is `$pcc` with the offset set to the jump target.  No change to permissions.

	120026434:   49a30800        cmove   c3,c1
	120026438:   49a41000        cmove   c4,c2

No modification of `c12`

	12002643c:   48f16000        cjalr   c17,c12                <--- Crash here

And now C12 does not have the execute permission set.  The error I was seeing was subtly different, though it was on the same set of code.  I was seeing appeared to have `CGetPCC` return the value of `$c0`, but it still looks as if there's some kind of register forwarding bug.  I would assign the bug to Jon, but he doesn't appear to have a GitHub account?

jonwoodruff, 2015-10-27:
Do we know what bitfile was used for this failure?  As davidchisnall mentioned, recent bitfiles have register file fixes.

davidchisnall, 2015-10-27:
I was seeing a similar issue, but it went away with `20151019-cheri-de4-jdw57.sof.bz2`.  Not sure if @rwatson is using an older one?

brooksdavis, 2015-10-29:
It looks like this was -cheri-linker bug.  There might have also been a hardware issue, but I can't replicate with the latest compiler and cherubs and the 20151019 bitfile.

closed 2015-10-29 by brooksdavis