CTSRD-CHERI/cheribsd #47: /bin/df experiences alignment error with CheriABI

rwatson, 2015-10-25:
When running a CheriABI version of /bin/df, I experience bus errors:

```
> ./df
Filesystem          1K-blocks    Used    Avail Capacity  Mounted on
Bus error
```

Compiled with the following modifications:

```
--- a/bin/df/Makefile
+++ b/bin/df/Makefile
@@ -4,6 +4,9 @@
 MOUNT= ${.CURDIR}/../../sbin/mount
 .PATH: ${MOUNT}
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  df
 SRCS=  df.c vfslist.c
```

Console output reports an alignment error:

```
FreeBSD/mips (beri1) (ttyj0)

login: cpuid = 0
C00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
C01: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12009eca0 t:0
C02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
C03: v:1 s:0 p:7fff81fd b:000000012009ede0 l:0000000000000900 o:48 t:0
C04: v:1 s:0 p:7fff81fd b:0000007fffffb7a0 l:0000000000000260 o:20 t:0
C05: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120085bc0 t:0
C06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12006bda0 t:0
C07: v:1 s:0 p:7fff81fd b:0000007fffffbbdc l:0000000000000004 o:0 t:0
C08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffa9c8 t:0
C09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffa9f8 t:0
C10: v:1 s:0 p:7fff81fd b:0000007fffffc2e4 l:0000000000000004 o:0 t:0
C11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
C12: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120052d78 t:0
C13: v:1 s:0 p:000081d5 b:0000007fffffd320 l:0000000000000008 o:0 t:0
C14: v:1 s:0 p:7fff81fd b:0000007fffffc380 l:0000000000000020 o:0 t:0
C15: v:1 s:0 p:7fff81fd b:0000007fffffc360 l:0000000000000020 o:0 t:0
C16: v:1 s:0 p:7fff81fd b:0000007fffffc1e0 l:0000000000000020 o:0 t:0
C17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12004aec8 t:0
C18: v:1 s:0 p:7fff81fd b:000000012009ede0 l:0000000000000900 o:48 t:0
C19: v:1 s:0 p:7fff81fd b:0000007fffffbba0 l:0000000000000020 o:0 t:0
C20: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120099870 t:0
C21: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120085bc0 t:0
C22: v:1 s:0 p:7fff81ff b:00000001600ef000 l:0000000000004000 o:25 t:0
C23: v:1 s:0 p:7fff81ff b:00000001600ef000 l:0000000000004000 o:25 t:0
C24: v:1 s:0 p:7fff81fd b:0000007fffffbb90 l:0000000000000004 o:0 t:0
C26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
C31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120052ea0 t:0
cpuid = 0
C00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
C01: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12009eca0 t:0
C02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
C03: v:1 s:0 p:7fff81fd b:000000012009ede0 l:0000000000000900 o:48 t:0
C04: v:1 s:0 p:7fff81fd b:0000007fffffb7a0 l:0000000000000260 o:20 t:0
C05: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120085bc0 t:0
C06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12006bda0 t:0
C07: v:1 s:0 p:7fff81fd b:0000007fffffbbdc l:0000000000000004 o:0 t:0
C08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffa9c8 t:0
C09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffa9f8 t:0
C10: v:1 s:0 p:7fff81fd b:0000007fffffc2e4 l:0000000000000004 o:0 t:0
C11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
[... 106 lines cut ...]
   120052e98:   48810047        cfromptr        c1,c0,at
   120052e9c:   d8411000        clc     c2,v0,0(c1)
   120052ea0:   f8520000        csc     c2,zero,0(c18)                   <--- Crash here
   120052ea4:   fa411000        csc     c18,v0,0(c1)
   120052ea8:   03c0e82d        move    sp,s8
   120052eac:   da2be820        clc     c17,sp,32(c11)
   120052eb0:   da4be840        clc     c18,sp,64(c11)
   120052eb4:   da6be860        clc     c19,sp,96(c11)
   120052eb8:   da8be880        clc     c20,sp,128(c11)
   120052ebc:   daabe8a0        clc     c21,sp,160(c11)
   120052ec0:   67a100d0        daddiu  at,sp,208
   120052ec4:   ca0b0803        cld     s0,at,0(c11)
   120052ec8:   67a100d8        daddiu  at,sp,216
   120052ecc:   cbcb0803        cld     s8,at,0(c11)
   120052ed0:   49008800        cjr     c17
   120052ed4:   67bd00e0        daddiu  sp,sp,224
```

The value in $c18 is indeed inadequately aligned for a CSC instruction with 0 offset:

```
C18: v:1 s:0 p:7fff81fd b:000000012009ede0 l:0000000000000900 o:48 t:0
```

Possibly a compiler bug, but another set of eyes would be appreciated.

davidchisnall, 2015-10-25:
The value in `$c18` is coming from `$c3` on function entry.  This is the first capability argument.  I don't know what `__Bfree_D2A` is - is it a MIPS mangling of `free()`?  If so, then one of the following is happening:

- `malloc()` is returning something that is not capability aligned.
- The argument to `free()` was not previously a return value of `malloc()`
- The value being passed here is the result of some incorrect pointer arithmetic on the return value of `malloc()` (either in the original code, or as a result of incorrect optimisation).

brooksdavis, 2015-10-29:
`__Bfree_D2A` is `Bfree` from `contrib/gdtoa/misc.c` with this define from `gdtoaimp.h` applied

```
#define      Bfree           __Bfree_D2A
```

closed 2015-10-29 by brooksdavis

rwatson, 2015-10-30:
Tagged as CheriCCompat to keep track of a case where source-code modification was required for CHERI compatibility: alignment requirement for pointers in a custom allocator.

commit aa8753a8 (2015-10-29): Fix gdtoa's internal allocator for CHERI.

The allocator incorrectly assumed that doubles hand equal or stronger
alignment constraints than pointers.

Increase the size of the static buffer to account for 256-bit pointers.

Fixes #47.