CTSRD-CHERI/cheribsd #48: intptr_t is not intcap_t in CheriABI mode

davidchisnall, 2015-10-26:
#include <stdint.h>
	
	_Static_assert(sizeof(uintptr_t) == sizeof(void*), "uintptr_t size is wrong");
	_Static_assert(sizeof(intptr_t) == sizeof(void*), "intptr_t size is wrong");

These static assertions fail and this is currently breaking some of the tests in Peter Sewell's C test suite.

brooksdavis, 2015-10-26:
This is near the top of my CheriABI todo list.  Last time I tried I produced a round of compiler crash bugs. :)

davidchisnall, 2015-11-03:
Are the relevant compiler bugs fixed?  If not, then can we have a `#define` to toggle it for simple test programs?  Peter Sewell's tests that use `intptr_t` are currently all failing and it would be nice to have good results for them soon.

brooksdavis, 2015-11-03:
It's still crashing.  I'll get a bug report submitted after breakfast.

brooksdavis, 2015-11-03:
The current patch is:

```C
commit c4d99163955224b89e9553c7694be6edc484c744
Author: Brooks Davis <brooks@one-eyed-alien.net>
Date:   Tue Oct 27 22:41:32 2015 +0000

    Use __(u)intcap_t for pointer types in the pure capability ABI.

diff --git a/sys/mips/include/_types.h b/sys/mips/include/_types.h
index f2d67d5..6afe3b9 100644
--- a/sys/mips/include/_types.h
+++ b/sys/mips/include/_types.h
@@ -76,10 +76,16 @@ typedef     __int32_t       __clock_t;              /* clock()... */
 typedef        double          __double_t;
 typedef        float           __float_t;
 #ifdef __mips_n64
+#ifndef __CHERI_SANDBOX__
 typedef        __int64_t       __critical_t;
 typedef        __int64_t       __intfptr_t;
 typedef        __int64_t       __intptr_t;
 #else
+typedef        __intcap_t      __critical_t;
+typedef        __intcap_t      __intfptr_t;
+typedef        __intcap_t      __intptr_t;
+#endif
+#else
 typedef        __int32_t       __critical_t;
 typedef        __int32_t       __intfptr_t;
 typedef        __int32_t       __intptr_t;
@@ -106,9 +112,14 @@ typedef    __int64_t       __segsz_t;
 typedef        __int64_t       __semid_t;              /* historically intptr_t */
 typedef        __uint64_t      __size_t;
 typedef        __int64_t       __ssize_t;
+#ifndef __CHERI_SANDBOX__
 typedef        __uint64_t      __uintfptr_t;
 typedef        __uint64_t      __uintptr_t;
 #else
+typedef        __uintcap_t     __uintfptr_t;
+typedef        __uintcap_t     __uintptr_t;
+#endif
+#else
 typedef        __int32_t       __ptrdiff_t;            /* ptr1 - ptr2 */
 typedef        __int32_t       __segsz_t;              /* segment size (in pages) */
 typedef        __int32_t       __semid_t;              /* historically intptr_t */
```

brooksdavis, 2015-11-03:
I've added #60 as a pull request of a building but untested change.  My tablet's controller seems to be offline so I've opened this in hopes that someone else can test it.

brooksdavis, 2015-11-05:
Resolved in 97705997154b73e14bd1531db856054d71ade5db.  It differs somewhat from #60.

closed 2015-11-05 by brooksdavis