CTSRD-CHERI/cheribsd #49: CheriBSD needs a portable way to report CHERI exceptions

rwatson, 2015-10-26:
Currently, we add SIGPROT but require use of a non-portable field in the machine-dependent signal context to resolve different failure modes (e.g., untagged pointer, bounds check, and permissions error) from one another. In due course, it would be desirable to offer a portable API to query the lower-level code information. My leaning is towards  portable extension to sigcontext, but I could also yet be convinced that more signals would be appropriate (although this would increase the burden of maintaining additional signals in CheriBSD).

davidchisnall, 2015-10-26:
I'd like the `sigcontext` to contain the two things that we currently pass to the handler in the CHERI C test suite:

- A `void*` indicating the capability that faulted.
- A numeric code (with a set of associated `enum` values for possible fault types) with the cause.

Anything that's going to do recovery needs to know a lot more about the faulting instructions and so will poke the `mcontext`, but this is enough to get useful diagnostics.

rwatson, 2015-10-30:
This is roughly what I had in mind; I wonder if a portable representation of $pcc would be useful as well.

davidchisnall, 2015-10-30:
It's not clear to me that it would.  The existing stack unwind code can't currently unwind CHERI stack frames, but it ought to be able to, which would allow you to get a back trace to the signal delivery point.  Anything that is going to use `$pc` / `$pcc` to *recover* from the error will be machine-specific and will need to understand `ucontex_t`.  The one case where it might be useful portably is if `$pcc` is enough to uniquely identify the sandbox and no other mechanism exists.

rwatson, 2015-10-30:
For me this was less about stack unwinding, and more about being able to print the current program counter, and in the event of a bounds-check failure or similar being caught, other debugging information. The other interesting aspect would relate to the trusted stack, where because there are only fixed-size frames with return state, not a general stack, it can be processed without detailed unwind information.

rwatson, 2016-05-30:
Note that this applies to the signal-handling code, but also reporting of CHERI exception information in ```ktrace(1)``` as well.

rwatson, 2017-03-19:
A ``si_code`` implementation for ```SIGPROT`` is prototyped in cf6952c1c0b4cae27e736dd59fb5aa8af1db0ade, and tested by 43ab85fba906ffb67ccee6a1c91e71f9a391ad2c.

closed 2017-03-19 by rwatson