CTSRD-CHERI/cheribsd #50: Loaded capabilities for CJALR targets appear (sometimes) $c0-derived rather than $pcc-derived?

rwatson, 2015-10-27:
When I compile /bin/echo with CheriABI:

```
--- a/bin/echo/Makefile
+++ b/bin/echo/Makefile
@@ -1,6 +1,9 @@
 #      @(#)Makefile    8.1 (Berkeley) 5/31/93
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  echo
 
 .include <bsd.prog.mk>
```

(and also with other programs) I frequently see crashes involving trying to CJALR to a $c12 that doesn't have the execute bit set:

```
login: cpuid = 0
cpuid = 0
CHERI cause: ExcCode: 0x11 RegNum: $c12 (permit execute violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120069060 t:0
$c02: v:1 s:0 p:7fff81ff b:0000000160069fe0 l:0000000000000100 o:a0 t:0
$c03: v:1 s:0 p:7fff81ff b:0000000160069fe0 l:0000000000000100 o:20 t:0
$c04: v:1 s:0 p:7fff81ff b:000000016006aee0 l:0000000000000100 o:0 t:0
$c05: v:1 s:0 p:7fff81ff b:000000016006aee0 l:0000000000000100 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffdf40 l:0000000000000020 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffdf20 l:0000000000000020 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffdf00 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120058e10 t:0
$c10: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12004b660 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120011330 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120001dc4 t:0
$c18: v:1 s:0 p:7fff81ff b:0000000160069fe0 l:0000000000000100 o:20 t:0
$c19: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002040 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffddf4 l:0000000000000004 o:0 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffde60 l:0000000000000020 o:0 t:0
$c22: v:1 s:0 p:7fff81fd b:0000007fffffde98 l:0000000000000004 o:0 t:0
$c23: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffe240 t:0
$c24: v:1 s:0 p:7fff81fd b:0000007fffffde40 l:0000000000000020 o:0 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002078 t:0
Mar 14 21:44:56 beri1 kernel: USER_CHERI_EXCEPTION: pid 788 tid 100064 (echo), uid 999: CP2 fault (type 0x32)
Mar 14 21:44:56 beri1 kernel: Trapframe Register Dump:
Mar 14 21:44:56 beri1 kernel: zero: 0   at: 0x400       v0: 0x120002040 v1: 0
Mar 14 21:44:56 beri1 kernel: a0: 0x1   a1: 0x2 a2: 0x1002      a3: 0
Mar 14 21:44:56 beri1 kernel: a4: 0     a5: 0   a6: 0   a6: 0
Mar 14 21:44:56 beri1 kernel: t0: 0     t1: 0   t2: 0   t3: 0
Mar 14 21:44:56 beri1 kernel: t8: 0     t9: 0x120002040 s0: 0x2 s1: 0x120081110
Mar 14 21:44:56 beri1 kernel: s2: 0x400 s3: 0xffffffffffffffff  s4: 0x2 s5: 0x2
Mar 14 21:44:56 beri1 kernel: s6: 0x2   s7: 0x63        k0: 0   k1: 0
Mar 14 21:44:56 beri1 kernel: gp: 0x40  sp: 0x7fffffdca0        s8: 0x7fffffdca0
        ra: 0x80
Mar 14 21:44:56 beri1 kernel: sr: 0x4000fcf3    mullo: 0x8000   mulhi: 0       badvaddr: 0x16006a0e0
Mar 14 21:44:56 beri1 kernel: cause: 0x148      pc: 0x120002078
Mar 14 21:45:36 beri1 sshd[791]: error: Could not load host key: /etc/ssh/ssh_host_dsa_key
Mar 14 21:47:55 beri1 sshd[793]: error: Could not load host key: /etc/ssh/ssh_host_dsa_key
```

The code disassembles as follows:

```
0000000120002040 <writev>:
   120002040:   67bdffa0        daddiu  sp,sp,-96
   120002044:   ebcbeac3        csd     s8,sp,88(c11)  
   120002048:   fa2be820        csc     c17,sp,32(c11)
   12000204c:   03a0f02d        move    s8,sp
   120002050:   49b96002        cgetoffset      t9,c12
   120002054:   3c010008        lui     at,0x8
   120002058:   0039082d        daddu   at,at,t9
   12000205c:   6421f0d0        daddiu  at,at,-3888
   120002060:   64218068        daddiu  at,at,-32664
   120002064:   48810047        cfromptr        c1,c0,at
   120002068:   c8210003        cld     at,zero,0(c1)
   12000206c:   48810047        cfromptr        c1,c0,at
   120002070:   64010400        daddiu  at,zero,1024
   120002074:   d9810800        clc     c12,at,0(c1)
   120002078:   48f16000        cjalr   c17,c12                         <--- Boom here
   12000207c:   00000000        nop
   120002080:   03c0e82d        move    sp,s8
   120002084:   da2be820        clc     c17,sp,32(c11)
   120002088:   cbcbeac3        cld     s8,sp,88(c11)
   12000208c:   49008800        cjr     c17
   120002090:   67bd0060        daddiu  sp,sp,96
```

It looks like $c12 is loaded from a vector of initialised capabilities. $c12 points where one might expect, the __sys_writev system call stub:

```
0000000120011330 <__sys_writev>:
   120011330:   34020079        li      v0,0x79
   120011334:   0000000c        syscall
   120011338:   14e00003        bnez    a3,120011348 <err>
   12001133c:   00000000        nop
   120011340:   49008800        cjr     c17  
   120011344:   00000000        nop
```

This suggests that the pointer value for this capability is correct, but the permissions are wrong. I'm guessing that (perhaps) the linker isn't setting up capabilities to __sys_writev() and friends with suitable permissions? Is the C startup/runtime code always deriving statically initialised capabilities from $c0, even though sometimes they should be $pcc-derived? Or is the linker not passing on the right information to select the right originating capability?

(@davidchisnall may also take a view on this?)

davidchisnall, 2015-10-27:
If this is a global, then it will be initialised by data provided by `capsizefix`, which differentiates code and data capabilities by the section that of the thing that they're pointing to.  If you do `llvm-objdump -C` then that will show you what it's initialising it with.  The length appears correct, which implies that either:

- `capsizefix` has determined that it's a code capability.
- `capsizefix` hasn't been able to find the relevant pointer and so has left the length 0.

In the latter case, you should be given an error about it when `capsizefix` runs.

rwatson, 2015-10-27:
```llvm-objdump -C``` returns:

```
> ~/git/llvm/Build/bin/llvm-objdump -C ~/obj/mips.mips64/home/rnw24/git/cheribsd/bin/echoecho | grep -i writev
0x0000000120069460      Base: __sys_writev (0x0000000120011330) Offset: 0000000000000000        Length: 0000000000000000
```

rwatson, 2015-10-27:
One further data point: __sys_writev() is an assembly-language system-call stub generated from syscalls.master output during the libc build.

```
rnw24@vica:~/obj/mips.mips64/home/rnw24/git/cheribsd/lib/libc> more _writev.S
#include "compat.h"
#include "SYS.h"
PSEUDO(writev)
```

Which would be expanded by this macro in the MIPS version of SYS.h:

```
#define PSEUDO(x)                                                       \
LEAF(__sys_ ## x);                                                      \
        .weak _C_LABEL(x);                                              \
        _C_LABEL(x) = _C_LABEL(__CONCAT(__sys_,x));                     \
        .weak _C_LABEL(__CONCAT(_,x));                                  \
        _C_LABEL(__CONCAT(_,x)) = _C_LABEL(__CONCAT(__sys_,x));         \
        PIC_PROLOGUE(__sys_ ## x);                                      \
        SYSTRAP(x);                                                     \
        bne a3,zero,err;                                                \
        PIC_RETURN();                                                   \
err:                                                                    \
        PIC_TAILCALL(__cerror);                                         \
END(__sys_ ## x)
```

rwatson, 2015-10-28:
Off-hand, a bit of digging around in the ```csu``` code and the binary suggests that most code is ending up with useful function pointers, but this specific code is not. The implementation of ```__sys_writev``` does appear to be in the right section:

```
  2 .text         0004cfb0  0000000120001100  0000000120001100  00001100  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

brooksdavis, 2015-10-28:
In my copy of echo, the bad jump target is:

```
C12: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120011260 t:0
```

That is also __sys_writev:

```
entry: 1447
        st_name: __sys_writev
        st_value: 0x120011260
        st_size: 0
        st_info: STT_NOTYPE STB_GLOBAL
        st_shndx: 3
```

According to elfdump, it is in fact not tagged as a function (high bit of permissions is 0) so it is derived from DDC and fails as expected:

```
entry: 33
        capability location: 0x120069460
        object: 0x120011260
        offset: 0
        size: 0
        permissions: 0
```

@davidchisnall this clearly a toolchain bug, how best to debug if?

brooksdavis, 2015-12-01:
-cheri-linker was disabled by default in b4e6b192c6cc57b36d28a708301b479ab11dc288 due to this issue.

davidchisnall, 2015-12-01:
Does this still occur?  I modified `capsizefix` a while ago to use `pcc`-relative caps for things that are in the text section.

brooksdavis, 2015-12-01:
I saw the warnings about not finding sizes and wrongly assumed it wasn't fixed, but echo(1) now works.  I'll switch the default back to `WITH_CHERI_LINKER`.

closed 2015-12-01 by brooksdavis