CTSRD-CHERI/cheribsd #51: Surprising EACCES returned by open(2) on CheriABI

rwatson, 2015-10-29:
Having compiled ```nvi``` to use CheriABI:

```
--- a/usr.bin/vi/Makefile
+++ b/usr.bin/vi/Makefile
@@ -4,6 +4,9 @@
 
 .include <src.opts.mk>
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 SRCDIR=                ${.CURDIR}/../../contrib/nvi
 SUBDIR+=       catalog
```

I get a mysterious and unexpected ```EACCES``` back from its call to open a temporary file:

```
> ktrace ./nvi
ex/vi: Input encoding conversion not supported
ex/vi: Error: /tmp/vi.fnX81ZifUk: Permission denied
Service unavailable> 
```

When I inspect trace output, I see:

```
  1010 nvi      CALL  stat(0x16027a420,0x7fffffcd88)
  1010 nvi      NAMI  "/tmp"
  1010 nvi      STRU  struct stat {dev=106, ino=2, mode=041777, nlink=7, uid=0, gid=0, rdev=312, atime=1426476132, stime=1426481931.003911000, ctime=1426481931.003911000, birthtime=1426476132, size=1536, blksize=32768, blocks=8, flags=0x0 }
  1010 nvi      RET   stat 0
  1010 nvi      CALL  open(0x16027a420,0xa02<O_RDWR|O_CREAT|O_EXCL>,0xa<S_IXGRP|S_IWOTH>)
  1010 nvi      NAMI  "/tmp/vi.ElkbxBzqK6"
  1010 nvi      RET   open 3
  1010 nvi      CALL  fstat(0x3,0x7fffffd1a8)
  1010 nvi      STRU  struct stat {dev=106, ino=78, mode=0100010, nlink=1, uid=999, gid=0, rdev=0, atime=1426481931.008197000, stime=1426481931.008197000, ctime=1426481931.008197000, birthtime=1426481931.008167000, size=0, blksize=32768, blocks=0, flags=0x0 }
  1010 nvi      RET   fstat 0
  1010 nvi      CALL  close(0x3)
...
  1010 nvi      CALL  open(0x16027a420,0x4<O_NONBLOCK>,<unused>0x1b6)
  1010 nvi      NAMI  "/tmp/vi.ElkbxBzqK6"
  1010 nvi      RET   open -1 errno 13 Permission denied
  1010 nvi      CALL  stat(0x7fffffbdd8,0x7fffffc348)
```

Non-CheriABI ```nvi``` seems to work better:

```
  1032 nvi      CALL  stat(0x160884100,0x7fffffd0c0)
  1032 nvi      NAMI  "/tmp"
  1032 nvi      STRU  struct stat {dev=106, ino=2, mode=041777, nlink=7, uid=0, gid=0, rdev=312, atime=1426476132, stime=1426482151.223722000, ctime=1426482151.223722000, birthtime=1426476132, size=1536, blksize=32768, blocks=8, flags=0x0 }
  1032 nvi      RET   stat 0
  1032 nvi      CALL  open(0x160884100,0xa02<O_RDWR|O_CREAT|O_EXCL>,0x180<S_IRUSR|S_IWUSR>)
  1032 nvi      NAMI  "/tmp/vi.mFhBHhFxyQ"
  1032 nvi      RET   open 3
  1032 nvi      CALL  fstat(0x3,0x7fffffd6d0)
  1032 nvi      STRU  struct stat {dev=106, ino=79, mode=0100600, nlink=1, uid=999, gid=0, rdev=0, atime=1426482151.229216000, stime=1426482151.229216000, ctime=1426482151.229216000, birthtime=1426482151.229187000, size=0, blksize=32768, blocks=0, flags=0x0 }
  1032 nvi      RET   fstat 0
  1032 nvi      CALL  close(0x3)
...
  1032 nvi      CALL  open(0x160884100,0x4<O_NONBLOCK>,<unused>0x1b6)
  1032 nvi      NAMI  "/tmp/vi.mFhBHhFxyQ"
  1032 nvi      RET   open 3

```
This look pretty similar, but maybe there's a bug in the ABI wrapper for ```open(2)```...?

rwatson, 2015-10-29:
Oh - the requested permissions are different in initial creation -- CheriABI is seeing ```S_IXGRP|S_IWOTH```, whereas MIPS is seeing ```S_IRUSR|S_IWUSR```. I wonder if that's userspace or kernel-side. Will need to investigate in the morning. But this would explain the problem -- in CheriABI, we end up with a temporary file with the wrong permissions.

brooksdavis, 2015-10-30:
I think know what the problem is.  The posix definition of open is an abomination.  It is declared as `int open(char *path, int flags, ...);` but the syscall is `int open(char *path, int flags, int mode);`  Since @davidchisnall changed things to always spill all varargs to the stack, I suspect we're passing garbage in `a1` as mode.

With the current ABI we need a userspace wrapper to unpack mode and pass it in `a1`.

I do wonder how many more abuses like this there are.  This is an interesting de facto C artifact.

brooksdavis, 2015-10-31:
Reopening as I tested the wrong program.  I believe the fix should work...

brooksdavis, 2015-11-12:
93b590767136cab357084c999e5d352041552ac6 was not the solution.  An appropriate (and virtually identical) wrapper already existed in for form of the interposers to support libc.  Those wrappers were not being linked due to the bugs fixed in 4e8e13c90fc6a80e1520de44a6864cfd78b3b56d.

commit 93b59076 (2015-10-31): Wrap open() to extract mode from varargs in the CheriABI case.

The open() syscall is declared publically as:

int	open(const char *path, int flags, ...);

put the system call is declared as:

int	open(const char *path, int flags, int mode);

This has worked because the common varargs API passes as many arguments
as possible in registers and only spills to the stack when required to
so the calling conventions are identical.  In the pure-capabilty ABI we
always spill arguments to the stack and pass a capability to them which
allows for bounds checking and limited type checking.  Unfortunatly, t [... cut]
closed 2015-11-12 by brooksdavis