CTSRD-CHERI/cheribsd #52: Possible continuing nsswitch problem for CheriABI

rwatson, 2015-10-30:
I've noticed that a number of CheriABI programs are having problems looking up user and group information, suggesting a problem in the ```nsswitch``` mechanism. For example, ```id(1)``` doesn't list usernames/groups:

```
> ./id
uid=999 gid=999 groups=999,0
> id
uid=999(ctsrd) gid=999(ctsrd) groups=999(ctsrd),0(wheel)
```

And the ```finger(1)``` command also can't look up users:

```
> ./finger ctsrd
finger: ctsrd: no such user
```

rwatson, 2015-11-27:
After diagnosing this via ISA-level tracing for a bit, it looks like we are encountering a bug in the implementation of CFromPtr in which a capability derived from an integer has a length of zero, which means a later cast back to an integer via CToPtr results in NULL rather than the expected integer 1. This reflects a couple of different problems: first, that the hardware is not correctly implementing CFromPtr as specified (bug report filed with Jon), and second, that we likely should not be using CFromPtr/CToPtr in pure-capability binaries as $c0 may not be meaningful -- instead, we might need a CFromInt that returns an untagged capability NULL with a suitable CSetOffset (also discussed with @davidchisnall). I will follow up on this report if a hardware fix resolves this problem.

brooksdavis, 2015-12-04:
With @davidchisnall's update `id` works:

```
root@beri1:/home/ctsrd # id
uid=0(root) gid=0(wheel) groups=0(wheel),5(operator)
root@beri1:/home/ctsrd # id ctsrd
uid=999(ctsrd) gid=999(ctsrd) groups=999(ctsrd),0(wheel)
```

but finger seems to still be broken:

```
root@beri1:/home/ctsrd # ./finger 
No one logged on.
root@beri1:/home/ctsrd # ./finger root
finger: root: no such user
```

brooksdavis, 2015-12-04:
Bah.  That `id` was the system `id` so both are still broken.

rwatson, 2015-12-07:
I can confirm that my ```getpwnam``` and ```getpwuid``` tests continue to fail. This is also a blocker on ```ssh``` working (as it queries account information for the current UID).

rwatson, 2015-12-07:
With compiler updates, it looks like this bug has evolved a bit such that we now get back an untagged capability with offset 1and length 0 that, when used with CToPtr, turns into NULL instead of the value tested for in the switch statement in NSS. More details to follow. In essence, though, we are still not quite getting static initialisation or casts to/from enums right in CheriABI.

davidchisnall, 2015-12-07:
I wonder if we should use `cgetoffset` for `int` / `enum` casts from pointers in the pure-capability ABI.  There is no expectation that we be able to round-trip a pointer via anything other than `intptr_t`, so it's probably a bit more consistent.

rwatson, 2015-12-07:
Sounds plausible to me, and would be an immediate fix for this problem. Not sure if it might introduce others?

davidchisnall, 2015-12-09:
This may be fixed now that CTSRD-CHERI/clang#74 is fixed.

brooksdavis, 2015-12-09:
The current bug is fixed, but CTSRD-CHERI/llvm#142 and some bdb bugs are still causing problems with nsswitch.

brooksdavis, 2015-12-16:
I've fixed one issues which was breaking `getpwnam()` in e3a69027cc5a384431156d61c90d4304387a9b9d.  I've also worked around a compiler bug that was breaking `getgrent()` and friends in eab55f11dbe238c6b4534f65a5b31dd4546ae2e4.

brooksdavis, 2015-12-16:
id and finger now work

closed 2015-12-16 by brooksdavis