CTSRD-CHERI/cheribsd #53: units(1) crashes with a length violation on CheriABI - possibly varargs related?

rwatson, 2015-10-30:
When compiling ```units(1)``` for CheriABI:

```
--- a/usr.bin/units/Makefile
+++ b/usr.bin/units/Makefile
@@ -2,6 +2,9 @@
 
 .include <src.opts.mk>
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  units
 FILES= units.lib
 FILESDIR=      ${SHAREDIR}/misc
```

I encounter a tag violation on running it:

```
> ./units
Signal 34
```

With the following console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c02 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007ffffee000 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:000081d5 b:0000007ffffee140 l:0000000000000008 o:0 t:0
$c03: v:1 s:0 p:000081d5 b:0000007ffffee140 l:0000000000000008 o:20 t:0
$c04: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120244c60 t:0
$c06: v:1 s:0 p:7fff81ff b:00000001601fa000 l:0000000000000400 o:360 t:0
$c07: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7ffffed980 t:0
$c08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7ffffed918 t:0
$c09: v:1 s:0 p:7fff81ff b:0000007fffffed04 l:0000000000000014 o:13 t:0
$c10: v:1 s:0 p:7fff81ff b:0000000160214000 l:0000000000000400 o:2e t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120018840 t:0
$c13: v:1 s:0 p:000081d5 b:0000007ffffee140 l:0000000000000008 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201b7e1f t:0
$c15: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201b7ea7 t:0
$c16: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201abd1c t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120003050 t:0
$c18: v:1 s:0 p:7fff81ff b:00000001601f1000 l:0000000000001000 o:20 t:0
$c19: v:1 s:0 p:7fff81fd b:0000007ffffee3bc l:0000000000000004 o:0 t:0
$c20: v:1 s:0 p:7fff81ff b:00000001601ef000 l:0000000000000200 o:20 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201a4330 t:0
$c23: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120002a00 t:0
$c24: v:1 s:0 p:7fff81fd b:0000007ffffedffc l:0000000000000004 o:0 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120018b08 t:0
Mar 16 20:56:04 beri1 kernel: USER_CHERI_EXCEPTION: pid 2840 tid 100068 (units), uid 999: CP2 fault (type 0x32)
Mar 16 20:56:04 beri1 kernel: Trapframe Register Dump:
Mar 16 20:56:04 beri1 kernel: zero: 0   at: 0x120010000 v0: 0xffffffffffffffe0  v1: 0x4
Mar 16 20:56:04 beri1 kernel: a0: 0x120018ae0   a1: 0x1201a0000 a2: 0x4 a3: 0
Mar 16 20:56:04 beri1 kernel: a4: 0     a5: 0x8 a6: 0x7ffffebed8        a6: 0x7ffffebed0
Mar 16 20:56:04 beri1 kernel: t0: 0x10  t1: 0xc t2: 0x7ffffebac0        t3: 0x7ffffeba40
Mar 16 20:56:04 beri1 kernel: t8: 0xc   t9: 0x120018840 s0: 0   s1: 0x8
Mar 16 20:56:04 beri1 kernel: s2: 0x1201f49d0   s3: 0x8 s4: 0x1 s5: 0x1201f0000
Mar 16 20:56:04 beri1 kernel: s6: 0x120000000   s7: 0xffffffffffffffff  k0: 0   k1: 0
Mar 16 20:56:04 beri1 kernel: gp: 0x7ffffee380  sp: 0x7ffffedac0        s8: 0x7ffffedac0        ra: 0x7ffffee380
Mar 16 20:56:04 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0        badvaddr: 0x120003000
Mar 16 20:56:04 beri1 kernel: cause: 0x148      pc: 0x120018b08
```

LLVM has helpfully provided an uninformative block name ```$BB99_13```, for which there isn't an obvious jump source:
[... 135 lines cut ...]

```
                el = el_init(argv[0], stdin, stdout, stderr);
                el_set(el, EL_PROMPT, &prompt);
                el_set(el, EL_EDITOR, "emacs");
                el_set(el, EL_SIGNAL, 1);
                el_set(el, EL_HIST, history, inhistory);
                el_source(el, NULL);
                history(inhistory, &ev, H_SETSIZE, 800);
```

It seems a good bet we're looking at an issue with ```varargs``` and hence possibly a toolchain issue. I can't easily tell from the available debugging information to hand which it is, but it's a good bet that it's the first, which passes a pointer to a function into ```libeditline```:
```
static const char * prompt(EditLine *e __unused) {
        return promptstr;
}
```

A close second best guess would be the passing in of ```inhistory```:

```
        History *inhistory;
```

Perhaps @davidchisnall  could opine on this use of ```varargs``` by ```units(1)```?

davidchisnall, 2015-10-30:
I'll take a closer look once I've finished the merge (just a few clang bugs to go and we're back to passing the LLVM / Clang test suites...).  At first glance, this looks like it really is a memory safety failure in variadic function calling: `$c13` contains the capability to the variadic arguments and it appears as if a single 8-byte value is passed, then it attempts to read it as a capability.

rwatson, 2015-10-30:
```units(1)``` code in question here:

    https://github.com/CTSRD-CHERI/cheribsd/blob/master/usr.bin/units/units.c#L820

```libeditline(3)``` code here:

    https://github.com/CTSRD-CHERI/cheribsd/blob/master/lib/libedit/eln.c#L113

brooksdavis, 2015-10-30:
I've spotted one libedit bug.  `el_set` contains the following, but the manpage says they all take `int`, not `int *`.

```c
        case EL_SIGNAL:         /* int * */
        case EL_EDITMODE:
        case EL_UNBUFFERED:
        case EL_PREP_TERM:
                ret = el_wget(el, op, va_arg(ap, int *));
                break;
```

brooksdavis, 2015-10-30:
The above wasn't detected by the compiler because `el_wget` is also a varargs function so there isn't any type checking.  It works be accident on non-cheri systems because the full register is copied over and then `el_wget` (declared `FUN(el,get)`) calls `va_arg(ap, int)` and pulls the right value out.  Fixing this doesn't fix unit as it then blows up in ncurses, but that deserves it's own bug report.

brooksdavis, 2015-11-05:
After discussion with NetBSD developers about the libedit change I've concluded it was wrong.  Backing it out I'm no longer seeing the crash so I suspect it may have been fixed by a compiler change or perhaps by 4f2bb124dcb0e4376fb79f2d2693e66f52ee90a7.  I'm removing the CheriCCompat label for now because I don't think this was actually that sort of issue.

brooksdavis, 2015-11-05:
Arg, spoke too soon.  I was testing a regular mips binary.

brooksdavis, 2015-11-05:
The problem turns out to be that the compiler things the size of a pointer to a function `&func` is 8.  I've committed a test in CTSRD-CHERI/cheri-c-tests@bac6b9abfa41361306cc34af4cea63a2c32edaa1.

davidchisnall, 2015-11-06:
Thanks.  This kind of thing doesn't need to run (a `_Static_assert` will do), so makes more sense as part of the Clang test suite than as an executable test.

davidchisnall, 2015-11-06:
In the following test cases:

	typedef void(*ptr)(void);
	
	void bar(void);
	
	ptr x = bar;
	
	_Static_assert(sizeof(x) == sizeof(void*), "Size of function pointer is incorrect");
	_Static_assert(sizeof(ptr) == sizeof(void*), "Size of function pointer is incorrect");
	_Static_assert(sizeof(const ptr) == sizeof(void*), "Size of function pointer is incorrect");
	_Static_assert(sizeof(&bar) == sizeof(void*), "Size of function pointer is incorrect");
	_Static_assert(sizeof(bar) == sizeof(void*), "Size of function pointer is incorrect");

Only the last two fail.  This is because functions are always in address space 0, and are only cast to address space 200 during function to pointer decay, which does not happen here.

davidchisnall, 2015-11-08:
Ooops, this was not filed as a clang bug, so I've just closed the wrong bug in a commit.

Please file clang bugs in the clang bug tracker.

commit ee66bd3d (2015-10-30): The ops EL_SIGNAL, EL_EDITMODE, EL_UNBUFFERED, and EL_PREP_TERM all take
an int, not an int*.

Fixes #53 (new bug will be filed for the next crash).
closed 2015-11-08 by davidchisnall