CTSRD-CHERI/cheribsd #55: CheriABI stty(1) prints incorrect baud

rwatson, 2015-10-30:
In running ```stty(1)``` compiled with CheriABI:

```
--- a/bin/stty/Makefile
+++ b/bin/stty/Makefile
@@ -1,6 +1,9 @@
 #      @(#)Makefile    8.1 (Berkeley) 5/31/93
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  stty
 SRCS=  cchar.c gfmt.c key.c modes.c print.c stty.c util.c
 
```

I've discovered that it prints the baud information incorrectly:

```
> ./stty
#-64594 disc; speed 9600 baud;
lflags: echoe echoke echoctl
iflags: -icrnl inlcr -ixon -ixany -imaxbel -brkint inpck
oflags: -opost ocrnl tab3
cflags: cs8 -parenb
```

Compare with non-CheriABI ```stty(1)```:
```
> stty
speed 9600 baud;
lflags: echoe echoke echoctl
oflags: tab0
cflags: cs8 -parenb
```

It is not clear to me whether this is a CheriABI bug (e.g., in ```ioctl``` handling), an ```stty(1)``` bug that manifests only with CHERI-style compilation (e.g., relating to ```printf```), or a toolchain bug.

(@davidchisnall may wish to opine.)

brooksdavis, 2015-10-31:
I see similar results.  It's clear a bogus line discipline is being passed to the `print` function (hence the `#-64594`.  Examining the `main()`, the line discipline is being queried a couple lines above.

I suspect ioctl is suffering from the non-register based varargs ABI since it is defined as

```
int     ioctl(int, unsigned long, ...);
```

brooksdavis, 2015-10-31:
I'm not entirely sure what we want to do with `ioctl`.  The third argument is either a pointer or an `int` depending on the declaration of the ioctl: http://fxr.watson.org/fxr/source/kern/sys_generic.c#L697.

I guess we can construct a tag-less capability with an uint64_t offset in the `IOC_VOID` case inside a wrapper.

rwatson, 2015-11-01:
I guess we can argue to ourselves that ```ioctl``` uses ```long``` (or sometimes ```int``` -- another type problem for ```ioctl```) because it predates ```intptr_t```, and what they really meant was ```intptr_t```. I suppose a legitimate question for CheriABI is whether we actually change the API for ```ioctl``` to accept something that is explicitly an ```intptr_t``` type -- but thus far we have tried to change only the ABI, not the API, for widely used interfaces, and I think should continue to try to do that?

brooksdavis, 2015-11-02:
I don't think we can change `ioctl`'s API here unless we're willing to go all the way to defining it as as `int ioctl(int, unsigned long, void *);`.  The problem is that the type of the argument is "what ever the heck the compiler decides to pass" since it's a varargs function.  Given that `ioctl` takes many different types this seems a bit more pardonable than `open`s optional third argument (apparently typing `, 0` was too much).

closed 2015-11-06 by brooksdavis