CTSRD-CHERI/cheribsd #57: CheriABI /usr/bin/host experiences tag violation

rwatson, 2015-10-30:
Compiling /usr/bin/host for CheriABI:

```
--- a/usr.bin/host/Makefile
+++ b/usr.bin/host/Makefile
@@ -1,5 +1,8 @@
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 LDNSDIR=       ${.CURDIR}/../../contrib/ldns
 LDNSHOSTDIR=   ${.CURDIR}/../../contrib/ldns-host
 
```

I experience a tag violation when I run ```host .```:
```
> ./host .
Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x02 RegNum: $c03 (tag violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffc870 l:0000000000000008 o:0 t:0
$c03: v:0 s:0 p:7fff81ff b:0000000160377060 l:0000000000000080 o:20 t:0
$c04: v:1 s:0 p:7fff81ff b:000000016037a100 l:0000000000000100 o:a0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffcd90 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffbd48 l:0000000000000008 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffbd40 l:0000000000000008 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffbcb0 l:0000000000000008 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffbca8 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffbd28 l:0000000000000008 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12002c508 t:0
$c13: v:1 s:0 p:7fff81ff b:000000016037a100 l:0000000000000100 o:7c t:0
$c14: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120363d50 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffbd08 l:0000000000000008 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffbd00 l:0000000000000008 o:0 t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120029de4 t:0
$c18: v:1 s:0 p:7fff81fd b:0000007fffffce78 l:0000000000000008 o:0 t:0
$c19: v:1 s:0 p:7fff81ff b:000000016037a100 l:0000000000000100 o:20 t:0
$c20: v:0 s:0 p:7fff81ff b:0000000160377060 l:0000000000000080 o:20 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffc8e0 l:0000000000000020 o:0 t:0
$c22: v:1 s:0 p:7fff81ff b:0000000160370fe0 l:0000000000000400 o:20 t:0
$c23: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120029d08 t:0
$c24: v:1 s:0 p:7fff81fd b:0000007fffffcdd8 l:0000000000000001 o:0 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12002c534 t:0
Mar 17 06:06:31 beri1 kernel: USER_CHERI_EXCEPTION: pid 4359 tid 100067 (host), uid 999: CP2 fault (type 0x32)
Mar 17 06:06:31 beri1 kernel: Trapframe Register Dump:
Mar 17 06:06:31 beri1 kernel: zero: 0   at: 0x1 v0: 0x1203ac508 v1: 0
Mar 17 06:06:31 beri1 kernel: a0: 0     a1: 0x80        a2: 0x4 a3: 0x7fffffc110
Mar 17 06:06:31 beri1 kernel: a4: 0x3800e000    a5: 0x3751c00   a6: 0xffffffffe8f2e3c0  a6: 0xe0e000
Mar 17 06:06:31 beri1 kernel: t0: 0x90ac62187d869c42    t1: 0   t2: 0xee0000    t3: 0
Mar 17 06:06:31 beri1 kernel: t8: 0x70e0000     t9: 0x12002c508 s0: 0x35        s1: 0x20
Mar 17 06:06:31 beri1 kernel: s2: 0x1   s3: 0x1203b1990 s4: 0   s5: 0x2
Mar 17 06:06:31 beri1 kernel: s6: 0x1c  s7: 0   k0: 0   k1: 0
Mar 17 06:06:31 beri1 kernel: gp: 0x1203b1990   sp: 0x7fffffc820        s8: 0x7fffffc820        ra: 0
Mar 17 06:06:31 beri1 kernel: sr: 0x4000fcf3    mullo: 0xf3a9   mulhi: 0        badvaddr: 0x1200a82f0
Mar 17 06:06:31 beri1 kernel: cause: 0x148      pc: 0x12002c534
```

$c03 is indeed untagged, although appears to have been a valid pointer previously:
```
$c03: v:0 s:0 p:7fff81ff b:0000000160377060 l:0000000000000080 o:20 t:0
```

Code as follows:
```
000000012002c508 <ldns_rdf_get_type>:
   12002c508:   67bdffa0        daddiu  sp,sp,-96
   12002c50c:   ebcbeac3        csd     s8,sp,88(c11)
   12002c510:   fa2be820        csc     c17,sp,32(c11)
   12002c514:   03a0f02d        move    s8,sp
   12002c518:   49b96002        cgetoffset      t9,c12
   12002c51c:   3c010038        lui     at,0x38
   12002c520:   0039102d        daddu   v0,at,t9
   12002c524:   48810007        cfromptr        c1,c0,zero
   12002c528:   49c11841        cne     at,c3,c1
   12002c52c:   10200007        beqz    at,12002c54c <ldns_rdf_get_type+0x44>
   12002c530:   00000000        nop
   12002c534:   c8430046        clw     v0,zero,8(c3)                                    <--- boom here
   12002c538:   03c0e82d        move    sp,s8
   12002c53c:   da2be820        clc     c17,sp,32(c11)
   12002c540:   cbcbeac3        cld     s8,sp,88(c11)
   12002c544:   49008800        cjr     c17
   12002c548:   67bd0060        daddiu  sp,sp,96
   12002c54c:   64415488        daddiu  at,v0,21640
   12002c550:   6422e8d8        daddiu  v0,at,-5928
   12002c554:   48810087        cfromptr        c1,c0,v0
   12002c558:   c8410003        cld     v0,zero,0(c1)
   12002c55c:   dc238038        ld      v1,-32712(at)
   12002c560:   48000807        cgetpcc c1
   12002c564:   49ac0881        csetoffset      c12,c1,v0
   12002c568:   dc228038        ld      v0,-32712(at)
   12002c56c:   6463b4c9        daddiu  v1,v1,-19255
   12002c570:   dc218038        ld      at,-32712(at)
   12002c574:   488300c7        cfromptr        c3,c0,v1
   12002c578:   6442b47e        daddiu  v0,v0,-19330
   12002c57c:   48840087        cfromptr        c4,c0,v0
   12002c580:   6421b4be        daddiu  at,at,-19266
   12002c584:   48850047        cfromptr        c5,c0,at
   12002c588:   48f16000        cjalr   c17,c12
   12002c58c:   64040021        daddiu  a0,zero,33
```

Possibly we have an errant non-```memcpy``` version of ```memcpy``` somewhere? (or a memory overwrite bug involving wherever $c3 was loaded from?).

brooksdavis, 2015-10-30:
Hmm.  It works for me:

```
root@beri1:/home/ctsrd # ./host .
root@beri1:/home/ctsrd # ./host vica.cl.cam.ac.uk
vica.cl.cam.ac.uk has address 128.232.64.163
vica.cl.cam.ac.uk has IPv6 address 2001:630:212:238:21e:67ff:fe56:305c
root@beri1:/home/ctsrd # file host
host: ELF 64-bit MSB executable, *unknown arch 0xc256* version 1 (FreeBSD), statically linked, for FreeBSD 11.0 (1100077), not stripped
```

rwatson, 2015-10-31:
I will rebuild from scratch and retry.

brooksdavis, 2015-11-13:
A freshly built `host` is again working for me so I'm going to assume this is fixed.  It's certainly possible that CL DNS servers are producing results that hit another code path, but in that case I won't be able to debug this easily

closed 2015-11-13 by brooksdavis

rwatson, 2015-11-14:
Confirmed that it is also fixed for me -- we will have to keep an eye on some of these prior reports when we switch back to cheri-linker though.