CTSRD-CHERI/cheribsd #61: Many programs experience alignment exceptions

rwatson, 2015-11-06:
A fair number of programs I've tested experience alignment exceptions when storing capabilities in CheriABI. For example, ```fetch(1)```, compiled for CheriABI, experiences an alignment problem in the ```getservent(3)``` code:
```
--- a/usr.bin/fetch/Makefile
+++ b/usr.bin/fetch/Makefile
@@ -1,5 +1,8 @@
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=          fetch
 CSTD?=         c99
 LIBADD=                fetch
```

When run casually:
```
> ./fetch http://www.watson.org/
Bus error
```

Console output:
```
cpuid = 0
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c03: v:1 s:0 p:7fff81ff b:00000001603e9000 l:0000000000000800 o:2d t:0
$c04: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203bf505 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffa77c l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffa780 l:0000000000000020 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffa778 l:0000000000000004 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203bf507 t:0
$c09: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120374158 t:0
$c10: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203c6468 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:1202c1630 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffa9e0 l:0000000000000008 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffaaa0 l:0000000000000020 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203c0703 t:0
$c16: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12039cd58 t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120284f84 t:0
$c18: v:1 s:0 p:7fff81fd b:0000007fffffa8e0 l:0000000000000020 o:0 t:0
$c19: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1603e93f8 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1603e9058 t:0
$c22: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1603e9058 t:0                  <-- Not 32-byte aligned; 24-byte aligned
$c23: v:1 s:0 p:7fff81fd b:0000007fffffbab8 l:0000000000000004 o:0 t:0
$c24: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203bf505 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120285080 t:0
Mar 23 04:13:37 beri1 kernel: ALIGNMENT_FIX_ERR: pid 16610 tid 100049 (fetch), uid 999: pc 0x120285080 got a write fault (type 0x5) at 0x1603e9058
Mar 23 04:13:37 beri1 kernel: Trapframe Register Dump:
Mar 23 04:13:37 beri1 kernel: zero: 0   at: 0   v0: 0   v1: 0x1
Mar 23 04:13:37 beri1 kernel: a0: 0x9   a1: 0xa a2: 0x7 a3: 0xccccccccccccccc
Mar 23 04:13:37 beri1 kernel: a4: 0x1   a5: 0   a6: 0x1 a6: 0x88
Mar 23 04:13:37 beri1 kernel: t0: 0     t1: 0x17        t2: 0x7fffffc2e0        t3: 0x7fffffc260
Mar 23 04:13:37 beri1 kernel: t8: 0x15  t9: 0x1202d6840 s0: 0xffffffffffffffff  s1: 0x9
Mar 23 04:13:37 beri1 kernel: s2: 0x1203c0000   s3: 0x1 s4: 0x20        s5: 0x20
Mar 23 04:13:37 beri1 kernel: s6: 0xffffffffffffb600    s7: 0xffd7      k0: 0   k1: 0
Mar 23 04:13:37 beri1 kernel: gp: 0x120420e50   sp: 0x7fffffa7c0        s8: 0x7fffffa7c0        ra: 0x1204b0000
Mar 23 04:13:37 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0        badvaddr: 0x1603e9058
Mar 23 04:13:37 beri1 kernel: cause: 0x114      pc: 0x120285080
Mar 23 04:13:37 beri1 kernel: Page table info for bad address 0x1603e9058: pde = 0x98000000050bc000, pte = 0x800000001253de
```

Disassembly of the pertinent function:
```
0000000120284d20 <servent_unpack>:
   120284d20:   67bdfd80        daddiu  sp,sp,-640
[... 229 lines cut ...]
   1202850b8:   67a10240        daddiu  at,sp,576
   1202850bc:   ca0b0803        cld     s0,at,0(c11)
   1202850c0:   67a10248        daddiu  at,sp,584
   1202850c4:   ca2b0803        cld     s1,at,0(c11)
   1202850c8:   67a10250        daddiu  at,sp,592
   1202850cc:   ca4b0803        cld     s2,at,0(c11)
   1202850d0:   67a10258        daddiu  at,sp,600
   1202850d4:   ca6b0803        cld     s3,at,0(c11)
   1202850d8:   67a10260        daddiu  at,sp,608
   1202850dc:   ca8b0803        cld     s4,at,0(c11)
   1202850e0:   67a10268        daddiu  at,sp,616
   1202850e4:   caab0803        cld     s5,at,0(c11)
   1202850e8:   67a10270        daddiu  at,sp,624
   1202850ec:   cb8b0803        cld     gp,at,0(c11)
   1202850f0:   67a10278        daddiu  at,sp,632
   1202850f4:   cbcb0803        cld     s8,at,0(c11)
   1202850f8:   49008800        cjr     c17
   1202850fc:   67bd0280        daddiu  sp,sp,640
   120285100:   fa950000        csc     c20,zero,0(c21)
   120285104:   24010022        li      at,34
   120285108:   080a1422        j       120285088 <servent_unpack+0x368>
   12028510c:   e8370002        csw     at,zero,0(c23)
```

$c22 is not 32-byte aligned, so the capability store fails.

rwatson, 2015-11-06:
This may be because the MIPS ```_align.h``` defines the minimum alignment for the most aligned thing to be 8-bit alignment rather than 32-byte alignment, which for CheriABI will inappropriately align capabilities. ```_ALIGN()``` is used by the ```getservent(3)``` code to align aliases within string buffers; it is also used in other ```libc``` databases. I will try this patch:

```
--- a/sys/mips/include/_align.h
+++ b/sys/mips/include/_align.h
@@ -47,7 +47,7 @@
  * data types (int, long, ...).          The result is u_long and must be cast to
  * any desired pointer type.
  */
-#define        _ALIGNBYTES     7
+#define        _ALIGNBYTES     (sizeof(void *) - 1)
 #define        _ALIGN(p)       (((u_long)(p) + _ALIGNBYTES) &~ _ALIGNBYTES)
 
 #endif /* !_MIPS_INCLUDE__ALIGN_H_ */
```

And see if it helps.

rwatson, 2015-11-06:
This does seem to help.

rwatson, 2015-11-06:
Fix committed as 12494209b2099a49110f1171227cee14a2e0548e, which appears to resolve the issue, allowing ```fetch(1)``` to successfully download a file via HTTP.

closed 2015-11-06 by rwatson

rwatson, 2015-11-07:
Annotated as CheriCCompat as _ALIGN() and friends are not very CHERI-friendly.

davidchisnall, 2015-11-07:
Related: CTSRD-CHERI/llvm#85