CTSRD-CHERI/cheribsd #62: fetch(1) experiences a capability length violation for an HTTPS download

rwatson, 2015-11-06:
When compiling ```fetch(1)``` for CheriABI:
```
--- a/usr.bin/fetch/Makefile
+++ b/usr.bin/fetch/Makefile
@@ -1,5 +1,8 @@
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=          fetch
 CSTD?=         c99
 LIBADD=                fetch
```

And experience a length violation exception when trying to download using HTTPS:
```
root@beri1:/tmp # ./fetch https://www.google.com/
Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c03 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffb478 l:0000000000000004 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffb1bc l:0000000000000004 o:0 t:0
$c03: v:1 s:0 p:7fff81ff b:0000000160467000 l:0000000000001000 o:20 t:0
$c04: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203ba0fe t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffb468 l:0000000000000008 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:1 s:0 p:7fff81ff b:0000000160466c00 l:0000000000000080 o:20 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffb2c0 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203bb0a4 t:0
$c10: v:1 s:0 p:7fff81ff b:00000001603edc00 l:0000000000000100 o:88 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffb860 l:0000000000000008 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:1 s:0 p:7fff81ff b:000000016045c700 l:0000000000000100 o:94 t:0
$c16: v:1 s:0 p:7fff81ff b:000000016045c700 l:0000000000000100 o:68 t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:1201ea674 t:0
$c18: v:1 s:0 p:7fff81ff b:000000016045d400 l:0000000000000200 o:20 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffb4e0 l:0000000000000020 o:0 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffb600 l:0000000000000020 o:0 t:0
$c22: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffb5e0 t:0
$c23: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:1201507b0 t:0
$c24: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:1201ea298 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:1201ea6ac t:0
Mar 23 05:38:29 beri1 kernel: USER_CHERI_EXCEPTION: pid 706 tid 100062 (fetch), uid 0: CP2 fault (type 0x32)
Mar 23 05:38:29 beri1 kernel: Trapframe Register Dump:
Mar 23 05:38:29 beri1 kernel: zero: 0   at: 0xfe0       v0: 0x40        v1: 0x7f
Mar 23 05:38:29 beri1 kernel: a0: 0x7f  a1: 0x1 a2: 0xfe0       a3: 0xffffffffffffffe0
Mar 23 05:38:29 beri1 kernel: a4: 0x20  a5: 0x40        a6: 0x200007f   a6: 0x7fffff8ba8
Mar 23 05:38:29 beri1 kernel: t0: 0     t1: 0x10        t2: 0   t3: 0x18e9cb54
Mar 23 05:38:29 beri1 kernel: t8: 0x2   t9: 0x12038eb78 s0: 0x4 s1: 0x80
Mar 23 05:38:29 beri1 kernel: s2: 0x40  s3: 0x120420e50 s4: 0   s5: 0x8
Mar 23 05:38:29 beri1 kernel: s6: 0x14  s7: 0x20        k0: 0   k1: 0
Mar 23 05:38:29 beri1 kernel: gp: 0x320 sp: 0x7fffffb440        s8: 0x7fffffb440        ra: 0x1
Mar 23 05:38:29 beri1 kernel: sr: 0x4000fcf3    mullo: 0x80001  mulhi: 0x3f     badvaddr: 0x16046a000
Mar 23 05:38:29 beri1 kernel: cause: 0x148      pc: 0x1201ea6ac
```

brooksdavis, 2015-11-24:
This is a malloc bug.  Allocating 4096 bytes results in 4064 bytes of usable space.

brooksdavis, 2015-12-01:
Hmm, When I try to allocate 4096 bytes directly I get the expected thing (0x2000-0x20 bytes) so we may be seeing a different bug.

commit bc27e518 (2015-12-03): Compute the allocated size correctly.

Fixes #62.
closed 2015-12-03 by brooksdavis