CTSRD-CHERI/cheribsd #64: fetch(1) experiences length violation in setproctitle(1)

rwatson, 2015-11-13:
When trying a CheriABI ```fetch(1)```, I experience a new length violation:
```
--- a/usr.bin/fetch/Makefile
+++ b/usr.bin/fetch/Makefile
@@ -1,5 +1,8 @@
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=          fetch
 CSTD?=         c99
 LIBADD=                fetch
```

Output:
```
> ./fetch http://www.watson.org/
fetch: http://www.watson.org/: size of remote file is not known
fetch.out                                     Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c21 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffc8b0 l:0000000000000008 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12041c860 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffca18 l:0000000000000068 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffcc90 l:0000000000000008 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffcc98 l:0000000000000008 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffc03c l:0000000000000002 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203bf8e7 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffc49c l:0000000000000004 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffff8470 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1202f1570 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffcb80 l:0000000000000020 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffaaa0 l:0000000000000020 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1203c0ae3 t:0
$c16: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12039d138 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120266aa0 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:1 s:0 p:7fff81fd b:0000007fffffcc98 l:0000000000000008 o:0 t:0
$c20: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120313700 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffcc90 l:0000000000000008 o:0 t:0
$c22: v:1 s:0 p:7fff81ff b:000000016045e000 l:0000000000001000 o:20 t:0
$c23: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1202d5d20 t:0
$c24: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1202668c8 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120266abc t:0
Mar 27 10:17:25 beri1 kernel: USER_CHERI_EXCEPTION: pid 1038 tid 100066 (fetch), uid 999: CP2 fault (type 0x32)
Mar 27 10:17:25 beri1 kernel: Trapframe Register Dump:
Mar 27 10:17:25 beri1 kernel: zero: 0   at: 0xffffffffffffffff  v0: 0   v1: 0
Mar 27 10:17:25 beri1 kernel: a0: 0x2   a1: 0   a2: 0x5d        a3: 0
Mar 27 10:17:25 beri1 kernel: a4: 0x7fffffc358  a5: 0x7fffffc350        a6: 0x7fffffc318        a6: 0x7fffffc310
Mar 27 10:17:25 beri1 kernel: t0: 0x19  t1: 0x1a        t2: 0x7fffffbf00        t3: 0x7fffffbe80
Mar 27 10:17:25 beri1 kernel: t8: 0x1   t9: 0x1202d5d20 s0: 0x120420e70 s1: 0x1204b0000
Mar 27 10:17:25 beri1 kernel: s2: 0x1   s3: 0x8 s4: 0x1204b0000 s5: 0x4
Mar 27 10:17:25 beri1 kernel: s6: 0x8   s7: 0x120420000 k0: 0   k1: 0
Mar 27 10:17:25 beri1 kernel: gp: 0x120420e70   sp: 0x7fffffcb80        s8: 0x7fffffcb80        ra: 0x7fffffd4a0
Mar 27 10:17:25 beri1 kernel: sr: 0x4000fcf3    mullo: 0x298    mulhi: 0        badvaddr: 0x1203bdeb9
Mar 27 10:17:25 beri1 kernel: cause: 0x148      pc: 0x120266abc
Mar 27 10:18:07 beri1 sshd[1039]: error: Could not load host key: /etc/ssh/ssh_host_dsa_key
```

The crash appears to happen in ```setproctitle(3)```:
```
[... 391 lines cut ...]
   120266cc8:   ca2b0803        cld     s1,at,0(c11)
   120266ccc:   67a10298        daddiu  at,sp,664
   120266cd0:   ca4b0803        cld     s2,at,0(c11)
   120266cd4:   67a102a0        daddiu  at,sp,672
   120266cd8:   ca6b0803        cld     s3,at,0(c11)
   120266cdc:   67a102a8        daddiu  at,sp,680
   120266ce0:   ca8b0803        cld     s4,at,0(c11)
   120266ce4:   67a102b0        daddiu  at,sp,688
   120266ce8:   caab0803        cld     s5,at,0(c11)
   120266cec:   67a102b8        daddiu  at,sp,696
   120266cf0:   cacb0803        cld     s6,at,0(c11)
   120266cf4:   67a102c0        daddiu  at,sp,704
   120266cf8:   caeb0803        cld     s7,at,0(c11)
   120266cfc:   67a102c8        daddiu  at,sp,712
   120266d00:   cb8b0803        cld     gp,at,0(c11)
   120266d04:   67a102d0        daddiu  at,sp,720
   120266d08:   cbcb0803        cld     s8,at,0(c11)
   120266d0c:   67a102d8        daddiu  at,sp,728
   120266d10:   cbeb0803        cld     ra,at,0(c11)
   120266d14:   49008800        cjr     c17
   120266d18:   67bd02e0        daddiu  sp,sp,736
   120266d1c:   00000000        nop
```

Looks a bit like an attempt to read a long and write back a pointer to the same address?

brooksdavis, 2015-11-13:
A bit more analysis after build with -g, it looks like a compiler bug.  It looks like `sysctl("kern.ps_strings", ...)` is succeeding and then the code tries to load a capability through `&ul_ps_strings` rather than converting `ul_ps_strings` into a pointer using `cfromptr`.

I've submitted CTSRD-CHERI/llvm#135 with a reduced test case for @davidchisnall 

```
CHERI cause: ExcCode: 0x01 RegNum: $c21 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffd7d0 l:0000000000000008 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120070038 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffd938 l:0000000000000068 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffdbb0 l:0000000000000008 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffdbb8 l:0000000000000008 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffcf5c l:0000000000000002 o:0 t:0
$c08: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12004fbe8 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffd3bc l:0000000000000004 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120053e78 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120013470 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffdaa0 l:0000000000000020 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120053e93 t:0
$c15: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120053f1e t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120003380 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:1 s:0 p:7fff81fd b:0000007fffffdbb8 l:0000000000000008 o:0 t:0
$c20: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12002ec10 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffdbb0 l:0000000000000008 o:0 t:0
$c22: v:1 s:0 p:7fff81ff b:000000016006bfe0 l:0000000000001000 o:20 t:0
$c23: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120005220 t:0
$c24: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:120003124 t:0
$c26: v:1 s:0 p:7fff81ff b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff81d7 b:0000000000000000 l:0000010000000000 o:12000339c t:0
Oct 24 05:39:10 beri1 kernel: USER_CHERI_EXCEPTION: pid 2967 tid 100042 (helloworld), uid 0: CP2 fault (type 0x32)
Oct 24 05:39:10 beri1 kernel: Trapframe Register Dump:
Oct 24 05:39:10 beri1 kernel: zero: 0   at: 0xffffffffffffffff  v0: 0   v1: 0
Oct 24 05:39:10 beri1 kernel: a0: 0x2   a1: 0   a2: 0x65        a3: 0
Oct 24 05:39:10 beri1 kernel: a4: 0x7fffffd278  a5: 0x7fffffd270        a6: 0x7fffffd238  a6: 0x7fffffd230
Oct 24 05:39:10 beri1 kernel: t0: 0xc   t1: 0x11        t2: 0x7fffffce20        t3: 0x7fffffcda0
Oct 24 05:39:10 beri1 kernel: t8: 0x9   t9: 0x120005220 s0: 0x120076d70 s1: 0x120070000
Oct 24 05:39:10 beri1 kernel: s2: 0x1   s3: 0x8 s4: 0x120070000 s5: 0x4
Oct 24 05:39:10 beri1 kernel: s6: 0x8   s7: 0   k0: 0   k1: 0
Oct 24 05:39:10 beri1 kernel: gp: 0     sp: 0x7fffffdaa0        s8: 0x7fffffdaa0 ra: 0
Oct 24 05:39:10 beri1 kernel: sr: 0x4000fcf3    mullo: 0x298    mulhi: 0        badvaddr: 0x12006f8d8
Oct 24 05:39:10 beri1 kernel: cause: 0x148      pc: 0x12000339c
```

```
       if (ps_strings == NULL) {
   120003324:   de148038        ld      s4,-32712(s0)
   120003328:   668100c0        daddiu  at,s4,192
   12000332c:   48810047        cfromptr        c1,c0,at
   120003330:   d8210000        clc     c1,zero,0(c1)
   120003334:   49c10c81        cne     at,c1,c18
   120003338:   1420001c        bnez    at,1200033ac <setproctitle+0x41c>
   12000333c:   00000000        nop
   120003340:   67c10110        daddiu  at,s8,272
                len = sizeof(ul_ps_strings);
                if (sysctlbyname("kern.ps_strings", &ul_ps_strings, &len, NULL,
   120003344:   66028130        daddiu  v0,s0,-32464
   120003348:   48810087        cfromptr        c1,c0,v0
   12000334c:   c8410003        cld     v0,zero,0(c1)
   120003350:   49a15841        csetoffset      c1,c11,at
   120003354:   48350cc0        csetbounds      c21,c1,s3
        oid[2] = KERN_PROC_ARGS;
        oid[3] = getpid();
        sysctl(oid, 4, 0, 0, kbuf, strlen(kbuf) + 1);

        if (ps_strings == NULL) {
                len = sizeof(ul_ps_strings);
[... 8 lines cut ...]
   120003374:   49a4a800        cmove   c4,c21
   120003378:   48f16000        cjalr   c17,c12
   12000337c:   49a59800        cmove   c5,c19
                    0) == -1)
   120003380:   2401ffff        li      at,-1
        oid[3] = getpid();
        sysctl(oid, 4, 0, 0, kbuf, strlen(kbuf) + 1);

        if (ps_strings == NULL) {
                len = sizeof(ul_ps_strings);
                if (sysctlbyname("kern.ps_strings", &ul_ps_strings, &len, NULL,
   120003384:   14410005        bne     v0,at,12000339c <setproctitle+0x40c>
   120003388:   00000000        nop
                    0) == -1)
                        ul_ps_strings = PS_STRINGS;
   12000338c:   64010001        daddiu  at,zero,1
   120003390:   0001097c        dsll32  at,at,0x5
   120003394:   6421eec0        daddiu  at,at,-4416
   120003398:   e8350003        csd     at,zero,0(c21)
                ps_strings = (struct ps_strings *)ul_ps_strings;
   12000339c:   d8350000        clc     c1,zero,0(c21)
   1200033a0:   668100c0        daddiu  at,s4,192
   1200033a4:   48820047        cfromptr        c2,c0,at
   1200033a8:   f8220000        csc     c1,zero,0(c2)
```

rwatson, 2015-11-14:
Slightly more code context from ```lib/libc/setproctitle.c```:
```
void
setproctitle(const char *fmt, ...)
{
    static struct ps_strings *ps_strings;
    ...
    unsigned long ul_ps_strings;
    ...
```

davidchisnall, 2015-11-14:
So the issue is casting to a pointer from an `unsigned long`?  In the sandbox ABI, this is not necessarily expected to work, as there's no guarantee that `$c0` will contain anything (though currently it must at least contain globals).

brooksdavis, 2015-11-16:
The whole `ps_strings` ABI is broken and I'll need to replace this code, but the compiler's `-O` behavior is clearly wrong as well.  I don't mind if we break creation of pointers from `$c0` via a cast from int in the sandbox ABI, but we need to break it via a meaningful compiler error.

On the `ps_strings` front, I'm debating between two options.  One is to define the `__ps_strings` variable as arm and power do in `crt1.o`.  The other is to add a syscall along the lines of `int get_psstrings(void **pssp, size_t *pssl)`.

brooksdavis, 2015-11-24:
@davidchisnall has fixed the underlying long issue and I've switched the way we find `ps_strings` to the NetBSD way in 41be23fb6f64996ecb3f1aaad456deec5159b3ba so this now works.

closed 2015-11-24 by brooksdavis