CTSRD-CHERI/cheribsd #65: PRIiPTR and friends are not sensible

davidchisnall, 2015-11-20:
The `PRI?PTR` macros defined in CheriABI assume that the variadic argument is a long, but it is not (this causes some very odd values to be printed).

brooksdavis, 2015-11-20:
It looks like this is going to require introducing a new printf modifier for `intptr_t`.  Fortunately `'P'` appears to be available.

brooksdavis, 2015-11-20:
Resolved in 64c53c89880fe31243e00db2099a3b1184fa53ca and CTSRD-CHERI/clang@139ee891f4e1a059b7816f0679cb12580bf325ba.

brooksdavis, 2015-11-20:
We'll likely want to write up a defect report for WG14.  It's clearly impractical to support any CHERI-like architecture where pointers are integers without a modifier like `'P'`.

closed 2015-11-20 by brooksdavis

brooksdavis, 2015-11-21:
Hmm, the alternative standards compliment thing to do would be to not define the `*PTR` macros in CheriABI.  According to the posix page for `inttypes.h`:

> For each type that the implementation provides in <stdint.h>, the corresponding fprintf() and fwprintf() macros shall be defined and the corresponding fscanf() and fwscanf() macros shall be defined unless the implementation does not have a suitable modifier for the type.

I'd tend to argue that the lack of a "suitable modifier for the type" is a bug though.