CTSRD-CHERI/cheribsd #67: openssl branchless pointer selection pattern doesn't work

brooksdavis, 2015-12-03:
OpenSSL uses a branchless assignment pattern to select between pointer values the looks like:

```C
int             a_bool;
intptr_t        mask;
void            *ptr;
a_bool = <0 or 1>;
mask = 0 - a_bool;
ptr = (val_a & mask | val_b & ~mask);
```

I've modified the code in a few places to remove or work around this idiom not working in a0eac17db0649fe134bed192bff305da84c4125d.

It's probably possible to get some cases working (where `val_a` and `val_b` are the same object as with the current model of function pointers, but in the general case, too much magic is likely required.

brooksdavis, 2015-12-03:
We need an alternative idiom, ideally as a compiler builtin.

rmn30, 2015-12-03:
MIPS has the sel instruction which could be useful here.

River

rmn30, 2015-12-03:
Actually I mean movn but perhaps cheri dues not have it.

brooksdavis, 2015-12-03:
It looks like a pair of `cmovn` instructions could do what is required if we implemented them, had a way to generate them in C, and modified openssl to have a sensible macro for branchless conditional assignment.

davidchisnall, 2015-12-04:
The slightly depressing thing is that simply writing this as an `if` statement will most likely generate the branchless version on most compilers (if-conversion will turn it into a select).  This is probably needed for architectures that don't have a conditional move, where you need to fudge it with xor.  Note that we currently don't have a `ccmove` instruction, and probably should.

brooksdavis, 2015-12-16:
I've opened an issue to track the need for an analyzer as CTSRD-CHERI/clang#78.

closed 2015-12-16 by brooksdavis