CTSRD-CHERI/cheribsd #72: Attempting to build mandoc(1) fails with linker error

rwatson, 2015-12-11:
When I attempt to build ```mandoc(1)``` with CheriABI support:
```
diff --git a/usr.bin/mandoc/Makefile b/usr.bin/mandoc/Makefile
index faa46ed..8ec2eeb 100644
--- a/usr.bin/mandoc/Makefile
+++ b/usr.bin/mandoc/Makefile
@@ -5,6 +5,9 @@
 MDOCMLDIR=     ${.CURDIR}/../../contrib/mdocml
 .PATH: ${MDOCMLDIR}
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  mandoc
 FILES= example.style.css style.css
 FILESDIR=      ${SHAREDIR}/mdocml
```

I get the following error:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker -O2 -pipe  -DHAVE_CONFIG_H  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../lib/libohash/  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../contrib/sqlite3 -Wno-error  -std=gnu99  -Qunused-arguments -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include -c /home/rnw24/git/cheribsd/usr.bin/mandoc/../../contrib/mdocml/mansearch_const.c -o mansearch_const.o
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker -O2 -pipe  -DHAVE_CONFIG_H  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../lib/libohash/  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../contrib/sqlite3 -Wno-error  -std=gnu99  -Qunused-arguments -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include -c /home/rnw24/git/cheribsd/usr.bin/mandoc/../../contrib/mdocml/manpath.c -o manpath.o
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/mandoc/.depend, 1494: ignoring stale .depend for /usr/libcheri/libc.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/mandoc/.depend, 1494: ignoring stale .depend for /usr/libcheri/libprivatesqlite3.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/mandoc/.depend, 1494: ignoring stale .depend for /usr/libcheri/libpthread.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/mandoc/.depend, 1494: ignoring stale .depend for /usr/libcheri/libz.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/mandoc/.depend, 1494: ignoring stale .depend for /usr/libcheri/libmalloc_simple.a
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker -O2 -pipe  -DHAVE_CONFIG_H  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../lib/libohash/  -I/home/rnw24/git/cheribsd/usr.bin/mandoc/../../contrib/sqlite3 -Wno-error  -std=gnu99  -Qunused-arguments -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include  -Wl,--no-warn-mismatch -static -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/lib -o mandoc man.o man_hash.o man_macro.o man_validate.o att.o lib.o mdoc.o mdoc_argv.o mdoc_hash.o mdoc_macro.o mdoc_validate.o st.o eqn.o roff.o tbl.o tbl_data.o tbl_layout.o tbl_opts.o chars.o mandoc.o mandoc_aux.o msec.o preconv.o read.o eqn_html.o html.o man_html.o mdoc_html.o tbl_html.o mdoc_man.o eqn_term.o man_term.o mdoc_term.o term.o term_ascii.o term_ps.o tbl_term.o main.o out.o tree.o mandocdb.o mansearch.o mansearch_const.o manpath.o -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libohash -lohash -lprivatesqlite3   -lz   -lpthread -lc -lmalloc_simple -legacy
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libohash/libohash.a: could not read symbols: File format not recognized
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

Stop.
make[3]: stopped in /home/rnw24/git/cheribsd/usr.bin/mandoc
*** Error code 1

Stop.
make[2]: stopped in /home/rnw24/git/cheribsd
*** Error code 1

Stop.
make[1]: stopped in /home/rnw24/git/cheribsd
*** Error code 1

Stop.
make: stopped in /home/rnw24/git/cheribsd
```

rwatson, 2015-12-11:
@brooksdavis, @davidchisnall: any idea what's going on here?

rwatson, 2015-12-11:
Similar problem for ```strings(1)```:
```
===> usr.bin/strings (obj,depend,all,install)
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/strings created for /home/rnw24/git/cheribsd/usr.bin/strings
CC='/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker ' mkdep -f strings.dep_c   -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/libelftc -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/common -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include -std=gnu99   /home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/strings/strings.c
cat strings.dep_c > .depend
echo strings: /usr/libcheri/libc.a /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libelftc/libelftc.a /usr/libcheri/libelf.a /usr/libcheri/libc.a /usr/libcheri/libmalloc_simple.a /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/lib/libegacy.a >> .depend
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker -O2 -pipe  -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/libelftc -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/common -Wno-error  -std=gnu99  -Qunused-arguments -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include -c /home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/strings/strings.c -o strings.o
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/strings/.depend, 52: ignoring stale .depend for /usr/libcheri/libc.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/strings/.depend, 52: ignoring stale .depend for /usr/libcheri/libelf.a
make[3]: /home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/usr.bin/strings/.depend, 52: ignoring stale .depend for /usr/libcheri/libmalloc_simple.a
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float -mabi=sandbox -cheri-linker -O2 -pipe  -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/libelftc -I/home/rnw24/git/cheribsd/usr.bin/strings/../../contrib/elftoolchain/common -Wno-error  -std=gnu99  -Qunused-arguments -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/include  -Wl,--no-warn-mismatch -static -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/legacy/usr/lib -o strings strings.o -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libelftc -lelftc  -lelf   -lc -lmalloc_simple -legacy
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libelftc/libelftc.a: could not read symbols: File format not recognized
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1
```

davidchisnall, 2015-12-14:
What does file say that the file type is?

brooksdavis, 2015-12-17:
I think this is caused by the definition of `ROOTOBJDIR` in share/mk/bsd.libnames.mk being wrong in the WANT_CHERI=pure case causing us to pick up the mips64 `libelftc.a` or `libohash.a`.  To make this concrete, it is using:

```
-L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/home/rnw24/git/cheribsd/lib/libelftc
```
but should be using
```
-L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/worldcheri/home/rnw24/git/cheribsd/lib/libelftc
```

commit 29d4718a (2015-12-17): Set ROOTOBJDIR in bsd.cheri.mk when building for CHERI.

This allows programs that use internal libraries to be linked for CheriABI
and fixes #72.
closed 2015-12-17 by brooksdavis