CTSRD-CHERI/cheribsd #73: telnet(1) build fails with linker assertion failures

rwatson, 2015-12-12:
Attempting to build ```telnet(1)``` using CheriABI:
```
diff --git a/usr.bin/telnet/Makefile b/usr.bin/telnet/Makefile
index a24a966..cfa263b 100644
--- a/usr.bin/telnet/Makefile
+++ b/usr.bin/telnet/Makefile
@@ -5,6 +5,9 @@
 TELNETDIR=     ${.CURDIR}/../../contrib/telnet
 .PATH:         ${TELNETDIR}/telnet
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=          telnet
 
 SRCS=          commands.c main.c network.c ring.c sys_bsd.c \
```

Fails due to linker assertion failures and truncated relocations:

```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker -O -pipe -G0   -DKLUDGELINEMODE -DUSE_TERMIO -DENV_HACK -DOPIE  -I/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet -I/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/libtelnet/ -DINET6 -DIPSEC -DENCRYPTION -DAUTHENTICATION -DIPSEC -DKRB5 -DFORWARD -Dnet_write=telnet_net_write -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Qunused-arguments  -Wl,--no-warn-mismatch -static -o telnet commands.o main.o network.o ring.o sys_bsd.o telnet.o terminal.o utilities.o authenc.o -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/lib/libtelnet -ltelnet  -lncursesw  -lipsec  -lmp  -lcrypto  -lcrypto  -lipsec  -lpam  -lradius  -lcrypto  -ltacplus  -lopie  -lmd  -lmd  -lutil  -lkrb5  -lasn1  -lcom_err  -lcrypt  -lcrypto  -lhx509  -lroken  -lwind  -lheimbase -lprivateheimipcc   -lprivatessh  -lcrypto  -lcrypt -lprivateldns  -lcrypto  -lz  -lypclnt  -lkrb5  -lasn1  -lcom_err  -lcrypt  -lcrypto  -lhx509  -lroken  -lwind  -lheimbase -lprivateheimipcc    -lroken   -lpthread -lc -lmalloc_simple
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2767
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:7455
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
commands.o: In function `bye':
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:1401: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
commands.o: In function `tn':
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2458: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2427: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2438: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2442: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2462: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2278: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2496: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/commands.c:2432: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
network.o: In function `netflush':
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/network.c:159: relocation truncated to fit: R_MIPS_GOT_DISP against `NetClose'
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
terminal.o: In function `ttyflush':
/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/telnet/terminal.c:163: additional relocation overflows omitted from the output
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
[... cut ...]
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2484
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

Stop.
make[4]: stopped in /home/rnw24/git/cheribsd/usr.bin/telnet
```

rwatson, 2015-12-12:
@brooksdavis and @davidchisnall may wish to take a look at this.

davidchisnall, 2015-12-12:
Can you try adding `-mxgot` to the CFLAGS?  That will make it emit large relocations, which should prevent the 'truncated to fit' errors.

rwatson, 2015-12-12:
Now getting:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker -O -pipe -G0   -mxgot -DKLUDGELINEMODE -DUSE_TERMIO -DENV_HACK -DOPIE  -I/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet -I/home/rnw24/git/cheribsd/usr.bin/telnet/../../contrib/telnet/libtelnet/ -DINET6 -DIPSEC -DENCRYPTION -DAUTHENTICATION -DIPSEC -DKRB5 -DFORWARD -Dnet_write=telnet_net_write -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Qunused-arguments  -Wl,--no-warn-mismatch -static -o telnet commands.o main.o network.o ring.o sys_bsd.o telnet.o terminal.o utilities.o authenc.o -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/lib/libtelnet -ltelnet  -lncursesw  -lipsec  -lmp  -lcrypto  -lcrypto  -lipsec  -lpam  -lradius  -lcrypto  -ltacplus  -lopie  -lmd  -lmd  -lutil  -lkrb5  -lasn1  -lcom_err  -lcrypt  -lcrypto  -lhx509  -lroken  -lwind  -lheimbase -lprivateheimipcc   -lprivatessh  -lcrypto  -lcrypt -lprivateldns  -lcrypto  -lz  -lypclnt  -lkrb5  -lasn1  -lcom_err  -lcrypt  -lcrypto  -lhx509  -lroken  -lwind  -lheimbase -lprivateheimipcc    -lroken   -lpthread -lc -lmalloc_simple
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2767
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:7455
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
[... cut ...]
/home/rnw24/sdk/sdk//bin/cheri-unknown-freebsd-ld: BFD 2.17.50 [FreeBSD] 2007-07-03 assertion fail /home/rnw24/sdk/cheribsd/gnu/usr.bin/binutils/libbfd/../../../../contrib/binutils/bfd/elfxx-mips.c:2472
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

Stop.
make[4]: stopped in /home/rnw24/git/cheribsd/usr.bin/telnet
*** Error code 1
```

With threading-related missing symbols, this likely won't work for a bit -- but continuing assertion problems are present which may be worth investigating.

davidchisnall, 2015-12-14:
It looks as if libcrypto also needs to be compiled with `-mxgot`.  It might be worth enabling this globally for CHERIABI things until we have a new linker.

brooksdavis, 2015-12-17:
Recompiling everything with `-mxgot` doesn't seem to fix the asserts for me.

brooksdavis, 2017-03-20:
Links with LLD so closing.

closed 2017-03-20 by brooksdavis