CTSRD-CHERI/cheribsd #74: Attempting to compile CheriABI pthread-liked ntpd(8) fails

rwatson, 2015-12-12:
Attempting to compile a pthread-linked CheriABI program:
```
diff --git a/usr.sbin/ntp/ntpd/Makefile b/usr.sbin/ntp/ntpd/Makefile
index 905ff3c..6114bb1 100644
--- a/usr.sbin/ntp/ntpd/Makefile
+++ b/usr.sbin/ntp/ntpd/Makefile
@@ -7,6 +7,9 @@ MAN=
 .PATH: ${.CURDIR}/../../../contrib/ntp/ntpd \
        ${.OBJDIR}
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  ntpd
 
 SRCS=  cmd_args.c ntp_config.c ntp_control.c ntp_crypto.c ntp_filegen.c \
```

Fails due to unimplemented malloc-related hooks in the threading library:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker -O -pipe -G0   -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/ntpd  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/pthreads/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/unix/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/sntp/libopts  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd -DSYS_FREEBSD -DPARSE -DHAVE_CONFIG_H  -DOPENSSL -DUSE_OPENSSL_CRYPTO_RAND -DAUTOKEY -Wno-error  -std=gnu99 -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Wno-parentheses -Qunused-arguments -c version.c -o version.o
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker -O -pipe -G0   -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/ntpd  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/pthreads/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/lib/isc/unix/include  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../../../contrib/ntp/sntp/libopts  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd/../  -I/home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd -DSYS_FREEBSD -DPARSE -DHAVE_CONFIG_H  -DOPENSSL -DUSE_OPENSSL_CRYPTO_RAND -DAUTOKEY -Wno-error  -std=gnu99 -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Wno-parentheses -Qunused-arguments  -Wl,--no-warn-mismatch -static -o ntpd cmd_args.o ntp_config.o ntp_control.o ntp_crypto.o ntp_filegen.o ntp_io.o ntp_leapsec.o ntp_loopfilter.o ntp_monitor.o ntp_parser.o ntp_peer.o ntp_proto.o ntp_refclock.o ntp_request.o ntp_restrict.o ntp_scanner.o ntp_signd.o ntp_timer.o ntp_util.o ntpd-opts.o ntpd.o rc_cmdlength.o refclock_acts.o refclock_arbiter.o refclock_arc.o refclock_as2201.o refclock_atom.o refclock_bancomm.o refclock_chronolog.o refclock_chu.o refclock_conf.o refclock_datum.o refclock_dumbclock.o refclock_fg.o refclock_gpsdjson.o refclock_gpsvme.o refclock_heath.o refclock_hopfpci.o refclock_hopfser.o refclock_hpgps.o refclock_irig.o refclock_jjy.o refclock_jupiter.o refclock_leitch.o refclock_local.o refclock_nmea.o refclock_neoclock4x.o refclock_oncore.o refclock_palisade.o refclock_parse.o refclock_pcf.o refclock_pst.o refclock_ripencc.o refclock_shm.o refclock_tpro.o refclock_true.o refclock_tsyncpci.o refclock_tt560.o refclock_ulink.o refclock_wwv.o refclock_wwvb.o refclock_zyfer.o version.o -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/usr.sbin/ntp/libparse -lparse -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/usr.sbin/ntp/libntp -lntp  -lm -L/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/usr.sbin/ntp/libopts -lopts   -lcrypto   -lpthread -lc -lmalloc_simple
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_rtld.o): In function `_thr_rtld_init':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_rtld.c:226: undefined reference to `_malloc_prefork'
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_rtld.c:227: undefined reference to `_malloc_postfork'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_fork.o): In function `__thr_fork':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_fork.c:170: undefined reference to `_malloc_prefork'
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_fork.c:239: undefined reference to `_malloc_postfork'
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_fork.c:217: undefined reference to `_malloc_postfork'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_create.o): In function `_pthread_create':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_create.c:71: undefined reference to `_malloc_first_thread'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_exit.o): In function `exit_thread':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_exit.c:281: undefined reference to `_malloc_thread_cleanup'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_ctrdtr.o): In function `_tcb_ctor':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_ctrdtr.c:45: undefined reference to `_rtld_allocate_tls'
/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp/usr/libcheri/libpthread.a(thr_ctrdtr.o): In function `_tcb_dtor':
/home/rnw24/git/cheribsd/lib/libthr/thread/thr_ctrdtr.c:55: undefined reference to `_rtld_free_tls'
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

Stop.
make[5]: stopped in /home/rnw24/git/cheribsd/usr.sbin/ntp/ntpd
*** Error code 1
```

This likely depends on completing the adaptation of ```jemalloc``` to run in CheriABI before it can be usefully resolved.

rwatson, 2015-12-12:
Likely depends on #75 to make useful progress.

brooksdavis, 2017-03-20:
`ntpd` does link with CheriABI these days.  It explodes when you point `ntpq -p` at it, but to does link.

closed 2017-03-20 by brooksdavis