CTSRD-CHERI/cheribsd #75: Use jemalloc in CheriABI

rwatson, 2015-12-12:
Currently, we use a simplified (```rtld```-derived) ```malloc()``` implementation rather than FreeBSD's ```jemalloc```. We will want to merge forward our version of FreeBSD to the latest version before performing that adaptation. However, a number of problems seem to relate to not yet using ```jemalloc```. This is a high-level issue to track adaptation of ```jemalloc``` to CheriABI, that other bugs can depend on.

brooksdavis, 2015-12-18:
With the latest compiler it builds...  No chance it works.

rwatson, 2016-04-06:
First cut appears in 495cc2b84f2538e8ce1310e863926af23d4a84e7.

rwatson, 2016-05-30:
```jemalloc``` enabled by default in CheriABI in 7018ae2eb4d237e73c8ac666e1adfec78383da48
Option to disable it removed in 8416b18a022bd14e70c75dca17b227195d0912d7

closed 2016-05-30 by rwatson