CTSRD-CHERI/cheribsd #76: ministat(1) crashes with a capability length violation

rwatson, 2015-12-12:
Building ```ministat(1)``` with CheriABI used to work (pretty) well:
```
diff --git a/usr.bin/ministat/Makefile b/usr.bin/ministat/Makefile
index 2a63797..0c7cf0e 100644
--- a/usr.bin/ministat/Makefile
+++ b/usr.bin/ministat/Makefile
@@ -1,6 +1,9 @@
 # $FreeBSD$
 PROG=  ministat
 
+WANT_CHERI=     pure
+WANT_DUMP=      yes
+
 LIBADD=        m
 
 .include <bsd.prog.mk>
```

However, I now experience crashes when doing trivial processing:
```
> cat > foo
1
23
4
5
2
23
3
> ./ministat foo
x foo
Signal 34
```

Console output suggests a bounds problem:
```
root@beri1:~ # cpuid = 0
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c03 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120083960 t:0
$c02: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd990 t:0
$c03: v:1 s:0 p:7fff81ff b:0000000160189000 l:0000000000000020 o:20 t:0
$c04: v:1 s:0 p:7fff81ff b:0000000160189fe0 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81ff b:0000000160189fc0 l:0000000000000020 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffd938 l:0000000000000008 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffd96c l:0000000000000004 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffd958 l:0000000000000004 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffd950 l:0000000000000004 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffd964 l:0000000000000004 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12003e430 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffd954 l:0000000000000004 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd938 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120004744 t:0
$c18: v:1 s:0 p:7fff81ff b:0000000160072000 l:0000000000000080 o:20 t:0
$c19: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd998 t:0
$c20: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd99c t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:1 s:0 p:7fff81fd b:0000007fffffd99c l:0000000000000004 o:0 t:0
$c23: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd994 t:0
$c24: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:7fffffd98c t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120004760 t:0

Apr 12 07:04:21 beri1 kernel: USER_CHERI_EXCEPTION: pid 11498 tid 100052 (ministat), uid 999: CP2 fault (type 0x32)
Apr 12 07:04:21 beri1 kernel: Trapframe Register Dump:
Apr 12 07:04:21 beri1 kernel: zero: 0   at: 0   v0: 0x20        v1: 0x2
Apr 12 07:04:21 beri1 kernel: a0: 0     a1: 0   a2: 0x7fffffd8f0        a3: 0x7fffffd640
Apr 12 07:04:21 beri1 kernel: a4: 0     a5: 0x69ebf966  a6: 0x4ec08e    a6: 0x69ebf965
Apr 12 07:04:21 beri1 kernel: t0: 0x5   t1: 0x16        t2: 0x7fffffd120        t3: 0x7fffffd0a0
Apr 12 07:04:21 beri1 kernel: t8: 0x1   t9: 0x120054bc0 s0: 0x1 s1: 0x4
Apr 12 07:04:21 beri1 kernel: s2: 0x7fffffd990  s3: 0   s4: 0x120080000 s5: 0x1
Apr 12 07:04:21 beri1 kernel: s6: 0     s7: 0x40        k0: 0   k1: 0
Apr 12 07:04:21 beri1 kernel: gp: 0x12008a5f0   sp: 0x7fffffd900        s8: 0x7fffffd900        ra: 0x7fffffde3c
Apr 12 07:04:21 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0        badvaddr: 0x160189000
Apr 12 07:04:21 beri1 kernel: cause: 0x148      pc: 0x120004760
```

This code appears to be experiencing the problem:
```
static void
PlotSet(struct dataset *ds, int val)
{
        struct plot *pl;
        int i, j, m, x;
        unsigned n;
        int bar;

        pl = &plot;
        if (pl->span == 0)
                return;

        if (pl->separate_bars)
                bar = val-1;
        else
                bar = 0;

        if (pl->bar == NULL)
                pl->bar = calloc(sizeof(char *), pl->num_datasets);
        if (pl->bar[bar] == NULL) {
                pl->bar[bar] = malloc(pl->width);                          <------ Boom here
                memset(pl->bar[bar], 0, pl->width);
        }
```

A bit of tracing and disassembly suggests that, indeed, $c3 has a surprisingly non-zero offset when returned from ```calloc(3)```:
```$c03: v:1 s:0 p:7fff81ff b:0000000160189000 l:0000000000000020 o:20 t:0```

There's a surprising CIncOffset at the end of the ```malloc(3)``` code in question:
```
# Some pointer magic:
#
#   120054870:   640100ef        daddiu  at,zero,239
Time=  42136420361459 : 0000000120054870: 640100ef      daddiu  at,zr,239  DestReg <- 0x00000000000000ef {254}

#   120054874:   e8210000        csb     at,zero,0(c1)
Time=  42136420361460 : 0000000120054874: e8210000      swc2    r1,0(at)        []  Address 0x0000000160189000 <- 0x00000000000000ef {254}

#        op->ov_index = bucket;
#   120054878:   ea010008        csb     s0,zero,1(c1)
Time=  42136420361461 : 0000000120054878: ea010008      swc2    r1,8(s0)        []  Address 0x0000000160189001 <- 0x0000000000000002 {254}

#         */
#        op->ov_size = (nbytes + RSLOP - 1) & ~(RSLOP - 1);
#        op->ov_rmagic = RMAGIC;
#        *(u_short *)((caddr_t)(op + 1) + op->ov_size) = RMAGIC;
##endif
#        return ((void *)(op + 1));
#   12005487c:   64010020        daddiu  at,zero,32
Time=  42136420361462 : 000000012005487c: 64010020      daddiu  at,zr,32  DestReg <- 0x0000000000000020 {254}

#   120054880:   08015226        j       120054898 <malloc+0x228>
Time=  42136420361499 : 0000000120054880: 08015226      j       0x0000000120054898 branch to 0x0000000120054898 {254}

# XXXRW: This code takes what looks 
#   120054884:   49b20840        cincoffset      c18,c1,at
Time=  42136420361504 : 0000000120054884: 49b20840        CapReg <- tag:1 u:0 perms:0x000000fe type:0x000000 offset:0x0000000100089020 base:0x0000000100089000 length:0x0000000000000020 {254}

#                amt = pagesz;
#                bucket = pagebucket;
#        }
#        n = -(sizeof (*op) + RSLOP);
#        while (nbytes > (size_t)amt + n) {
#                amt <<= 1;
```

This feels a bit like a ```malloc(3)``` bug but I'm not certain -- certainly, the resulting apparent offset of the allocation would ideally be zero, and I'm left wondering if we're adding a size more than once when we don't mean to be (e.g., for a pointer-size allocation?). Alternatively, maybe it's a compiler code-generation bug in compiling this malloc code (```lib/libc_cheri/malloc.c```).

Possibly @davidchisnall or @brooksdavis will take a view on this.

rwatson, 2015-12-12:
Or maybe this is a symptom of a ```malloc(3)``` bug when a bucket becomes almost empty, and we allocate too little space at the end of a bucket? (I'd assumed that ```malloc(3)``` was using CSetBounds relative only to the allocation, but reading the code, it looks like the bounds may also include metadata and other adjacent allocations in the same bucket?).

rwatson, 2015-12-12:
Oh, I wonder if this is an issue when selecting a bucket size, in which the size of capability extensions to the overhead structure are not properly taken into account -- either by ```malloc(3)``` or the compiler in calculating the size of the overhead structure (which is a union containing a capability):
```
/*
 * The overhead on a block is at least 4 bytes.  When free, this space
 * contains a pointer to the next free block, and the bottom two bits must
 * be zero.  When in use, the first byte is set to MAGIC, and the second
 * byte is the size index.  The remaining bytes are for alignment.
 * If range checking is enabled then a second word holds the size of the
 * requested block, less 1, rounded up to a multiple of sizeof(RMAGIC).
 * The order of elements is critical: ov_magic must overlay the low order
 * bits of ov_next, and ov_magic can not be a valid ov_next bit pattern.
 */ 
union   overhead {
        union   overhead *ov_next;      /* when free */
        struct { 
                u_char  ovu_magic;      /* magic number */
                u_char  ovu_index;      /* bucket # */
#ifdef RCHECK
                u_short ovu_rmagic;     /* range magic number */
                u_int   ovu_size;       /* actual block size */
#endif
        } ovu;
};
```

rwatson, 2015-12-12:
Yes, seems likely this needs updating:
```
        n = pagesz - sizeof (*op) - RSLOP;
        if (nbytes <= (unsigned long)n) {
#ifndef RCHECK
                amt = 32;       /* size of first bucket */
                bucket = 2;
#else
                amt = 16;       /* size of first bucket */
                bucket = 1;
#endif
        } else {
                amt = pagesz;
                bucket = pagebucket;
        }
```

rwatson, 2015-12-12:
In particular, perhaps this:
```
#ifdef RCHECK
#define RSLOP           sizeof (u_short)
#else
#define RSLOP           0
#endif
```

rwatson, 2015-12-12:
Indeed, it looks like we're running into trouble when ```morecore()``` is used with bucket '2' -- i.e., requesting 32-byte allocations, in ```ministat(1)```. Oddly, when I write a simple stand-alone ```malloc()``` test program trying various sizes, I don't encounter the problem as here -- I guess there is some difference in code paths (perhaps to do with reallocation). Regardless, it sounds like we have a problem allocating or reallocating 32-byte memory allocations, in which we end up with a bound being set to ```sz``` of 0x20, rather than the desired size of 0x20 + the overhead of 0x20. Possibly, we simply need to adjust the behaviour of the allocator to allow for a larger overhead somewhere, but I've not quite spotted where.

rwatson, 2015-12-15:
Assigning to @davidchisnall as dependent on fix to https://github.com/CTSRD-CHERI/llvm/issues/144. Have not yet tried -O0 workaround, but will do so shortly.

rwatson, 2015-12-15:
(Summary from submitted LLVM issue: due to a possible optimisation bug, ```calloc()``` is always requesting zero-sized allocation from ```malloc()```, hence the apparently successful allocation having a bounds that is too short to hold the desired pointer store. Bounds checking in action!)

rwatson, 2015-12-15:
I can confirm that recompiling ```ministat(1)``` with ```-O0``` leads to a working binary:
```
root@beri1:/tmp # ./ministat foo
x foo
+------------------------------------------------------------------------------+
|                                                                             x|
|       x  x  x  x   x                                                        x|
||_______________M______________A_______________________________|              |
+------------------------------------------------------------------------------+
    N           Min           Max        Median           Avg        Stddev
x   7             1            23             4     8.7142857     9.8440216
root@beri1:/tmp # ./ministat foo foo
x foo
+ foo
+------------------------------------------------------------------------------+
|                                                                             *|
|       *  *  *  *   *                                                        *|
||_______________M______________A_______________________________|              |
+------------------------------------------------------------------------------+
    N           Min           Max        Median           Avg        Stddev
x   7             1            23             4     8.7142857     9.8440216
+   7             1            23             4     8.7142857     9.8440216
No difference proven at 95.0% confidence
```

rwatson, 2015-12-15:
Worked around for now by forcing ```-O0``` for all CheriABI applications in b04dd8adf42f16781e3a1ac4369c95dce7fef5d2.

davidchisnall, 2015-12-16:
This is hopefully fixed by the fix to CTSRD-CHERI/llvm#144

rwatson, 2015-12-16:
I can confirm that, even with ```-O0``` removed, ```ministat(1)``` now appears to work properly (at least in casual use).

closed 2015-12-16 by rwatson