CTSRD-CHERI/cheribsd #78: CheriABI ping6(8): sendmsg: Bad Address

rwatson, 2015-12-14:
When compiling ```ping6(8)``` for CheriABI:
```
diff --git a/sbin/ping6/Makefile b/sbin/ping6/Makefile
index 35a76e4..58e2d97 100644
--- a/sbin/ping6/Makefile
+++ b/sbin/ping6/Makefile
@@ -3,6 +3,9 @@
 PROG=  ping6
 MAN=   ping6.8
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 CFLAGS+=-DIPSEC -DKAME_SCOPEID -DUSE_RFC2292BIS \
        -DHAVE_ARC4RANDOM
 WARNS?=        3
```

An unexpected ```Bad Address``` address occurs:
```
root@beri1:/tmp # ./ping6 localhost
PING6(56=40+8+8 bytes) ::1 --> :
ping6: sendmsg: Bad address
ping6: wrote localhost 16 chars, ret=-1
ping6: sendmsg: Bad address
ping6: wrote localhost 16 chars, ret=-1
^C
--- localhost ping6 statistics ---
2 packets transmitted, 0 packets received, 100.0% packet loss
```

A non-CheriABI ```ping6(8)``` appears to work fine:
root@beri1:/tmp # ping6 localhost
PING6(56=40+8+8 bytes) ::1 --> ::1
16 bytes from ::1, icmp_seq=0 hlim=64 time=5.876 ms
^C
--- localhost ping6 statistics ---
1 packets transmitted, 1 packets received, 0.0% packet loss
round-trip min/avg/max/std-dev = 5.876/5.876/5.876/0.000 ms
```

rwatson, 2015-12-17:
This is likely the same underlying cause as the problem in #84.

brooksdavis, 2015-12-17:
It looks like this one has been fixed and is unrelated to the current #84 blocker.

```
root@beri1:/home/ctsrd # ./ping6 localhost
PING6(56=40+8+8 bytes) ::1 --> ::1
16 bytes from ::1, icmp_seq=0 hlim=64 time=8.773 ms
16 bytes from ::1, icmp_seq=1 hlim=64 time=5.446 ms
16 bytes from ::1, icmp_seq=2 hlim=64 time=5.402 ms
16 bytes from ::1, icmp_seq=3 hlim=64 time=5.364 ms
16 bytes from ::1, icmp_seq=4 hlim=64 time=5.425 ms
16 bytes from ::1, icmp_seq=5 hlim=64 time=5.442 ms
16 bytes from ::1, icmp_seq=6 hlim=64 time=5.384 ms
16 bytes from ::1, icmp_seq=7 hlim=64 time=5.482 ms
16 bytes from ::1, icmp_seq=8 hlim=64 time=5.396 ms
16 bytes from ::1, icmp_seq=9 hlim=64 time=5.311 ms
16 bytes from ::1, icmp_seq=10 hlim=64 time=5.419 ms
16 bytes from ::1, icmp_seq=11 hlim=64 time=5.313 ms
16 bytes from ::1, icmp_seq=12 hlim=64 time=5.353 ms
16 bytes from ::1, icmp_seq=13 hlim=64 time=5.337 ms
^CIllegal instruction
root@beri1:/home/ctsrd # file ping
ping: ELF 64-bit MSB executable, *unknown arch 0xc256* version 1 (FreeBSD), statically linked, for FreeBSD 11.0 (1100077), not stripped
```

closed 2015-12-17 by brooksdavis