CTSRD-CHERI/cheribsd #80: ncal(1) shows 1900 rather than the current year in its default output

rwatson, 2015-12-14:
When compiling ```ncal(1)``` for CheriABI:
```
diff --git a/usr.bin/ncal/Makefile b/usr.bin/ncal/Makefile
index 23b5da2..1c399c9 100644
--- a/usr.bin/ncal/Makefile
+++ b/usr.bin/ncal/Makefile
@@ -2,6 +2,9 @@
 
 .include <src.opts.mk>
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  ncal
 
 LIBADD=        calendar ncursesw
```

The output of the default command shows 1900 instead of 2015 as expected:
```
root@beri1:/tmp # ./ncal
    January 1900      
Mo  1  8 15 22 29   
Tu  2  9 16 23 30   
We  3 10 17 24 31   
Th  4 11 18 25      
Fr  5 12 19 26      
Sa  6 13 20 27      
Su  7 14 21 28      
```

MIPS ```ncal(1)```:
```
    January 1900      
Mo  1  8 15 22 29   
Tu  2  9 16 23 30   
We  3 10 17 24 31   
Th  4 11 18 25      
Fr  5 12 19 26      
Sa  6 13 20 27      
Su  7 14 21 28      
```

rwatson, 2015-12-14:
Possibly related: forcing CheriABI ```ncal(3)``` to show the current year still doesn't lead to a highlight of the current date (not visible in copy-and-paste output). This suggests that ```ncal(3)``` is having trouble retrieving the current date (or interpreting it?).

rwatson, 2015-12-15:
Worked around for now by forcing ```-O0``` for all CheriABI applications in b04dd8adf42f16781e3a1ac4369c95dce7fef5d2.

rwatson, 2015-12-15:
(Relates to CTSRD-CHERI/llvm#144)

davidchisnall, 2015-12-16:
May be fixed now that CTSRD-CHERI/llvm#144 is believed fixed.

rwatson, 2015-12-16:
I can confirm that even with ```-O0``` removed from the build, ```ncal(1)``` now appears to work as intended (at least in casual use).

closed 2015-12-16 by rwatson