CTSRD-CHERI/cheribsd #81: CheriABI elfdump(1) reports all files are not ELF

rwatson, 2015-12-14:
Building ```elfdump(1)``` for CheriABI:
```
diff --git a/usr.bin/elfdump/Makefile b/usr.bin/elfdump/Makefile
index 22e1ca9..2257aa7 100644
--- a/usr.bin/elfdump/Makefile
+++ b/usr.bin/elfdump/Makefile
@@ -2,4 +2,7 @@
 
 PROG=  elfdump
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 .include <bsd.prog.mk>
```

Appears to give a version of ```elfdump(1)``` that no longer recognises ELF files:
```
root@beri1:/tmp # ./elfdump -a /bin/echo                                        
elfdump: not an elf file
```

In contrast, the MIPS ```elfdump(1)``` appears to work as expected:
```
root@beri1:/tmp # elfdump -a /bin/echo | head -10                               

elf header:

        e_ident: ELFCLASS64 ELFDATA2MSB ELFOSABI_FREEBSD
        e_type: ET_EXEC
        e_machine: EM_MIPS
        e_version: EV_CURRENT
        e_entry: 0x1200009a0
        e_phoff: 64
        e_shoff: 5912
```

rwatson, 2015-12-15:
Looks like it fails here:
```
        e = mmap(NULL, sb.st_size, PROT_READ, MAP_SHARED, fd, 0);
        if (e == MAP_FAILED)
                err(1, NULL);
        if (!IS_ELF(*(Elf32_Ehdr *)e))
                errx(1, "not an elf file");
```

The ```mmap(2)``` succeeds, and according to ```ktrace(1)```, is even for the right file. Further investigation required, but sounds like it might relate to types/casts/etc. ```struct Elf32_Ehdr``` doesn't appear to contain any pointers (for good reason) early in the structure, so it's not an obvious pointer-size mismatch (preprocessed output from build):
```
/*
 * ELF header.
 */
typedef struct {
typedef uint32_t Elf32_Addr;
typedef uint16_t Elf32_Half;
typedef uint32_t Elf32_Off;
typedef int32_t Elf32_Sword;
typedef uint32_t Elf32_Word;
typedef uint64_t Elf32_Lword;
typedef Elf32_Word Elf32_Hashelt;
typedef Elf32_Word Elf32_Size;
typedef Elf32_Sword Elf32_Ssize;

typedef struct {
 unsigned char e_ident[16];
 Elf32_Half e_type;
 Elf32_Half e_machine;
 Elf32_Word e_version;
 Elf32_Addr e_entry;
 Elf32_Off e_phoff;
 Elf32_Off e_shoff;
 Elf32_Word e_flags;
 Elf32_Half e_ehsize;
 Elf32_Half e_phentsize;
 Elf32_Half e_phnum;
 Elf32_Half e_shentsize;
 Elf32_Half e_shnum;
 Elf32_Half e_shstrndx;
} Elf32_Ehdr;
```

Note that ```elfdump(1)``` is casting to a 32-bit ELF header in order to decide if the file is ELF, which should also work for 64-bit ELF headers that are similarly laid out for the first few fields.

```IS_ELF()``` appears relatively innocuous, on face value:
```
/* e_ident */
#define IS_ELF(ehdr)    ((ehdr).e_ident[EI_MAG0] == ELFMAG0 && \
                         (ehdr).e_ident[EI_MAG1] == ELFMAG1 && \
                         (ehdr).e_ident[EI_MAG2] == ELFMAG2 && \   
                         (ehdr).e_ident[EI_MAG3] == ELFMAG3)
```

rwatson, 2015-12-15:
Pre-processed version of the pertinent code:
```
e = mmap(((void *)0), sb.st_size, 0x01, 0x0001, fd, 0);
 if (e == ((void *)-1))
  err(1, ((void *)0));
 if (!((*(Elf32_Ehdr *)e).e_ident[0] == 0x7f && (*(Elf32_Ehdr *)e).e_ident[1] == 'E' && (*(Elf32_Ehdr *)e).e_ident[2] == 'L' && (*(Elf32_Ehdr *)e).e_ident[3] == 'F'))
  errx(1, "not an elf file");
```

The code generation here is a bit confusing, and makes me wonder if the test following the call to ```mmap(2)``` has been optimised away:
```
# RW: Set up call to mmap(2)
   120003498:   66418d48        daddiu  at,s2,-29368
   12000349c:   48810047        cfromptr        c1,c0,at
   1200034a0:   c8210003        cld     at,zero,0(c1)
   1200034a4:   d82bf080        clc     c1,s8,128(c11)
   1200034a8:   c8810243        cld     a0,zero,72(c1)    # mmap() integer argument 1: size from stat
   1200034ac:   48000807        cgetpcc c1
   1200034b0:   49ac0841        csetoffset      c12,c1,at
   1200034b4:   64050001        daddiu  a1,zero,1     # mmap() integer argument 2: PROT_READ
   1200034b8:   64060001        daddiu  a2,zero,1     # mmap() integer argument 3: MAP_READ
   1200034bc:   64080000        daddiu  a4,zero,0     # mmap() integer argument 5: offset = 0
   1200034c0:   da4bf120        clc     c18,s8,288(c11)
   1200034c4:   49a39000        cmove   c3,c18    # mmap() pointer argument 1: NULL pointer
# RW: Call to mmap(2)
   1200034c8:   48f16000        cjalr   c17,c12
   1200034cc:   02003825        move    a3,s0    # mmap() integer argument 4: file-descriptor number
   1200034d0:   d82bf040        clc     c1,s8,64(c11)
   1200034d4:   f8610000        csc     c3,zero,0(c1)
        if (e == MAP_FAILED)
# RW: Construct NULL-relative (-1) for MAP_FAILED
   1200034d8:   6401ffff        daddiu  at,zero,-1
   1200034dc:   49a19041        csetoffset      c1,c18,at
   1200034e0:   49c11841        cne     at,c3,c1
# RW: If not equal (i.e., success) then branch below
   1200034e4:   14200034        bnez    at,1200035b8 <$BB0_22+0x498>
   1200034e8:   00000000        nop
                err(1, NULL);
# RW: Otherwise, call err().
   1200034ec:   664187b0        daddiu  at,s2,-30800
   1200034f0:   48810047        cfromptr        c1,c0,at
   1200034f4:   c8210003        cld     at,zero,0(c1)
   1200034f8:   48000807        cgetpcc c1
   1200034fc:   49ac0841        csetoffset      c12,c1,at
   120003500:   48830007        cfromptr        c3,c0,zero
   120003504:   48f16000        cjalr   c17,c12

# Long gap
...
        if (!IS_ELF(*(Elf32_Ehdr *)e))
                errx(1, "not an elf file");
# Unconditionally call errx() rather than performing test of c3->{fields}?
   1200035b8:   66418f98        daddiu  at,s2,-28776
   1200035bc:   48810047        cfromptr        c1,c0,at
   1200035c0:   c8210003        cld     at,zero,0(c1)
   1200035c4:   de428050        ld      v0,-32688(s2)
   1200035c8:   48000807        cgetpcc c1
   1200035cc:   49ac0841        csetoffset      c12,c1,at
   1200035d0:   64411a4b        daddiu  at,v0,6731
   1200035d4:   48830047        cfromptr        c3,c0,at   # errx() pointer argument: format string?
   1200035d8:   48f16000        cjalr   c17,c12 
   1200035dc:   64040001        daddiu  a0,zero,1    # errx() integer argument 1: -1
```

On face value, and assuming I've read the disassembly right, this suggests a compiler bug; assigning to @davidchisnall so he can opine on my reading.

Compiled as:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker -O -pipe -G0   -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wmissing-variable-declarations -Wthread-safety -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Qunused-arguments -c /home/rnw24/git/cheribsd/usr.bin/elfdump/elfdump.c -o elfdump.o
```

Preprocessed C: https://gist.github.com/rwatson/2bbc1c238afece3bac8b

Full objdump: https://gist.github.com/rwatson/1e7a915a8b77d6f1e735

davidchisnall, 2015-12-15:
This looks like it's another example of the GVN issue.

rwatson, 2015-12-15:
I can confirm that recompiling with ```-O0``` leads to a much happier CheriABI ```elfdump(1)```:
```
root@beri1:/tmp # ./elfdump -e elfdump                                          

elf header:

        e_ident: ELFCLASS64 ELFDATA2MSB ELFOSABI_FREEBSD
        e_type: ET_EXEC
        e_machine: (unknown machine) -- type 0xc256
        e_version: EV_CURRENT
        e_entry: 0x1200022e0
        e_phoff: 64
        e_shoff: 1410104
        e_flags: 805306375
        e_ehsize: 64
        e_phentsize: 56
        e_phnum: 4
        e_shentsize: 64
        e_shnum: 29
        e_shstrndx: 26
```

The GVN issue referred to by @davidchisnall is https://github.com/CTSRD-CHERI/llvm/issues/144.

rwatson, 2015-12-15:
Worked around for now by forcing ```-O0``` for all CheriABI applications in b04dd8adf42f16781e3a1ac4369c95dce7fef5d2.

davidchisnall, 2015-12-16:
Hopefully fixed by CTSRD-CHERI/llvm#144

rwatson, 2015-12-16:
I can confirm that, even with ```-O0``` removed from the build command line, ```elfdump(1)``` now appears to work (at least in casual use).

closed 2015-12-16 by rwatson