CTSRD-CHERI/cheribsd #82: CheriABI getfacl(1) experiences tag violation in ACL brand code

rwatson, 2015-12-14:
A CheriABI-compiled ```getfacl(1)```:
```
diff --git a/bin/getfacl/Makefile b/bin/getfacl/Makefile
index b7b5e983..c8e6175 100644
--- a/bin/getfacl/Makefile
+++ b/bin/getfacl/Makefile
@@ -2,4 +2,7 @@
 
 PROG=  getfacl
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 .include <bsd.prog.mk>
```

Experiences a tag violation when run on ```/tmp```:
```
root@beri1:/tmp # ./getfacl .
# file: .
# owner: 0
# group: wheel
Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x02 RegNum: $c03 (tag violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12001c9a0 t:0
$c02: v:1 s:0 p:7fff81ff b:0000000160170000 l:0000000000001000 o:38 t:0
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffda80 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffd90c l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:000081d5 b:0000007fffffdc00 l:0000000000000060 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffd6e0 l:0000000000000080 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12012e538 t:0
$c09: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120132a8c t:0
$c10: v:1 s:0 p:7fff81ff b:000000016016b000 l:0000000000000800 o:35 t:0
$c11: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12001c8d0 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffdc00 l:0000000000000038 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12001c9f8 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12001c8d0 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffda80 l:0000000000000020 o:0 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffda60 l:0000000000000020 o:0 t:0
$c22: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12000c0b0 t:0
$c23: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12000aea8 t:0
$c24: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12000b788 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12001c8e0 t:0
Apr 14 14:29:07 beri1 kernel: USER_CHERI_EXCEPTION: pid 17398 tid 100063 (getfacl), uid 0: CP2 fault (type 0x32)
Apr 14 14:29:07 beri1 kernel: Trapframe Register Dump:
Apr 14 14:29:07 beri1 kernel: zero: 0   at: 0xffc       v0: 0xffffffffffffe000  v1: 0x80008000
Apr 14 14:29:07 beri1 kernel: a0: 0x1   a1: 0x7fffffd8d0        a2: 0x4 a3: 0
Apr 14 14:29:07 beri1 kernel: a4: 0     a5: 0x552d2430  a6: 0x3a664970  a6: 0x552d2430
Apr 14 14:29:07 beri1 kernel: t0: 0x8   t1: 0x800000000000      t2: 0x5df2a6a200000000  t3: 0x5524eb5c
Apr 14 14:29:07 beri1 kernel: t8: 0     t9: 0x12001c9a0 s0: 0x1 s1: 0x120180760
Apr 14 14:29:07 beri1 kernel: s2: 0xffffffffffffffff    s3: 0xffffffffffffffff  s4: 0xffd7      s5: 0x20
Apr 14 14:29:07 beri1 kernel: s6: 0     s7: 0x120180760 k0: 0   k1: 0
Apr 14 14:29:07 beri1 kernel: gp: 0x2   sp: 0x7fffffd820        s8: 0x7fffffd820        ra: 0
Apr 14 14:29:07 beri1 kernel: sr: 0x4000fcf3    mullo: 0x298    mulhi: 0        badvaddr: 0x12000c0b0
Apr 14 14:29:07 beri1 kernel: cause: 0x148      pc: 0x12001c8e0
```

This appears to be in the ACL branding code:
```
000000012001c8d0 <_acl_brand>:
/*
 * Return brand of an ACL.
 */
int
_acl_brand(const acl_t acl)
{
   12001c8d0:   67bdffe0        daddiu  sp,sp,-32
   12001c8d4:   ebcbe8c3        csd     s8,sp,24(c11)
   12001c8d8:   03a0f025        move    s8,sp

        return (acl->ats_brand);
   12001c8dc:   64010ffc        daddiu  at,zero,4092
   12001c8e0:   c8430806        clw     v0,at,0(c3)           <--- Boom here
   12001c8e4:   03c0e825        move    sp,s8
   12001c8e8:   cbcbe8c3        cld     s8,sp,24(c11)
   12001c8ec:   49008800        cjr     c17
   12001c8f0:   67bd0020        daddiu  sp,sp,32
   12001c8f4:   00000000        nop

000000012001c8f8 <_entry_brand>:
}
```

Due to a NULL pointer in $c3:
```
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
```

The caller hinted at by $c17 is ```_acl_brand_may_be()```:
```
000000012001c9a0 <_acl_brand_may_be>:
/*
 * Return 1, iff branding ACL as "brand" is ok.
 */
int
_acl_brand_may_be(const acl_t acl, int brand)
{
   12001c9a0:   67bdff60        daddiu  sp,sp,-160
   12001c9a4:   67a50098        daddiu  a1,sp,152
   12001c9a8:   ebcb2803        csd     s8,a1,0(c11)
   12001c9ac:   67a50090        daddiu  a1,sp,144
   12001c9b0:   ea0b2803        csd     s0,a1,0(c11)
   12001c9b4:   fa6be860        csc     c19,sp,96(c11)
   12001c9b8:   fa4be840        csc     c18,sp,64(c11)
   12001c9bc:   fa2be820        csc     c17,sp,32(c11)
   12001c9c0:   03a0f025        move    s8,sp
   12001c9c4:   49b96002        cgetoffset      t9,c12
   12001c9c8:   3c010016        lui     at,0x16 
   12001c9cc:   0039082d        daddu   at,at,t9
   12001c9d0:   64213dc0        daddiu  at,at,15808

        if (_acl_brand(acl) == ACL_BRAND_UNKNOWN)
   12001c9d4:   6421a028        daddiu  at,at,-24536
   12001c9d8:   48810047        cfromptr        c1,c0,at
   12001c9dc:   c8210003        cld     at,zero,0(c1)
   12001c9e0:   00808025        move    s0,a0
   12001c9e4:   48000807        cgetpcc c1
   12001c9e8:   49b30841        csetoffset      c19,c1,at
   12001c9ec:   49ac9800        cmove   c12,c19
   12001c9f0:   48f16000        cjalr   c17,c12                  <-- Call to _acl_brand() here
   12001c9f4:   49b21800        cmove   c18,c3
   12001c9f8:   00400825        move    at,v0
   12001c9fc:   10200006        beqz    at,12001ca18 <_acl_brand_may_be+0x78>
   12001ca00:   24020001        li      v0,1
                return (1);
```

This is slightly surprising, as many consumers of ```_acl_brand_may_be()``` guard it against NULL pointers being called.

rwatson, 2015-12-15:
From prior conversations with @brooksdavis, it looks like the POSIX.1e ACL code uses interesting allocation and lower-bit tricks to associate an implied "brand" with allocated ACL memory (```lib/libc/posix1e/acl_branding.c```), which is likely tripping up the compiler (or, perhaps, our C-language approach?):
```
/*
 * An ugly detail of the implementation - fortunately not visible
 * to the API users - is the "branding": libc needs to keep track
 * of what "brand" ACL is: NFSv4, POSIX.1e or unknown.  It happens
 * automatically - for example, during acl_get_file(3) ACL gets
 * branded according to the "type" argument; during acl_set_permset
 * ACL, if its brand is unknown it gets branded as NFSv4 if any of the
 * NFSv4 permissions that are not valid for POSIX.1e ACL are set etc.
 * Branding information is used for printing out the ACL (acl_to_text(3)),
 * veryfying acl_set_whatever arguments (checking against setting
 * bits that are valid only for NFSv4 in ACL branded as POSIX.1e) etc.
 */
```

And, in particular, visible in the implementation of ```entry2acl()```:
```
static acl_t
entry2acl(acl_entry_t entry)
{
        acl_t aclp;

        aclp = (acl_t)(((long)entry >> _ACL_T_ALIGNMENT_BITS) << _ACL_T_ALIGNMEN
T_BITS);

        return (aclp);
}
```

This code appears to accept a valid ```entry```, but will always return NULL in CheriABI. Compiled code for ```entry2acl()```:
```
000000012001c970 <entry2acl>:
 * bits that are valid only for NFSv4 in ACL branded as POSIX.1e) etc.
 */

static acl_t
entry2acl(acl_entry_t entry)
{
   12001c970:   67bdffa0        daddiu  sp,sp,-96
   12001c974:   ebcbeac3        csd     s8,sp,88(c11)
   12001c978:   03a0f025        move    s8,sp
        acl_t aclp;

        aclp = (acl_t)(((long)entry >> _ACL_T_ALIGNMENT_BITS) << _ACL_T_ALIGNMEN
T_BITS);
   12001c97c:   49a11802        cgetoffset      at,c3                    <--- Got back 0x38 here; I pass no judgement on whether this is a good answer 
   12001c980:   6402e000        daddiu  v0,zero,-8192              <--- $v0 becomes 0xffffffffffffe000
   12001c984:   00220824        and     at,at,v0                          <--- $at becomes zero here regardless
   12001c988:   48810007        cfromptr        c1,c0,zero           <--- $zero as an argument here, rather than $at? Regardless, NULL capability generated.
   12001c98c:   49a30841        csetoffset      c3,c1,at               <--- Still a NULL capability here.

        return (aclp);
   12001c990:   03c0e825        move    sp,s8
   12001c994:   cbcbeac3        cld     s8,sp,88(c11)
   12001c998:   49008800        cjr     c17
   12001c99c:   67bd0060        daddiu  sp,sp,96
```

In the above annotations, $c3 is passed in as an actual tagged capability with an offset of 0x38 relative to its base (note slight loss of bits in tracing means that base/address aren't quite complete addresses):
```
Time=  62157264566890 : 000000012000af60: 49a39000        CapReg <- tag:1 u:0 perms:0x000000fe type:0x000000 offset:0x0000000100070038 base:0x0000000100070000 length:0x0000000000001000 {76}
```

Assigning to @davidchisnall as this is a C-oriented trick that we in principle support in the CHERI C model, but may not currently handle in the compiler (or perhaps don't). Regardless, this is an idiom that is used in FreeBSD's ```libc```.

More misc information:

```acl_branding.c```, as used in ```libc_cheri```, is build as follows:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -cheri-linker  -O -pipe -G0  -I/home/rnw24/git/cheribsd/lib/libc/include -I/home/rnw24/git/cheribsd/lib/libc/../../include -I/home/rnw24/git/cheribsd/lib/libc/mips -DNLS  -DSOFTFLOAT -D__DBINTERFACE_PRIVATE -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/gdtoa -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/libc-vis -DINET6 -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/worldcheri/home/rnw24/git/cheribsd/lib/libc -I/home/rnw24/git/cheribsd/lib/libc/resolv -D_ACL_PRIVATE -DPOSIX_MISTAKE -I/home/rnw24/git/cheribsd/lib/libc/../libmd -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/tzcode/stdtime -I/home/rnw24/git/cheribsd/lib/libc/stdtime  -I/home/rnw24/git/cheribsd/lib/libc/locale -DBROKEN_DES -DPORTMAP -DDES_BUILTIN -I/home/rnw24/git/cheribsd/lib/libc/rpc -I/home/rnw24/git/cheribsd/lib/libc/mips/softfloat  -I/home/rnw24/git/cheribsd/lib/libc/softfloat -DSOFTFLOAT_FOR_GCC -DYP -DNS_CACHING -DSYMBOL_VERSIONING -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Qunused-arguments -I/home/rnw24/git/cheribsd/lib/libutil -I/home/rnw24/git/cheribsd/lib/msun/mips -I/home/rnw24/git/cheribsd/lib/msun/src -c /home/rnw24/git/cheribsd/lib/libc/posix1e/acl_branding.c -o acl_branding.o
```

Preprocessed ```acl_branding.c```: https://gist.github.com/rwatson/2a7f59698f5139424d77

Objdump of ```getfacl(3)```: ```~rnw24/20151216-getfacl.dump``` on ```vica```, as too big for gist!

Tagged as CheriCCompat; can remove the tag if this turns out to be a compiler issue rather than something to fix in CheriBSD.

davidchisnall, 2015-12-15:
It looks as if the compiler is doing what I'd expect here.

rwatson, 2015-12-15:
In the abstract, presumably what the C programmer wants here is simply:
```
# Trim off use of lower-order bits in the pointer so that we can restore it to a de-referenceable state
cgetoffset $at, $c3
daddiu $v0, $zero, -8192
and $at, $at, $v0
csetoffset $c3, $c3, $at
```

This would end up with a version of $c3 that had a zero offset but remain a valid capability pointer to the intended memory -- without an attempt to re-derive using CFromPtr / NULL via $c0.

davidchisnall, 2015-12-15:
The issue is the cast via `long`, not via `intptr_t`.  Any cast to a pointer from a `long` will give an invalid pointer (a valid `intcap_t` with the numerical value stored in the pointer).

rwatson, 2015-12-15:
Will try switching to an ```(intptr_t)``` cast and see if life gets better. We'll keep an eye on the ```long``` cast issue and I guess find out whether there are too many places we need to change... Thanks!

rwatson, 2015-12-15:
Fix to ```libc``` committed as b156f8b5af8855e597246c4851df09d77bc08f27.

closed 2015-12-15 by rwatson