CTSRD-CHERI/cheribsd #84: CheriABI ping(8): unable to open system.dns service: Bad address

rwatson, 2015-12-15:
This issue picks up where #83 leaves off in debugging why ```system.dns``` fails to initialise when using a CheriABI ```ping(8)```.

Changes to ```ping(8)``` build:
```
diff --git a/sbin/ping/Makefile b/sbin/ping/Makefile
index 533f8b8..077dec6 100644
--- a/sbin/ping/Makefile
+++ b/sbin/ping/Makefile
@@ -3,6 +3,9 @@

 .include <src.opts.mk>

+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  ping
 MAN=   ping.8
 BINOWN=        root
```

Gives:
```
root@beri1:/tmp # ./ping 8.8.8.8
ping: unable to open system.dns service: Bad address
```

A system-call trace using ```ktrace(1)``` suggests that there is a problem with argument handling for ```sendmsg(2)```:
```
   726 ping     CALL  fcntl(0x6,F_GETFL,0)
   726 ping     RET   fcntl 2
   726 ping     CALL  select(0x6,0,0x7ffffeb058,0,0)
   726 ping     RET   select 1
   726 ping     CALL  sendmsg(0x5,0x7ffffeb640,0)
   726 ping     RET   sendmsg -1 errno 14 Bad address
```

rwatson, 2015-12-16:
This appears to occur because the kernel stack memory, into which the ```msghdr_c``` is copied, is not sufficiently capability-aligned. It may be that marking capability alignment on the type is sufficient -- but it could also be that we should just ```malloc()``` temporary storage with suitable alignment -- or play stack allocation tricks to get suitably aligned space.

brooksdavis, 2015-12-16:
If so that's a compiler bug since msghdr_c should be forced to the correct alignment no matter where it is allocated.

brooksdavis, 2015-12-16:
I found another alignment and sizing issue in the control message.  I believe the message data is nonsense due to the use of `_ALIGN` in the `CMSG_` defines.

brooksdavis, 2015-12-17:
3aeaec742940164a102668d267cbe4e506b40c95 fixes a libnv bug and 867e380fe89495c3bd93f81b3ace325e8bcf5bc1 at least partially addresses the control message issue.

brooksdavis, 2015-12-17:
It appears that libnv is broken by a hardware bug where CLD silently doesn't support unaligned accesses.  I've sent @jonwoodruff a bug report.

brooksdavis, 2015-12-17:
I've pushed a minor update to the sendmsg/recvmsg code in 30bb97fabddae5aad95a6e4bc87273bf4feec264.

rwatson, 2015-12-17:
This is likely the same problem as experienced in #78, so tagging it here for reference.

rwatson, 2015-12-17:
Looks like we now get a different problem, but still relating to ```casperd``` communications:
```
root@beri1:/tmp # ktrace 8.8.8.8
ping: unable to open system.dns service: No such file or directory
```

```ktrace``` excerpt:
```
   729 ping     CALL  socket(PF_LOCAL,SOCK_STREAM,0)
   729 ping     RET   socket 5
   729 ping     CALL  connect(0x5,0x7ffffed010,0x6a)
   729 ping     STRU  struct sockaddr { AF_LOCAL, /var/run/casper }
   729 ping     NAMI  "/var/run/casper"
   729 ping     RET   connect 0
   729 ping     CALL  fcntl(0x5,F_GETFL,0)
   729 ping     RET   fcntl 2
   729 ping     CALL  mmap(0,0x8000,0x3<PROT_READ|PROT_WRITE>,0x1002<MAP_PRIVATE|MAP_ANON>,0xffffffffffffffff,0)
   729 ping     RET   mmap 5907177472/0x160186000
   729 ping     CALL  fcntl(0x2,F_GETFL,0)
   729 ping     RET   fcntl 2
   729 ping     CALL  fcntl(0x2,F_GETFL,0)
   729 ping     RET   fcntl 2
   729 ping     CALL  fcntl(0x2,F_DUPFD_CLOEXEC,0)
   729 ping     RET   fcntl 6
   729 ping     CALL  select(0x6,0,0x7ffffecb00,0,0)
   729 ping     RET   select 1
   729 ping     CALL  sendto(0x5,0x160187120,0x61,0,0,0)
   729 ping     GIO   fd 5 wrote 97 bytes
       0x0000 6c00 8000 0000 0000 0000 0100 0000 0000  |l...............|
       0x0010 0000 4e04 0004 0000 0000 0000 0005 636d  |..N...........cm|
       0x0020 6400 6f70 656e 0004 0008 0000 0000 0000  |d.open..........|
       0x0030 000b 7365 7276 6963 6500 7379 7374 656d  |..service.system|
       0x0040 2e64 6e73 0006 0009 0000 0000 0000 0008  |.dns............|
       0x0050 7374 6465 7272 6664 0000 0000 0000 0000  |stderrfd........|
       0x0060 00                                       |.|

   729 ping     RET   sendto 97/0x61
   729 ping     CALL  fcntl(0x6,F_GETFL,0)
   729 ping     RET   fcntl 2
   729 ping     CALL  select(0x6,0,0x7ffffec740,0,0)
   729 ping     RET   select 1
   729 ping     CALL  sendmsg(0x5,0x7ffffec9c0,0)
   729 ping     GIO   fd 5 wrote 1 byte
       "\0"
   729 ping     RET   sendmsg 1
   729 ping     CALL  close(0x6)
   729 ping     RET   close 0
   729 ping     CALL  select(0x6,0x7ffffeca40,0,0,0)
   729 ping     RET   select 1
   729 ping     CALL  recvfrom(0x5,0x7ffffecd88,0x13,0,0,0)
   729 ping     GIO   fd 5 read 19 bytes
       0x0000 6c00 8000 0000 0000 0000 0100 0000 0000  |l...............|
       0x0010 0000 33                                  |..3|

   729 ping     RET   recvfrom 19/0x13
   729 ping     CALL  close(0x5)
   729 ping     RET   close 0
   729 ping     CALL  write(0x2,0x7ffffec4e0,0x6)
   729 ping     GIO   fd 2 wrote 6 bytes
       "ping: "
   729 ping     RET   write 6
   729 ping     CALL  write(0x2,0x7ffffec520,0x21)
   729 ping     GIO   fd 2 wrote 33 bytes
       "unable to open system.dns service"
   729 ping     RET   write 33/0x21
   729 ping     CALL  write(0x2,0x7ffffec4e0,0x2)
   729 ping     GIO   fd 2 wrote 2 bytes
       ": "
   729 ping     RET   write 2
   729 ping     CALL  stat(0x7ffffec618,0x7ffffecb88)
   729 ping     NAMI  "/usr/share/nls/C/libc.cat"
   729 ping     RET   stat -1 errno 2 No such file or directory
   729 ping     CALL  stat(0x7ffffec618,0x7ffffecb88)
   729 ping     NAMI  "/usr/share/nls/libc/C"
   729 ping     RET   stat -1 errno 2 No such file or directory
   729 ping     CALL  stat(0x7ffffec618,0x7ffffecb88)
   729 ping     NAMI  "/usr/local/share/nls/C/libc.cat"
   729 ping     RET   stat -1 errno 2 No such file or directory
   729 ping     CALL  stat(0x7ffffec618,0x7ffffecb88)
   729 ping     NAMI  "/usr/local/share/nls/libc/C"
   729 ping     RET   stat -1 errno 2 No such file or directory
   729 ping     CALL  write(0x2,0x7ffffec4e0,0x1a)
   729 ping     GIO   fd 2 wrote 26 bytes
       "No such file or directory
       "
   729 ping     RET   write 26/0x1a
   729 ping     CALL  exit(0x1)
```

It appears something goes wrong in/around ```recvmsg(2)```. The console reports, albeit only via ```/var/log/messages```:
```
Apr 17 01:28:19 beri1 kernel: Unaligned Capability Store Doubleword: pc=0x12001f85c, badvaddr=0x160187179
```

This also appears to be an ```nvpair``` problem:
```
        PJDLOG_ASSERT(*leftp >= sizeof(value));
   12001f84c:   c8260003        cld     at,zero,0(c6)
   12001f850:   2c210008        sltiu   at,at,8
   12001f854:   14200045        bnez    at,12001f96c <nvpair_pack_descriptor+0x1
b4>
   12001f858:   00000000        nop
        memcpy(ptr, &value, sizeof(value));  
   12001f85c:   e8640003        csd     v1,zero,0(c4)
        ptr += sizeof(value);
        *leftp -= sizeof(value);
   12001f860:   c8260003        cld     at,zero,0(c6)
                (*fdidxp)++;
        }
```

As this is in recently changed ```nvpair``` code, I likely need to check that everything is up-to-date, but if so, we could still have a wrong ```sizeof()``` or type somewhere in that code.

brooksdavis, 2015-12-17:
The alignment issue is due to `nvlist`'s wire format being excessively compact.  All messages begin with:

```C
struct nvlist_header {
        uint8_t         nvlh_magic;
        uint8_t         nvlh_version;
        uint8_t         nvlh_flags;
        uint64_t        nvlh_descriptors;
        uint64_t        nvlh_size;
} __packed;
```

This insures that every access will be misaligned unless subsequent `nvpairs` accidentally align the buffer.

brooksdavis, 2015-12-17:
The `CLD` alignment bug is causing `nvlh_size` to be misread as 0x0000010000000000 which ends up passed to `malloc` which returns `NULL`.

rwatson, 2015-12-18:
With the latest bit file, I'm now seeing:
```
root@beri1:/tmp # ./ping 8.8.8.8
[ERROR] Unable to receive message from client: Invalid argument.
ping: unable to limit access to system.dns service: Socket is not connected
```

brooksdavis, 2015-12-18:
Hmm.  With the latest bitfile and a fresh build ping works for me modulo #87
```
root@beri1:/home/ctsrd # ./ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: icmp_seq=0 ttl=52 time=23.293 ms
64 bytes from 8.8.8.8: icmp_seq=1 ttl=52 time=23.489 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=52 time=28.469 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=52 time=19.168 ms
^CIllegal instruction
```

rwatson, 2015-12-20:
Tagging as CheriCCompat due to part of the problem being resolved by 3aeaec7, although the majority of the problem addressed here relates to ```sendmsg(2)```.

rwatson, 2017-03-19:
Closing this issue as it appears to have been resolved (although I'll shortly file a new issue on a new problem).

closed 2017-03-19 by rwatson