CTSRD-CHERI/cheribsd #85: CheriABI openssl(1) experiences a segmentation fault when using s_client

rwatson, 2015-12-17:
```openssl(1)```, built for CheriABI:
```
diff --git a/secure/usr.bin/openssl/Makefile b/secure/usr.bin/openssl/Makefile
index ccc1b6f..0b5374f 100644
--- a/secure/usr.bin/openssl/Makefile
+++ b/secure/usr.bin/openssl/Makefile
@@ -1,5 +1,8 @@
 # $FreeBSD$
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  openssl
 
 LIBADD=        ssl crypto
```

Experiences a segmentation fault when used with ```s_client```:
```
root@beri1:/tmp # ./openssl s_client
Segmentation fault
```

Console output:
```
cpuid = 0
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81ff b:00000001605a6000 l:0000000000008000 o:37a0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffb120 l:0000000000000020 o:0 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffb580 l:0000000000000020 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffb560 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffb580 l:cpuid = 0
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81ff b:00000001605a6000 l:0000000000008000 o:37a0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffb0e0 l:0000000000000020 o:0 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffb540 l:0000000000000020 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffb520 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffb540 l:0000000000000020 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffb520 l:0000000000000020 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f280 o:7c0e0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffb120 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffa788 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffa808 l:0000000000000008 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f280 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1201282a0 t:0
$c13: v:1 s:0 p:000081d5 b:0000007fffffb2e0 l:0000000000000008 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffce58 l:0000000000000004 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffa7e8 l:0000000000000008 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffa7e0 l:0000000000000008 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1201273fc t:0
$c18: v:1 s:0 p:7fff81fd b:0000007fffffb100 l:0000000000000020 o:0 t:0
$c19: v:1 s:0 p:7fff81ff b:00000001605a6000 l:0000000000008000 o:20 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffb140 l:0000000000000020 o:0 t:0
$c21: v:1 s:0 p:7fff81fd b:0000007fffffb160 l:0000000000000020 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:1 s:0 p:7fff81ff b:00000001605a6000 l:0000000000008000 o:37a0 t:0
$c24: v:1 s:0 p:7fff81ff b:00000001605a6000 l:0000000000008000 o:20 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120128420 t:0
Apr 17 03:11:19 beri1 kernel: BAD_PAGE_FAULT: pid 1008 tid 100064 (openssl), uid 0: pc 0x120128420 got a read fault (type 0x2) at 0x7c2e0
Apr 17 03:11:19 beri1 kernel: Trapframe Register Dump:
Apr 17 03:11:19 beri1 kernel: zero: 0   at: 0x7c0c0     v0: 0x1206182a0 v1: 0x20
Apr 17 03:11:19 beri1 kernel: a0: 0x3   a1: 0x80        a2: 0   a3: 0
Apr 17 03:11:19 beri1 kernel: a4: 0     a5: 0   a6: 0   a6: 0x1
Apr 17 03:11:19 beri1 kernel: t0: 0xeba32aa53bd57f1b    t1: 0x17de1ab9f3da1     t2: 0x599ed3181699d     t3: 0xfd998a06e869
Apr 17 03:11:19 beri1 kernel: t8: 0x27e4824f918b8       t9: 0x1201282a0 s0: 0x1 s1: 0
Apr 17 03:11:19 beri1 kernel: s2: 0     s3: 0x20        s4: 0xffffffffffffffff  s5: 0
Apr 17 03:11:19 beri1 kernel: s6: 0x80  s7: 0x20        k0: 0   k1: 0
Apr 17 03:11:19 beri1 kernel: gp: 0x7c078       sp: 0x7bf60     s8: 0x7bf60     ra: 0x7d400
Apr 17 03:11:19 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0x4      badvaddr: 0x7c2e0
[... 110 lines cut ...]
                continue;
            if (alg_mac && !(alg_mac & cp->algorithm_mac))
                continue;
            if (alg_ssl && !(alg_ssl & cp->algorithm_ssl))
                continue;
            if ((algo_strength & SSL_EXP_MASK)
                && !(algo_strength & SSL_EXP_MASK & cp->algo_strength))
                continue;
            if ((algo_strength & SSL_STRONG_MASK)
                && !(algo_strength & SSL_STRONG_MASK & cp->algo_strength))
                continue;
        }
...
```

Tagging @davidchisnall so he can opine. Further attachments:

Full ```objdump``` output for ```openssl```: too big for gist; available as ```~rnw24/20151217-openssl.dump``` on ```vica```.

Preprocessed source for ssl_ciph.c: https://gist.github.com/rwatson/0c19f684d06e7de34360

Normally this file is compiled with:
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker  -O -pipe -G0  -DTERMIOS -DANSI_SOURCE -I/home/rnw24/git/cheribsd/secure/lib/libssl/../../../crypto/openssl -I/home/rnw24/git/cheribsd/secure/lib/libssl/../../../crypto/openssl/crypto -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/worldcheri/home/rnw24/git/cheribsd/secure/lib/libssl -DOPENSSL_THREADS -DDSO_DLFCN -DHAVE_DLFCN_H -Wno-error  -std=gnu99 -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Wno-parentheses -Qunused-arguments -c /home/rnw24/git/cheribsd/secure/lib/libssl/../../../crypto/openssl/ssl/ssl_ciph.c -o ssl_ciph.o
```

davidchisnall, 2015-12-17:
`$s8` is more conventionally spelled `$fp`.  This load is at the offset that is the same as the `$sp` displacement for this function, so looks as if it's a problem with loading arguments from the stack.

davidchisnall, 2015-12-17:
Should be fixed as a result of fixing CTSRD-CHERI/llvm#145.

rwatson, 2015-12-17:
Indeed, rebuilding with the new LLVM version leaves ```openssl(1)```'s ```s_client``` feature much happier:
```
root@beri1:/tmp # ./openssl s_client
connect: Connection refused
connect:errno=61
```

closed 2015-12-17 by rwatson