CTSRD-CHERI/cheribsd #88: CheriABI file(1) fails with capability length violation

rwatson, 2015-12-18:
Building a CheriABI ```file(1)```:
```
diff --git a/usr.bin/file/Makefile b/usr.bin/file/Makefile
index 70c211b..f7369d5 100644
--- a/usr.bin/file/Makefile
+++ b/usr.bin/file/Makefile
@@ -24,6 +24,9 @@ SRCDIR=       ${.CURDIR}/../../contrib/file
 .PATH: ${SRCDIR}/src
 .PATH: ${SRCDIR}/doc
 
+WANT_CHERI=pure
+WANT_DUMP=yes
+
 PROG=  file
 
 MAGICPATH?=    /usr/share/misc
```

I get a capability length violation when running ```file(1)``` on its own binary:
```
root@beri1:/tmp # ./file file
Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c02 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffae40 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff81ff b:00000001601a7000 l:0000000000000800 o:220 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffae3c l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007ffffface0 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffac8c l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201a5c60 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffa984 l:0000000000000004 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffa960 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffada0 l:0000000000000020 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffad9c l:0000000000000004 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f280 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffad94 l:0000000000000006 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffad88 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffad08 l:0000000000000080 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffb338 l:0000000000000004 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12008e68c t:0
$c18: v:1 s:0 p:7fff81fd b:0000007fffffbf20 l:0000000000000020 o:0 t:0
$c19: v:1 s:0 p:7fff81fd b:0000007fffffbf18 l:0000000000000008 o:0 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffbf14 l:0000000000000004 o:0 t:0
$c21: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12011fae0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12008e740 t:0
Apr 17 23:13:09 beri1 kernel: USER_CHERI_EXCEPTION: pid 771 tid 100051 (file), uid 0: CP2 fault (type 0x32)
Apr 17 23:13:09 beri1 kernel: Trapframe Register Dump:
Apr 17 23:13:09 beri1 kernel: zero: 0   at: 0x3fffffe00 v0: 0x4 v1: 0
Apr 17 23:13:09 beri1 kernel: a0: 0x8   a1: 0x7bc80     a2: 0x7bc80     a3: 0x7b9f8
Apr 17 23:13:09 beri1 kernel: a4: 0x7baa8       a5: 0x7c4f8     a6: 0   a6: 0
Apr 17 23:13:09 beri1 kernel: t0: 0x10  t1: 0x8 t2: 0   t3: 0
Apr 17 23:13:09 beri1 kernel: t8: 0x8   t9: 0x7bc90     s0: 0   s1: 0
Apr 17 23:13:09 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 17 23:13:09 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 17 23:13:09 beri1 kernel: gp: 0x7cda0       sp: 0x7bd40     s8: 0x7bd40     ra: 0
Apr 17 23:13:09 beri1 kernel: sr: 0x4000fcf3    mullo: 0x48     mulhi: 0        badvaddr: 0x1601a7000
Apr 17 23:13:09 beri1 kernel: cause: 0x148      pc: 0x12008e740
```

$pc of 0x12008e740 occurs in the function ```computejumps()```, which is part of ```libc```'s ```regex``` implementation, called by ```regcomp(3)```:
[... 132 lines cut ...]
   12008e750:   24210001        addiu   at,at,1
   12008e754:   64020000        daddiu  v0,zero,0
   12008e758:   e8210002        csw     at,zero,0(c1)
   12008e75c:   e84bf203        csd     v0,s8,64(c11)
   12008e760:   080239bf        j       12008e6fc <computejumps+0x13c>
   12008e764:   00000000        nop
```

$c2 is as follows:
```
$c02: v:1 s:0 p:7fff81ff b:00000001601a7000 l:0000000000000800 o:220 t:0
```

And $at has a value of 0x3fffffe00. This is definitely beyond the bounds of $c2.

Assigning initially to @davidchisnall to get his reading on the possible nature of the problem as I'm not currently having success in following the generated code.

Full ```objdump``` for ```file(1)```: https://gist.github.com/rwatson/227bb9ba5109eaeea544

Preprocessed C for ```regcomp.c```: https://gist.github.com/rwatson/7808bd392d396e7b8117

Normal build command line for ```regcomp.c```: 
```
/home/rnw24/sdk/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd  -msoft-float --sysroot=/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker  -O -pipe -G0  -I/home/rnw24/git/cheribsd/lib/libc/include -I/home/rnw24/git/cheribsd/lib/libc/../../include -I/home/rnw24/git/cheribsd/lib/libc/mips -DNLS  -DSOFTFLOAT -D__DBINTERFACE_PRIVATE -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/gdtoa -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/libc-vis -DINET6 -I/home/rnw24/obj/mips.mips64/home/rnw24/git/cheribsd/worldcheri/home/rnw24/git/cheribsd/lib/libc -I/home/rnw24/git/cheribsd/lib/libc/resolv -D_ACL_PRIVATE -DPOSIX_MISTAKE -I/home/rnw24/git/cheribsd/lib/libc/../libmd -I/home/rnw24/git/cheribsd/lib/libc/../../contrib/tzcode/stdtime -I/home/rnw24/git/cheribsd/lib/libc/stdtime  -I/home/rnw24/git/cheribsd/lib/libc/locale -DBROKEN_DES -DPORTMAP -DDES_BUILTIN -I/home/rnw24/git/cheribsd/lib/libc/rpc -I/home/rnw24/git/cheribsd/lib/libc/mips/softfloat  -I/home/rnw24/git/cheribsd/lib/libc/softfloat -DSOFTFLOAT_FOR_GCC -DYP -DNS_CACHING -DSYMBOL_VERSIONING -Wno-error  -std=gnu99 -Wsystem-headers -Werror -Wall -Wno-format-y2k -Wno-uninitialized -Wno-pointer-sign -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Wno-tautological-compare -Wno-unused-value -Wno-parentheses-equality -Wno-unused-function -Wno-enum-conversion -Wno-unused-local-typedef -Wno-switch -Wno-switch-enum -Wno-knr-promoted-parameter -Qunused-arguments -I/home/rnw24/git/cheribsd/lib/libutil -I/home/rnw24/git/cheribsd/lib/msun/mips -I/home/rnw24/git/cheribsd/lib/msun/src -c /home/rnw24/git/cheribsd/lib/libc/regex/regcomp.c -o regcomp.o
```

davidchisnall, 2015-12-18:
This looks as if it might be CTSRD-CHERI/llvm#142, which is a bit less fixed than previously believed.

brooksdavis, 2015-12-18:
Works for me:
```
root@beri1:/home/ctsrd # ./file file
file: ELF 64-bit MSB executable, MIPS with CHERI (unofficial), version 1 (FreeBSD), statically linked, for FreeBSD 11.0 (1100077), not stripped
```

rwatson, 2015-12-20:
Hmm. Still experiencing it here despite a rebuild of both toolchain and world:
```
root@beri1:/tmp # ./file file
cpuid = 0
cpuid = 0
CHERI cause: ExcCode: 0x01 RegNum: $c02 (length violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffaf60 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff81ff b:00000001601a7000 l:0000000000000800 o:220 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffaf5c l:0000000000000004 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffae00 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffadac l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:1201a5c60 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffaaa4 l:0000000000000004 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffaa80 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffaec0 l:0000000000000020 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffaebc l:0000000000000004 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f3a0 o:0 t:0
$c12: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffaeb4 l:0000000000000006 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffaea8 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffae28 l:0000000000000080 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffb458 l:0000000000000004 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12008e68c t:0
$c18: v:1 s:0 p:7fff81fd b:0000007fffffc040 l:0000000000000020 o:0 t:0
$c19: v:1 s:0 p:7fff81fd b:0000007fffffc038 l:0000000000000008 o:0 t:0
$c20: v:1 s:0 p:7fff81fd b:0000007fffffc034 l:0000000000000004 o:0 t:0
$c21: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12011fae0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12008e740 t:0
Apr 19 15:54:43 beri1 kernel: USER_CHERI_EXCEPTION: pid 707 tid 100053 (file), uid 0: CP2 fault (type 0x32)
Apr 19 15:54:43 beri1 kernel: Trapframe Register Dump:
Apr 19 15:54:43 beri1 kernel: zero: 0   at: 0x3fffffe00 v0: 0x4 v1: 0
Apr 19 15:54:43 beri1 kernel: a0: 0x8   a1: 0x7bda0     a2: 0x7bda0     a3: 0x7bb18
Apr 19 15:54:43 beri1 kernel: a4: 0x7bbc8       a5: 0x7c618     a6: 0   a6: 0
Apr 19 15:54:43 beri1 kernel: t0: 0x10  t1: 0x8 t2: 0   t3: 0
Apr 19 15:54:43 beri1 kernel: t8: 0x8   t9: 0x7bdb0     s0: 0   s1: 0
Apr 19 15:54:43 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 19 15:54:43 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 19 15:54:43 beri1 kernel: gp: 0x7cec0       sp: 0x7be60     s8: 0x7be60     ra: 0
Apr 19 15:54:43 beri1 kernel: sr: 0x4000fcf3    mullo: 0x48     mulhi: 0        badvaddr: 0x1601a7000
Apr 19 15:54:43 beri1 kernel: cause: 0x148      pc: 0x12008e740
Signal 34
```

From the disassembly:
```
        /* If the character does not exist in the pattern, the jump         * is equal to the number of characters in the pattern.
         */
        for (ch = CHAR_MIN; ch < (CHAR_MAX + 1); ch++)
   12008e6ec:   2403ff80        li      v1,-128
   12008e6f0:   d84bf060        clc     c2,s8,96(c11)
   12008e6f4:   e8620002        csw     v1,zero,0(c2)
   12008e6f8:   e84bf283        csd     v0,s8,80(c11)
   12008e6fc:   d82bf060        clc     c1,s8,96(c11)
   12008e700:   c8210006        clw     at,zero,0(c1)
   12008e704:   28210080        slti    at,at,128
   12008e708:   10200017        beqz    at,12008e768 <computejumps+0x1a8>
   12008e70c:   00000000        nop
   12008e710:   080239c6        j       12008e718 <computejumps+0x158>
   12008e714:   00000000        nop
                g->charjump[ch] = g->mlen;
   12008e718:   d82bf0c0        clc     c1,s8,192(c11)
   12008e71c:   d8410000        clc     c2,zero,0(c1)
   12008e720:   64010140        daddiu  at,zero,320
   12008e724:   c8420806        clw     v0,at,0(c2)
   12008e728:   d86bf060        clc     c3,s8,96(c11)
   12008e72c:   c8230002        clwu    at,zero,0(c3)
   12008e730:   64030100        daddiu  v1,zero,256
   12008e734:   d8421800        clc     c2,v1,0(c2)
   12008e738:   000108b8        dsll    at,at,0x2
   12008e73c:   64030000        daddiu  v1,zero,0
   12008e740:   e8420802        csw     v0,at,0(c2)              <-- boom here
   12008e744:   e86bf243        csd     v1,s8,72(c11)
```

This appears to be the identical problem.

brooksdavis, 2016-01-23:
This is working for me with both `-O0` and the default `-O2`.

root@beri1:/home/ctsrd # ./file file
file: ELF 64-bit MSB executable, MIPS with CHERI (unofficial), version 1 (FreeBSD), statically linked, for FreeBSD 11.0 (1100094), not stripped

closed 2016-01-23 by brooksdavis