CTSRD-CHERI/cheribsd #89: Add alternate signal stack support to CheriABI

brooksdavis, 2015-12-18:
The current CheriABI `sendsig` implementation (`cheriabi_sendsig`) doesn't support alternate signal stacks.  This needs to be fixed for libcheri to be secure.

brooksdavis, 2015-12-18:
Re-reading the code, `cheriabi_sendsig` does support alternative signal stacks, but `td->td_pcb->pcb_cherisignal.csig_c11` isn't updated when the alternate stack is installed so it won't work.

rwatson, 2015-12-20:
Given @davidchisnall's recent comments on $c12, I wonder if we also need to arrange for a suitable value to be installed in that register before the signal handler is called? Presumably some piece of code (perhaps the trampoline code) is already setting up $t9?

brooksdavis, 2015-12-20:
We do install a proper $c12 and $c17 in `cheriabi_sendsig`.  Even the most trivial signal handlers fail otherwise:

```C
       cheri_capability_copy(&capreg->cf_c12, &psp->ps_sigcap[_SIG_IDX(sig)]);
        cheri_capability_copy(&capreg->cf_c17,
            &td->td_pcb->pcb_cherisignal.csig_sigcode);
```

rwatson, 2015-12-20:
Indeed!

rwatson, 2015-12-21:
While not yet tested, ```sigaltstack(2)``` is at least now returning success for ```cheriabitest```. Other compilation / runtime problems / issues in #91 prevent me from determining if all works well with ```libcheri``` compartmentalisation, but we should know fairly soon one way or the other.

rwatson, 2016-01-01:
The alternative signal stack is now believed to work properly in CheriABI, following various bug fixes to the kernel stack implementation, ```libcheri``` signal handling, and ```cheritest``` signal handling.

closed 2016-01-01 by rwatson