CTSRD-CHERI/cheribsd #91: Make a CheriABI cheritest(1) pass all non-XFAIL tests

rwatson, 2015-12-19:
Currently, ```cheritest(1)``` is compiled as hybrid-ABI, but ideally it could also be compiled to CheriABI to ensure that various compartmentalisation functions work in that environment. This may require improvements to the kernel, ```libc_cheri```, and the test suite itself.

rwatson, 2015-12-19:
Now compiles for CheriABI as of 3cf50e919dd66f671e8d39e706d3e6446fb157cd, which corrects warnings and adapts to the ```__CHERI_SANDBOX__``` definition for ```mcontext```.

rwatson, 2015-12-19:
With ede1440a16a11eaa7ab2bf113e43f893c22be6dc, there is now a CheriABI version of ```cheritest``` built as ```cheriabitest```.

rwatson, 2015-12-19:
... Which unfortunately segfaults due to a page fault when starting to run, likely a property of pointer use in the program:
```
> ./cheriabitest -a 
Segmentation fault
```

Console output:
```
cpuid = 0
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffe040 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffe020 l:0000000000000020 o:0 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffe020 l:0000000000000020 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffdfe0 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12013c2c0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffdfa0 l:0000000000000020 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:12011a0e8 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffe1e0 l:0000000000000020 o:0 t:0
$c09: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c10: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f2e0 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12005d510 t:0
$c13: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c14: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c15: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c16: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120003d80 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12005d514 t:0
Apr 19 05:38:32 beri1 kernel: BAD_PAGE_FAULT: pid 2239 tid 100069 (cheriabitest), uid 999: pc 0x12005d514 got a write fault (type 0x3) at 0x7ef98
Apr 19 05:38:32 beri1 kernel: Trapframe Register Dump:
Apr 19 05:38:32 beri1 kernel: zero: 0   at: 0   v0: 0xffffffffffffffff  v1: 0
Apr 19 05:38:32 beri1 kernel: a0: 0x7f098       a1: 0   a2: 0   a3: 0
Apr 19 05:38:32 beri1 kernel: a4: 0     a5: 0   a6: 0   a6: 0
Apr 19 05:38:33 beri1 kernel: t0: 0     t1: 0   t2: 0   t3: 0
Apr 19 05:38:33 beri1 kernel: t8: 0     t9: 0x1201351e0 s0: 0   s1: 0
Apr 19 05:38:33 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 19 05:38:33 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 19 05:38:33 beri1 kernel: gp: 0     sp: 0x7eb40     s8: 0x7efa0     ra: 0
Apr 19 05:38:33 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0        badvaddr: 0x7ef98
Apr 19 05:38:33 beri1 kernel: cause: 0x10c      pc: 0x12005d514
Apr 19 05:38:33 beri1 kernel: Page table info for bad address 0x7ef98: pde = 0, pte = 0
```

rwatson, 2015-12-19:
It appears that the binary is linking against a non-pure-capability compilation of ```libc_cheri```:
`z`
000000012005d510 <sandbox_init>:
 */
int sb_verbose;

__attribute__ ((constructor)) static void
sandbox_init(void)
{
   12005d510:   67bdfba0        daddiu  sp,sp,-1120
   12005d514:   ffbf0458        sd      ra,1112(sp)            <-- boom here
   12005d518:   ffbe0450        sd      s8,1104(sp)
   12005d51c:   ffbc0448        sd      gp,1096(sp)
   12005d520:   ffb30440        sd      s3,1088(sp)
   12005d524:   ffb20438        sd      s2,1080(sp)
   12005d528:   ffb10430        sd      s1,1072(sp)
   12005d52c:   ffb00428        sd      s0,1064(sp)
   12005d530:   03a0f025        move    s8,sp
   12005d534:   3c01000f        lui     at,0xf
   12005d538:   0039082d        daddu   at,at,t9
   12005d53c:   643c5950        daddiu  gp,at,22864
```

rwatson, 2015-12-19:
Tagging @brooksdavis in case he has a view on why other libraries are being linked to ```cheriabitest``` as pure-capability, but for ```libc_cheri```, the build appears to be picking up a MIPS/Hybrid-compiled library?

rwatson, 2015-12-20:
With changes b4728d86a45d1843f247fdb0ba0211c46856b4bf and  ef808e865c1f8be0b3af8347300895312385e494, ```libcheri``` is now compiled as pure-capability when generating the library for ```worldcheri``` for use with ```cheriabitest```.

rwatson, 2015-12-20:
```cheriabitest``` now fails with a capability tag violation:
```
> ./cheriabitest 
Signal 34
```

Console output:
```
cpuid = 0
CHERI cause: ExcCode: 0x02 RegNum: $c07 (tag violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffd7e0 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffd800 l:0000000000000020 o:0 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffd820 l:0000000000000008 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffd830 l:0000000000000008 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffd880 l:0000000000000020 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffd840 l:0000000000000020 o:0 t:0
$c07: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:120157e60 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffd320 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffd838 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffd830 l:0000000000000008 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f300 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1200d1f70 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffd828 l:0000000000000008 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffd820 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffd800 l:0000000000000020 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffd7e0 l:0000000000000020 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12006feb4 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12006ffb4 t:0
Apr 19 17:58:41 beri1 kernel: USER_CHERI_EXCEPTION: pid 1046 tid 100068 (cheriabitest), uid 999: CP2 fault (type 0x32)
Apr 19 17:58:41 beri1 kernel: Trapframe Register Dump:
Apr 19 17:58:41 beri1 kernel: zero: 0   at: 0   v0: 0x20        v1: 0x20
Apr 19 17:58:41 beri1 kernel: a0: 0x7e7a0       a1: 0x7e760     a2: 0x7e348     a3: 0x7e3b8
Apr 19 17:58:41 beri1 kernel: a4: 0     a5: 0x7d798     a6: 0   a6: 0
Apr 19 17:58:41 beri1 kernel: t0: 0     t1: 0x6 t2: 0   t3: 0
Apr 19 17:58:41 beri1 kernel: t8: 0x1f  t9: 0x7e310     s0: 0   s1: 0
Apr 19 17:58:41 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 19 17:58:41 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 19 17:58:41 beri1 kernel: gp: 0     sp: 0x7e540     s8: 0x7e540     ra: 0
Apr 19 17:58:41 beri1 kernel: sr: 0x4000fcf3    mullo: 0x1      mulhi: 0x8      badvaddr: 0x1200700a0
Apr 19 17:58:41 beri1 kernel: cause: 0x148      pc: 0x12006ffb4
```

Disassembly around 0x12006ffb4, which is part of ```sandbox_make_vtable```:
```
000000012006fae8 <sandbox_make_vtable>:

__capability intptr_t *
sandbox_make_vtable(void *dataptr, const char *class,
    struct sandbox_provided_classes *provided_classes)
{
   12006fae8:   67bdfc00        daddiu  sp,sp,-1024
   12006faec:   67a403f8        daddiu  a0,sp,1016
   12006faf0:   ebcb2003        csd     s8,a0,0(c11)
   12006faf4:   67a403c0        daddiu  a0,sp,960
   12006faf8:   fa2b2000        csc     c17,a0,0(c11)
   12006fafc:   03a0f025        move    s8,sp
   12006fb00:   49b96002        cgetoffset      t9,c12
   12006fb04:   3c01000f        lui     at,0xf
   12006fb08:   0039082d        daddu   at,at,t9
   12006fb0c:   64210538        daddiu  at,at,1336
   12006fb10:   67d903a0        daddiu  t9,s8,928
[... 344 lines cut ...]
   12006ff8c:   49a22002        cgetoffset      v0,c4
   12006ff90:   67c40260        daddiu  a0,s8,608
   12006ff94:   d88b2000        clc     c4,a0,0(c11)
   12006ff98:   e8440003        csd     v0,zero,0(c4)
                            provided_classes->spcs_base) / sizeof(*vtable);
                        vtable[index] = cheri_ccallee_base[index];
   12006ff9c:   67c40260        daddiu  a0,s8,608
   12006ffa0:   d88b2000        clc     c4,a0,0(c11)
   12006ffa4:   c8440003        cld     v0,zero,0(c4)
   12006ffa8:   d8cbf120        clc     c6,s8,288(c11)
   12006ffac:   d8e60000        clc     c7,zero,0(c6)
   12006ffb0:   00021178        dsll    v0,v0,0x5
   12006ffb4:   d8e71000        clc     c7,v0,0(c7)         <-- boom here
   12006ffb8:   d90bf140        clc     c8,s8,320(c11)
   12006ffbc:   d9280000        clc     c9,zero,0(c8)
   12006ffc0:   f8e91000        csc     c7,v0,0(c9)
   12006ffc4:   e82bf303        csd     at,s8,96(c11)
```

$c7 is:
```
$c07: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:120157e60 t:0
```

It seems likely that this is due to errant integer casting of a pointer in the function.

rwatson, 2015-12-20:
Now corrected by 4bc5e3787755a003d7d7106b9a68f8d179dc5fd6, which ensures that libcheri derives pointers for system classes from ```cheri_getdefault()```, rather than NULL.

rwatson, 2015-12-20:
Further update to ```cheri_maketype()``` to similarly derive object-type capabilities from ```cheri_getdefault()``` rather than NULL: 37c0765228bad3d7c14ce8246e68fb9db2f2f8c4

rwatson, 2015-12-20:
Tagging as CheriCCompat as a fair number of these bugs have to do with casts from integers to pointers, whose adaptation to CheriABI is a bit programmer-unfriendly.

rwatson, 2015-12-20:
I now experience a further tag violation during sandbox setup:
```
cpuid = 0
CHERI cause: ExcCode: 0x02 RegNum: $c03 (tag violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffdf58 l:0000000000000008 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffdf60 l:0000000000000020 o:0 t:0
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:7fffffeec0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffdf58 l:0000000000000008 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffdf50 l:0000000000000008 o:0 t:0
$c06: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffdad8 l:0000000000000008 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffdb5c l:0000000000000004 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffdae0 l:0000000000000020 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffdad8 l:0000000000000008 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f300 o:0 t:0
$c12: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:1200b19f0 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffdad4 l:0000000000000004 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffdac8 l:0000000000000008 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffd800 l:0000000000000020 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffd7e0 l:0000000000000020 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12006a31c t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12006a364 t:0
Apr 19 23:36:20 beri1 kernel: USER_CHERI_EXCEPTION: pid 1987 tid 100060 (cheriabitest), uid 999: CP2 fault (type 0x32)
Apr 19 23:36:20 beri1 kernel: Trapframe Register Dump:
Apr 19 23:36:20 beri1 kernel: zero: 0   at: 0xa0        v0: 0   v1: 0x1800
Apr 19 23:36:20 beri1 kernel: a0: 0x2   a1: 0   a2: 0x7ebd0     a3: 0
Apr 19 23:36:20 beri1 kernel: a4: 0     a5: 0x7d798     a6: 0   a6: 0
Apr 19 23:36:20 beri1 kernel: t0: 0     t1: 0x15        t2: 0   t3: 0
Apr 19 23:36:20 beri1 kernel: t8: 0x1f  t9: 0x7ead4     s0: 0   s1: 0
Apr 19 23:36:20 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 19 23:36:20 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 19 23:36:20 beri1 kernel: gp: 0     sp: 0x7eea0     s8: 0x7eea0     ra: 0
Apr 19 23:36:20 beri1 kernel: sr: 0x4000fcf3    mullo: 0x1      mulhi: 0x18     badvaddr: 0x1200932f0
Apr 19 23:36:20 beri1 kernel: cause: 0x148      pc: 0x12006a364
```

Code disassembly leading up to 0x12006a364:
```
__attribute__ ((constructor)) static void
sandbox_stat_init(void)
{
   12006a270:   67bdfee0        daddiu  sp,sp,-288
   12006a274:   67a50118        daddiu  a1,sp,280
   12006a278:   ebcb2803        csd     s8,a1,0(c11)
   12006a27c:   fa2be8e0        csc     c17,sp,224(c11)
   12006a280:   03a0f025        move    s8,sp
   12006a284:   49b96002        cgetoffset      t9,c12         <-- $t9 derived from (undefined) $c12???
   12006a288:   3c01000f        lui     at,0xf
   12006a28c:   0039082d        daddu   at,at,t9
   12006a290:   64215c50        daddiu  at,at,23632
   12006a294:   67d900c0        daddiu  t9,s8,192
   12006a298:   49ac5e41        csetoffset      c12,c11,t9      <-- $c12 offset set to ($c11 + $t9) -- stack-relative
   12006a29c:   64190020        daddiu  t9,zero,32
   12006a2a0:   482c6640        csetbounds      c12,c12,t9     <-- $c12 bounds set to 32 (capability size)
   12006a2a4:   67d900b8        daddiu  t9,s8,184
   12006a2a8:   49a15e41        csetoffset      c1,c11,t9
   12006a2ac:   64190008        daddiu  t9,zero,8
   12006a2b0:   48210e40        csetbounds      c1,c1,t9
   12006a2b4:   67c200b0        daddiu  v0,s8,176
   12006a2b8:   49a25881        csetoffset      c2,c11,v0
   12006a2bc:   48221640        csetbounds      c2,c2,t9
[... 46 lines cut ...]
         *
         * XXXRW: It would be nice if these pointer assignments had release
         * semantics.
         */
        ps_strings = (struct ps_strings *)ul_ps_strings;
   12006a338:   d82bf020        clc     c1,s8,32(c11)
   12006a33c:   c8210003        cld     at,zero,0(c1)
   12006a340:   64020000        daddiu  v0,zero,0
   12006a344:   48820007        cfromptr        c2,c0,zero
   12006a348:   49a21041        csetoffset      c2,c2,at
   12006a34c:   d86bf060        clc     c3,s8,96(c11)
   12006a350:   f8430000        csc     c2,zero,0(c3)
        ps_strings->ps_sbclasseslen = sizeof(sandbox_class_stats);
   12006a354:   d84bf060        clc     c2,s8,96(c11)        <-- $c2 loaded from ($c11 + 96) on the stack
   12006a358:   d8620000        clc     c3,zero,0(c2)         <-- $c3 loaded from ($c2 + 0)
   12006a35c:   640100a0        daddiu  at,zero,160
   12006a360:   64031800        daddiu  v1,zero,6144
   12006a364:   e8630803        csd     v1,at,0(c3)             <-- Boom due to untagged $c3
```

$c3 is untagged (NULL-derived):
```
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:7fffffeec0 t:0
```
The sequence is fairly odd: first, the compiler derives an offset from $c12 at the start of the function (which has undefined / uncertain value), but combines it with $c11 which is a valid capability, and later we end up with an untagged capability loaded from the (presumed) same spot. There's at least one compiler bug here (tagging @davidchisnall), which I've filed as https://github.com/CTSRD-CHERI/llvm/issues/146. There is something else going on here, however -- possibly also compiler linked.

rwatson, 2015-12-20:
This problem is resolved in 09eb7f0bf1f924ca18217088e38b047b42241386, which uses ```cheri_getdefault()``` and ```cheri_setoffset()``` to cast the ```u_long``` returned by the ```ps_strings``` ```sysctl()``` to a pointer. This is another instance of an integer-to-pointer cast leading to a NULL-derived dereference. This does leave me wondering if we have a larger-scale code-compatibility problem to resolve here -- e.g., whether we can in fact ditch $c0 as hoped.

rwatson, 2015-12-20:
I am now able to run ```cheriabitest``` and encounter:
```
> cheriabitest -a
cheriabitest: sigaltstack: Function not implemented
```

This is already covered by #89, so assigning to @brooksdavis until that is resolved.

rwatson, 2015-12-21:
Following 9f93f9c4fd5ef0e46629b8288c874695aae6d44f, I am now able to get further:
```
root@beri1:/tmp # cheriabitest test_initregs_default
Bus error
```

Console output:
```
cpuid = 0
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffd140 l:0000000000000020 o:0 t:0
$c02: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:48 t:0
$c03: v:1 s:0 p:7fff81fd b:0000007fffffd118 l:0000000000000008 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000007fffffd0c0 l:0000000000000020 o:0 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffd0e0 l:0000000000000020 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:120157738 t:0
$c07: v:1 s:0 p:7fff81ff b:0000000160176000 l:0000000000001000 o:20 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffcf70 l:0000000000000008 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffcf68 l:0000000000000008 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffcf40 l:0000000000000020 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f380 o:0 t:0
$c12: v:1 s:0 p:7fff81fd b:0000007fffffd140 l:0000000000000020 o:0 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffcf20 l:0000000000000020 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffccc0 l:0000000000000020 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffffcca0 l:0000000000000020 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffffcc80 l:0000000000000020 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12005ec28 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120070200 t:0
Apr 20 13:21:03 beri1 kernel: ALIGNMENT_FIX_ERR: pid 1964 tid 100066 (cheriabitest), uid 0: pc 0x120070200 got a write fault (type 0x5) at 0x120157738
Apr 20 13:21:03 beri1 kernel: Trapframe Register Dump:
Apr 20 13:21:03 beri1 kernel: zero: 0   at: 0   v0: 0   v1: 0
Apr 20 13:21:03 beri1 kernel: a0: 0x1   a1: 0x7df18     a2: 0x7de98     a3: 0
Apr 20 13:21:03 beri1 kernel: a4: 0x7cea8       a5: 0x7ced8     a6: 0xe58       a6: 0x1
Apr 20 13:21:03 beri1 kernel: t0: 0     t1: 0x16        t2: 0   t3: 0
Apr 20 13:21:03 beri1 kernel: t8: 0xf   t9: 0x20        s0: 0   s1: 0
Apr 20 13:21:03 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Apr 20 13:21:03 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Apr 20 13:21:03 beri1 kernel: gp: 0x7ec18       sp: 0x7dfa0     s8: 0x7dfa0     ra: 0
Apr 20 13:21:03 beri1 kernel: sr: 0x4000fcf3    mullo: 0x60     mulhi: 0        badvaddr: 0x120157738
Apr 20 13:21:03 beri1 kernel: cause: 0x114      pc: 0x120070200
Apr 20 13:21:03 beri1 kernel: Page table info for bad address 0x120157738: pde = 0x980000000409e000, pte = 0xa000000012bb5a
```

Disassembly:
```
000000012006fec0 <sandbox_set_required_method_variables>:

int
sandbox_set_required_method_variables(__capability void *datacap,
    struct sandbox_required_methods *required_methods)
{
   12006fec0:   67bdfe00        daddiu  sp,sp,-512
   12006fec4:   67a401f8        daddiu  a0,sp,504
   12006fec8:   ebcb2003        csd     s8,a0,0(c11)
   12006fecc:   fa2be9c0        csc     c17,sp,448(c11)
   12006fed0:   03a0f025        move    s8,sp
   12006fed4:   49b96002        cgetoffset      t9,c12
   12006fed8:   3c01000f        lui     at,0xf
   12006fedc:   0039082d        daddu   at,at,t9
   12006fee0:   64210000        daddiu  at,at,0
   12006fee4:   67d901a0        daddiu  t9,s8,416
   12006fee8:   49ac5e41        csetoffset      c12,c11,t9
[... 206 lines cut ...]
   1200701dc:   f8450000        csc     c2,zero,0(c5)
                    rmethods[i].srm_index_offset);
                *method_var_p = rmethods[i].srm_vtable_offset;
   1200701e0:   c8430003        cld     v0,zero,0(c3)
   1200701e4:   d8440000        clc     c2,zero,0(c4)
   1200701e8:   00021978        dsll    v1,v0,0x5
   1200701ec:   000211f8        dsll    v0,v0,0x7
   1200701f0:   0043102d        daddu   v0,v0,v1
   1200701f4:   d8421060        clc     c2,v0,96(c2)
   1200701f8:   d8abf080        clc     c5,s8,128(c11)
   1200701fc:   d8c50000        clc     c6,zero,0(c5)
   120070200:   f8460000        csc     c2,zero,0(c6)        <--- boom here
   120070204:   e82bf0c3        csd     at,s8,24(c11)
```

Interestingly, the faulting address is this in the ```objdump```:
```
0000000120157738  w    O .CHERI_CALLER  0000000000000008 __cheri_method.cherites
t.invoke_malloc
...
0000000120157738 <__cheri_method.cheritest.invoke_malloc>:
        ...
```

```__cheri_method.cheritest.invoke_malloc``` is not 32-byte aligned, hence the fault.

rwatson, 2015-12-21:
Looks like it's not just alignment, but also size that's a problem here -- it appears to be a symbol for an 8-byte object. @davidchisnall: should the storage size here be capability-sized (32), rather than double word-sized (8)for pure-capability code generation?

rwatson, 2015-12-21:
(or should ```libcheri``` be assuming they are words instead of pointers?)

davidchisnall, 2015-12-21:
The `__cheri_method.{class}.{method}` variable is legacy, and probably doesn't need to exist now that libcheri can fill in the correct field in the `sandbox_required_method` structure.  Regardless, the type of this variable should be a 64-bit integer - it's a vtable offset, not a pointer.

brooksdavis, 2015-12-21:
With aca36dc422a8515f40b5492a1ed4176f3efe6c62 we're now failing due to a pagefault in the sandbox and then again when freeing the vtable.

brooksdavis, 2015-12-22:
With ebaa2f5644158494b318c529c72f8a8a24db88a3 we can pass a simple test, but crash on a tag violation in a sandbox teardown:

```
./cheriabitest test_initregs_default
TEST: test_initregs_default: Test initial value of default capability
PASS: test_initregs_default
Signal 34
```

```
CHERI cause: ExcCode: 0x02 RegNum: $c03 (tag violation)
$c00: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c01: v:1 s:0 p:7fff81fd b:0000007fffffdac0 l:0000000000000020 o:0 t:0
$c02: v:1 s:0 p:7fff81fd b:0000007fffffd8dc l:0000000000000004 o:0 t:0
$c03: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c04: v:1 s:0 p:7fff81fd b:0000000120145a40 l:0000000000000900 o:300 t:0
$c05: v:1 s:0 p:7fff81fd b:0000007fffffd4f8 l:0000000000000004 o:0 t:0
$c06: v:1 s:0 p:7fff81fd b:0000007fffffd878 l:0000000000000004 o:0 t:0
$c07: v:1 s:0 p:7fff81fd b:0000007fffffc2b8 l:0000000000000008 o:0 t:0
$c08: v:1 s:0 p:7fff81fd b:0000007fffffc880 l:0000000000000020 o:0 t:0
$c09: v:1 s:0 p:7fff81fd b:0000007fffffc87c l:0000000000000004 o:0 t:0
$c10: v:1 s:0 p:7fff81fd b:0000007fffffc840 l:0000000000000020 o:0 t:0
$c11: v:1 s:0 p:7fff81fd b:0000007ffff7f000 l:000000000007f1e0 o:0 t:0
$c12: v:1 s:0 p:7fff81fd b:0000007fffffdae0 l:0000000000000020 o:0 t:0
$c13: v:1 s:0 p:7fff81fd b:0000007fffffc820 l:0000000000000020 o:0 t:0
$c14: v:1 s:0 p:7fff81fd b:0000007fffffc800 l:0000000000000020 o:0 t:0
$c15: v:1 s:0 p:7fff81fd b:0000007fffff6cb0 l:0000000000000008 o:0 t:0
$c16: v:1 s:0 p:7fff81fd b:0000007fffff92bc l:0000000000000004 o:0 t:0
$c17: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:120013534 t:0
$c18: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c19: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c20: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c21: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c22: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c23: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c24: v:0 s:0 p:00000000 b:0000000000000000 l:0000000000000000 o:0 t:0
$c26: v:1 s:0 p:7fff81fd b:0000000000000000 l:0000010000000000 o:0 t:0
$c31: v:1 s:0 p:7fff8197 b:0000000000000000 l:0000010000000000 o:12006259c t:0
Nov 29 18:36:46 beri1 kernel: USER_CHERI_EXCEPTION: pid 769 tid 100061 (cheriabitest), uid 0: CP2 fault (type 0x32)
Nov 29 18:36:46 beri1 kernel: Trapframe Register Dump:
Nov 29 18:36:46 beri1 kernel: zero: 0   at: 0x120160ec0 v0: 0   v1: 0x120160ec0
Nov 29 18:36:46 beri1 kernel: a0: 0x7eb30       a1: 0   a2: 0x7eea8     a3: 0
Nov 29 18:36:46 beri1 kernel: a4: 0     a5: 0xffffffffffffffff  a6: 0x7d930     a6: 0
Nov 29 18:36:46 beri1 kernel: t0: 0     t1: 0   t2: 0   t3: 0
Nov 29 18:36:46 beri1 kernel: t8: 0xa   t9: 0x20        s0: 0   s1: 0
Nov 29 18:36:46 beri1 kernel: s2: 0     s3: 0   s4: 0   s5: 0
Nov 29 18:36:46 beri1 kernel: s6: 0     s7: 0   k0: 0   k1: 0
Nov 29 18:36:46 beri1 kernel: gp: 0     sp: 0x7e9e0     s8: 0x7e9e0     ra: 0
Nov 29 18:36:46 beri1 kernel: sr: 0x4000fcf3    mullo: 0        mulhi: 0        badvaddr: 0x120062548
Nov 29 18:36:46 beri1 kernel: cause: 0x148      pc: 0x12006259c
```

```
void
sandbox_object_destroy(struct sandbox_object *sbop)
{
   120062548:   67bdfea0        daddiu  sp,sp,-352
   12006254c:   67a40158        daddiu  a0,sp,344
   120062550:   ebcb2003        csd     s8,a0,0(c11)
   120062554:   67a40150        daddiu  a0,sp,336
   120062558:   eb8b2003        csd     gp,a0,0(c11)
   12006255c:   fa2be920        csc     c17,sp,288(c11)
   120062560:   03a0f025        move    s8,sp
   120062564:   49b96002        cgetoffset      t9,c12
   120062568:   3c010010        lui     at,0x10
   12006256c:   0039082d        daddu   at,at,t9
   120062570:   6421e978        daddiu  at,at,-5768
   120062574:   67d90100        daddiu  t9,s8,256
   120062578:   49ac5e41        csetoffset      c12,c11,t9
   12006257c:   64190020        daddiu  t9,zero,32
   120062580:   482c6640        csetbounds      c12,c12,t9
   120062584:   67c200e0        daddiu  v0,s8,224
   120062588:   49a15881        csetoffset      c1,c11,v0
   12006258c:   48210e40        csetbounds      c1,c1,t9
   120062590:   64020000        daddiu  v0,zero,0
   120062594:   f86c0000        csc     c3,zero,0(c12)
        struct sandbox_class *sbcp;

        sbcp = sbop->sbo_sandbox_classp;
   120062598:   d86c0000        clc     c3,zero,0(c12)
   12006259c:   d8630040        clc     c3,zero,64(c3)
   1200625a0:   f8610000        csc     c3,zero,0(c1)
   1200625a4:   67dc00d8        daddiu  gp,s8,216
   1200625a8:   e82be003        csd     at,gp,0(c11)
   1200625ac:   67c100d0        daddiu  at,s8,208
   1200625b0:   e84b0803        csd     v0,at,0(c11)
   1200625b4:   f98bf0a0        csc     c12,s8,160(c11)
   1200625b8:   f82bf080        csc     c1,s8,128(c11)
...
```

rwatson, 2015-12-22:
Snag the issue back again now that ```sigaltstack()``` is believed resolved.

rwatson, 2015-12-22:
With my change 659b2ad2524d75c5a99567d27ab136bf1bda11aa, I no longer get a ```cheriabitest``` assertion failure during ```libcheri``` initialisation following my changes to divorce data and object-type capabilities (2d60da3e28a952c1040229744c4d30365c5ca550). However, I likewise get the crash that @brooksdavis is getting with tag violation after running a test.

rwatson, 2016-01-01:
Various improvements, including 8752587861c50778fe6937aeabd17244abd23fe5, now allow system-class invocation to work from sandboxes in CheriABI.

rwatson, 2016-01-01:
Further improvements to signal handling and ```cheritest``` itself now allow all signal-related test cases to pass in CheriABI: f9cdc807c729d84c6c69115d0b0fc1afa8bed243, 12693025baf4c9eef9f12df611070f28197be9f3, 8823287750cb93e7122e103ac380198fdbf2e022, 55fea6e3eb8f851819774b8824eb1d35c56069bb, b0a6d89096396281d7f27ff10b3412b0f448dd09, and 14e3ab80a0249a97b56b41ce7cb232b92973b10a leave all sandbox-related test cases passing in ```cheriabitest```.

rwatson, 2016-01-01:
Remaining (unexpected) failing test cases are:
```
FAIL: test_bounds_stack_static_524288: Test failed to set a success/failure status
FAIL: test_bounds_stack_static_1048576: Test failed to set a success/failure status
FAIL: test_bounds_stack_dynamic_524288: Test failed to set a success/failure status
FAIL: test_bounds_stack_dynamic_1048576: Test failed to set a success/failure status
FAIL: cheritest_vm_tag_shm_open_anon_shared: shm_open
FAIL: cheritest_vm_tag_shm_open_anon_private: shm_open
FAIL: cheritest_vm_cow_read: shm_open
FAIL: cheritest_vm_cow_write: shm_open
```

The former appear to relate to stack overflows, and the latter appear to relate to a problem with ```shm_open(2)```'s ```SHM_ANON``` argument.

rwatson, 2016-01-03:
Problems with shared-memory test cases have been resolved in #96, and related to an assembler bug worked around by f619a9056b38df4dcc6f88f0a05a1a117523096e.

rwatson, 2016-01-03:
With my most recent change (199c896), all expected success-case tests in ```cheriabitest``` now pass. There are some desired refinements to how tests are performed on CheriABI, but, in general, we now believe it to fully support the libcheri sandbox environment.

closed 2016-01-03 by rwatson