CTSRD-CHERI/cheribsd #93: CheriABI may not be running __attribute__ ((constructor)) functions on program startup

rwatson, 2015-12-21:
As of 2d60da3e28a952c1040229744c4d30365c5ca550, ```libcheri``` depends on ```cheri_type_init()``` running during process startup to request a type capability from the kernel. Unfortunately, this appears not to happen in ```cheriabitest```, but does work in ```cheritest```.

No diagnosis yet, but this leaves me wondering if ```__attribute__ ((constructor))``` functions may not be running. Alternatively, the constructor does run but ```sysarch()``` is not getting called properly. In ```cheritest```, the ```sysarch()``` call appears clearly in the trace, but is omitted from ```cheriabitest``` when it runs.

rwatson, 2015-12-21:
Tagging @davidchisnall and @brooksdavis since it's not clear whether this originates in a compiler or C-runtime problem.

davidchisnall, 2015-12-22:
I'm pretty sure `__attribute__((constructor))` functions are running (lots of stuff would break if they weren't), but it's easy to test by just having it set a global and then asserting that the global is set (make sure that it's not `static`, or the optimisers might be clever enough to know that it can only be set from there).  With regard to the `sysarch`, I think that this may be the first time the sysarch is used in CheriABI, so it might not yet be implemented.

rwatson, 2015-12-22:
Although there is no explicit ```sysarch()``` wrapper currently, the native FreeBSD ```sysarch()``` is hooked up and should in fact work for some of the simple cases (e.g., getting the timestamp counter) and even for some more complex ones (e.g., I think that the trusted-stack-related calls, as well as the one to query the root type capability) might "just work". Even if they don't work, it should appear in ```ktrace``` followed by a failure. Which suggests to me that even if the constructor is running, the ```sysarch()``` call is not taking place. Another possibility is that we have undocumented dependencies between constructors, and they are running in a different order, so a call dependent on the root type is being made before the root-type constructor runs. I'll investigate this possibility.

rwatson, 2015-12-22:
Resolved in ```libcheri``` with 659b2ad2524d75c5a99567d27ab136bf1bda11aa, which avoids using a constructor here. This was probably an (non-)ordering issue with the constructors rather than a problem using constructors at all, so close -- but more investigation needed to confirm this hypothesis.

closed 2015-12-22 by rwatson