CTSRD-CHERI/cheribsd #97: Merge packaged base branch?

davidchisnall, 2016-01-02:
It would be nice if we could build a package set from base so that people doing software work can do binary upgrades.

brooksdavis, 2016-01-07:
Is the base packaging work going on in a separate branch?  I'd presumably need to get us merged up to HEAD at a minimum even if it's in another branch. I think we may be in a good place to do that once I'm back next week.  (I'll need to hold a lock on the official head for a few days.)

davidchisnall, 2016-01-07:
Yes, in the release-pkg branch.  I'm not sure how easy it is to cherry-pick the relevant commits.  We could possibly encourage the people working on it to push it to head in an incomplete state...

brooksdavis, 2017-12-12:
The basics of this are in via head.  Actually making packaging work for cheribsd would be an additional bit of work since we'd have to add appropriate annotations for package metadata.  We should look into this, but I'm closing this issue for now.

closed 2017-12-12 by brooksdavis