CTSRD-CHERI/cheribsd #101: const arrays of pointers are ending up in read-only pages

brooksdavis, 2016-01-19:
With changes to the local code we're not ending up with const arrays in pages that are being mapped read-only.  This is breaking constructors.  I've not yet determined if this is a bug in the cheriabi mapping code, a compiler bug, or something else new.

davidchisnall, 2016-01-19:
It's possible that this is an effect of `-cheri-linker`.  We're now able to mark constant data as constant in the IR.  Longer term, this is the correct thing to do: data that contains pointers should be mapped read-write, the relocations applied, and then lose its write permission.

FreeBSD was mapping read-only data as read-write for a long time.  It may be that this was fixed in a recent merge?

brooksdavis, 2016-01-19:
I it looks like I accidentally broke branding of CheriABI programs in the merge which means they aren't being loaded as CheriABI binaries.  Thus the r/o sections are being mapped r/o.

brooksdavis, 2016-01-20:
The problem is due to GNU objcopy losing the actual ELF machine type when working on binaries with one of the alternate types (which I'm using for CheriABI).  Worked around in c30995a5c1f02504683d1e52cacf65ec9c9d1888.

closed 2016-01-20 by brooksdavis