CTSRD-CHERI/cheribsd #102: passing CPUTYPE=mips breaks the build

a-richardson, 2016-01-19:
The build_sdk.sh script does this so after the merge this no longer works.
The problem seems to be `${CPUTYPE:S/^mips//}` inside bsd.cpu.mk introduced by
```
commit 952142b8429e4692b0fa3c8ce70a9fd5e27809b7
Author: imp <imp@FreeBSD.org>
Date:   Thu Nov 19 03:11:20 2015 +0000

    Fix mips CPUTYPE so that we can pass it through to gcc.
    Keep old CPUTYPEs around for compatibility. Also include
    a list of typical values for FreeBSD.
    
    # Split out from other changes in D4155
    
    Differential Revision: https://reviews.freebsd.org/D4155
```
This results in an empty `-march=` being passed to the compiler.

For now I worked around this by setting CPUTYPE to mips64 which works but I am not sure whether this is correct.

brooksdavis, 2016-01-19:
I don't know why @davidchisnall is passing a CPUTYPE value in the sdk.  I've never used one by hand and don't know what it's expected to do.

brooksdavis, 2016-01-19:
Ah, it's supposed to select between fpu and non-fpu cases.  That is very much not supported in FreeBSD proper so it breaking in unsurprising.

davidchisnall, 2016-01-20:
The `build_sdk.sh` script just copied the buildworld instructions from the CHERIBSD documentation.  I'm not sure if it still says to set `CPUTYPE` in the software reference, but if it does then it probably needs fixing there too.

brooksdavis, 2016-01-20:
Worked around by removing CPUTYPE use from `build_sdk.sh`.  Fixed an unrelated issue in the sdk build in 579efe6026a50157595871a594fa6b275fd25fca.

closed 2016-01-20 by brooksdavis