CTSRD-CHERI/cheribsd #103: make installworld fails at cheritest-helper

a-richardson, 2016-01-19:
when I run `MAKEOBJDIRPREFIX=/home/alr48/cheri/qemu/obj/ make CHERI=256 CHERI_CC=/home/alr48/cheri/qemu/llvm-build/bin/clang CPUTYPE=mips64 -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR installworld DESTDIR=/home/alr48/cheri/qemu/rootfs` 

```
===> libexec/cheritest-helper (install)
install -N /home/alr48/cheri/cheribsd/etc -U -M /home/alr48/cheri/qemu/rootfs//METALOG -D /home/alr48/cheri/qemu/rootfs -o root -g wheel -m 555   cheritest-helper.stripped /home/alr48/cheri/qemu/rootfs/usr/libexec/cheritest-helper
install -N /home/alr48/cheri/cheribsd/etc -U -M /home/alr48/cheri/qemu/rootfs//METALOG -D /home/alr48/cheri/qemu/rootfs -T debug -o root -g wheel -m 444  cheritest-helper.debug /home/alr48/cheri/qemu/rootfs/usr/lib/debug/usr/libexec/cheritest-helper.debug
make[5]: /home/alr48/cheri/qemu/obj//mips.mips64/home/alr48/cheri/cheribsd/libexec/cheritest-helper/.depend, 75: ignoring stale .depend for /home/alr48/cheri/qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/libcheri/libMD.a
make[5]: /home/alr48/cheri/qemu/obj//mips.mips64/home/alr48/cheri/cheribsd/libexec/cheritest-helper/.depend, 75: ignoring stale .depend for /home/alr48/cheri/qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/libcheri/libZ.a
make[5]: /home/alr48/cheri/qemu/obj//mips.mips64/home/alr48/cheri/cheribsd/libexec/cheritest-helper/.depend, 75: ignoring stale .depend for /home/alr48/cheri/qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/libcheri/libC_CHERI.a
/home/alr48/cheri/qemu/llvm-build/bin/clang -g -integrated-as --target=cheri-unknown-freebsd -msoft-float -mabi=sandbox -mxgot -cheri-linker -O -pipe -G0 -march=mips64 -O0 -Wno-error -g -std=gnu99 -Wsystem-headers -Wall -Wno-format-y2k -W -Wno-unused-parameter -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch -Wshadow -Wunused-parameter -Wcast-align -Wchar-subscripts -Winline -Wnested-externs -Wredundant-decls -Wold-style-definition -Wno-pointer-sign -Wformat=2 -Wno-format-extra-args -Wmissing-variable-declarations -Wthread-safety -Wno-empty-body -Wno-string-plus-int -Wno-unused-const-variable -Qunused-arguments  -Wl,--no-warn-mismatch -Wl,--script=/home/alr48/cheri/cheribsd/libexec/cheritest-helper/../../lib/libc_cheri/sandbox.ld  -nostdlib -static -o cheritest-helper.full cheritest-helper.o stub.o   -lmd  -lz  -lc_cheri
/home/alr48/cheri/qemu/obj//mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/bin/ld: cannot find -lmd
clang-3.8: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1
```

If I search for libmd.a I get the following:

```
./qemu/rootfs/usr/lib/libmd.a
./qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/lib/libmd/libmd.a
./qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/worldcheri/home/alr48/cheri/cheribsd/lib/libmd/libmd.a
./qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/lib/libmd.a
./qemu/obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/libcheri/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/lib/libmd/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/libcheri/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/usr/lib/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/legacy/usr/lib/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmp/home/alr48/cheri/cheribsd/lib/libmd/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/worldcheri/home/alr48/cheri/cheribsd/lib/libmd/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmproot/usr/lib/libmd.a
./obj/mips.mips64/home/alr48/cheri/cheribsd/tmproot/usr/libcheri/libmd.a
```

brooksdavis, 2016-01-19:
Unless there's a bug in the build (entirely possible), it appears you have done an `installworld` without a `buildworld` as the `installworld` target should not be compiling things.  I'll give installworld a spin here.

a-richardson, 2016-01-19:
I did do a buildworld first. The other binaries in this directory already exist from buildworld. Seems like compiling `cheritest-helper.full` is not part of buildworld?

brooksdavis, 2016-01-19:
Just hit it myself.  Definitely a build system bug.  It looks like I need to disable @emaste's debug file stuff for helpers.

commit a93240ba (2016-01-20): Don't add bogus libUPPERCASE.a depends.

These depends were forcing rebuilds during installworld which failed.

Fixes #103.
closed 2016-01-20 by brooksdavis