CTSRD-CHERI/cheribsd #105: Build failure from linker removing needed sections

davidchisnall, 2016-01-22:
```
===> lib/libhelloworld/compartment (all)
/home/dc552/CHERISDK/sdk//bin/clang -g -integrated-as --target=cheri-unknown-freebsd -msoft-float --sysroot=/home/dc552/obj/mips.mips64/home/dc552/CHERISDK/cheribsd/tmp -mabi=sandbox -mxgot -cheri-linker -O -pipe -G0 -O0 -Wno-error -g -std=gnu99 -Qunused-arguments  -Wl,--no-warn-mismatch -Wl,--script=/home/dc552/CHERISDK/cheribsd/lib/libhelloworld/compartment/../../libc_cheri/sandbox.ld  -nostdlib -static -o helloworld.co.0.full helloworld.o   -lc_cheri
`__cheri_sandbox_required_methods' referenced in section `__cap_relocs' of /home/dc552/obj/mips.mips64/home/dc552/CHERISDK/cheribsd/tmp/usr/libcheri/libc_cheri.a(cheri_stdio.o): defined in discarded section `__cheri_sandbox_required_methods[.sandbox_required_method._cheri_system_object.cheri_system_puts]' of /home/dc552/obj/mips.mips64/home/dc552/CHERISDK/cheribsd/tmp/usr/libcheri/libc_cheri.a(cheri_stdio.o)
`__cheri_sandbox_required_methods' referenced in section `__cap_relocs' of /home/dc552/obj/mips.mips64/home/dc552/CHERISDK/cheribsd/tmp/usr/libcheri/libc_cheri.a(cheri_printf.o): defined in discarded section `__cheri_sandbox_required_methods[.sandbox_required_method._cheri_system_object.cheri_system_putchar]' of /home/dc552/obj/mips.mips64/home/dc552/CHERISDK/cheribsd/tmp/usr/libcheri/libc_cheri.a(cheri_printf.o)
```

It looks as if the linker script `cheribsd/lib/libc_cheri/sandbox.ld` does not contain instructions for the `__cheri_sandbox_required_methods`, `.CHERI_CALLEE`, `__cheri_sandbox_provided_methods`, and `.CHERI_CALLER` sections, so they are dropped.

brooksdavis, 2016-01-22:
TL;DR: there something special in your SDK.

I'm not seeing this with my hand built clang, my SDK just updated fine, and I can use the SDK clang to build world.  When I try to build world with your SDK's clang I do see this breakage.  I don't see if it if I run your clang manually against `helloworld.o`.

brooksdavis, 2016-01-28:
Is this still happening?

brooksdavis, 2016-02-16:
I can't reproduce this and I know @davidchisnall has built world so closing for now

closed 2016-02-16 by brooksdavis