CTSRD-CHERI/cheribsd #107: METALOG is not a valid mtree file for makefs (error at ./usr/libcheri/.debug)

RichardsonAlex, 2016-01-24:
When I build a disk image for QEMU (`cd /home/alr48 && makefs -M 1077936128 -B be -F /home/alr48/cheri/output/rootfs/METALOG -N /home/alr48/cheri/cheribsd/etc /home/alr48/cheri/output/disk.image /home/alr48/cheri/output/rootfs`) I get the following error: 
```
makefs: ./usr/libcheri/.debug: missing directory in specification
makefs: failed at line 4306 of the specification
```
In order to successfully build the disk image I need to move the `./usr/libcheri/.debug` line to the top of the file. Not sure why this is necessary but it fixes the build for me:
```
@@ -121,6 +121,7 @@
 ./usr/lib/libxo type=dir uname=root gname=wheel mode=0755
 ./usr/lib/libxo/encoder type=dir uname=root gname=wheel mode=0755
 ./usr/libcheri type=dir uname=root gname=wheel mode=0755
+./usr/libcheri/.debug type=dir mode=0755 tags=debug
 ./usr/libdata type=dir uname=root gname=wheel mode=0755
 ./usr/libdata/gcc type=dir uname=root gname=wheel mode=0755
 ./usr/libdata/ldscripts type=dir uname=root gname=wheel mode=0755
@@ -4434,7 +4435,6 @@
 ./usr/lib//libhelloworld_p.a type=file uname=root gname=wheel mode=0444 size=6614
 ./usr/include/cheri//helloworld.h type=file uname=root gname=wheel mode=0444 size=2370
 ./usr/libcheri/helloworld.co.0 type=file uname=root gname=wheel mode=0555 size=77984
-./usr/libcheri/.debug/ type=dir mode=0755 tags=debug
 ./usr/libcheri/.debug/helloworld.co.0.debug type=file uname=root gname=wheel mode=0444 size=67656 tags=debug
 ./usr/libcheri//helloworld.co.0.dump type=file uname=root gname=wheel mode=0444 size=1049489
 ./usr/lib//librpcsec_gss.a type=file uname=root gname=wheel mode=0444 size=53040
```

These are the commands executed by my build script (https://github.com/RichardsonAlex/cheri-scripts)
```bash
cd /home/alr48/cheri/qemu && git checkout HEAD po/ 
cd /home/alr48/cheri/qemu && git pull --rebase 
cd /home/alr48/cheri/output/qemu-build && /home/alr48/cheri/qemu/configure --target-list=cheri-softmmu --disable-linux-user --disable-linux-aio --disable-kvm --disable-xen --extra-cflags=-g --prefix=/home/alr48/cheri/output/host-tools 
cd /home/alr48/cheri/output/qemu-build && gmake -j16 
cd /home/alr48/cheri/output/qemu-build && gmake install 
cd /home/alr48/cheri/binutils && git pull --rebase 
cd /home/alr48/cheri/output/binutils-build && /home/alr48/cheri/binutils/configure --target=mips64 --disable-werror --prefix=/home/alr48/cheri/output/host-tools 
cd /home/alr48/cheri/output/binutils-build && make -j16 
cd /home/alr48/cheri/output/binutils-build && make install 
cd /home/alr48/cheri/llvm && git pull --rebase 
cd /home/alr48/cheri/llvm/tools/clang && git pull --rebase 
cd /home/alr48/cheri/output/llvm-build && cmake /home/alr48/cheri/llvm -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_COMPILER=clang++37 -DCMAKE_C_COMPILER=clang37 -DLLVM_DEFAULT_TARGET_TRIPLE=cheri-unknown-freebsd -DCMAKE_INSTALL_PREFIX=/home/alr48/cheri/output/host-tools -DDEFAULT_SYSROOT=/home/alr48/cheri/output/cheribsd-obj/mips.mips64/home/alr48/cheri/cheribsd/tmp 
cd /home/alr48/cheri/output/llvm-build && ninja -j16 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdint.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdalign.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdarg.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdatomic.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stddef.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdbool.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/stdnoreturn.h 
rm /home/alr48/cheri/output/llvm-build/lib/clang/3.8.0/include/limits.h 
cd /home/alr48/cheri/cheribsd && git pull --rebase 
# Set PATH to /home/alr48/cheri/output/host-tools/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/games:/usr/local/sbin:/usr/local/bin:/home/alr48/bin
rm -rf /home/alr48/cheri/output/rootfs 
cd /home/alr48/cheri/cheribsd && make CHERI=256 CHERI_CC=/home/alr48/cheri/output/llvm-build/bin/clang -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR DESTDIR=/home/alr48/cheri/output/rootfs KERNCONF=CHERI_MALTA64 buildworld -j16 
cd /home/alr48/cheri/cheribsd && make CHERI=256 CHERI_CC=/home/alr48/cheri/output/llvm-build/bin/clang -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR DESTDIR=/home/alr48/cheri/output/rootfs KERNCONF=CHERI_MALTA64 buildkernel -j16 
cd /home/alr48/cheri/cheribsd && make CHERI=256 CHERI_CC=/home/alr48/cheri/output/llvm-build/bin/clang -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR DESTDIR=/home/alr48/cheri/output/rootfs KERNCONF=CHERI_MALTA64 installworld 
cd /home/alr48/cheri/cheribsd && make CHERI=256 CHERI_CC=/home/alr48/cheri/output/llvm-build/bin/clang -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR DESTDIR=/home/alr48/cheri/output/rootfs KERNCONF=CHERI_MALTA64 installkernel 
cd /home/alr48/cheri/cheribsd && make CHERI=256 CHERI_CC=/home/alr48/cheri/output/llvm-build/bin/clang -DDB_FROM_SRC -DNO_ROOT -DNO_WERROR DESTDIR=/home/alr48/cheri/output/rootfs KERNCONF=CHERI_MALTA64 distribution 
makefs -M 1077936128 -B be -F /home/alr48/cheri/output/rootfs/METALOG -N /home/alr48/cheri/cheribsd/etc /home/alr48/cheri/output/disk.image /home/alr48/cheri/output/rootfs 
```

stderr is empty except for a few messages like this: `[1]: "/home/alr48/cheri/cheribsd/Makefile.inc1" line 122: warning: ctsrd/lib not added to SUBDIR list.  See UPDATING 20141121.` as well as the following messages during installworld:
```
chflags: /home/alr48/cheri/output/rootfs/usr/bin/chpass: Operation not permitted
chflags: /home/alr48/cheri/output/rootfs/usr/bin/passwd: Operation not permitted
```

staceyson, 2016-01-24:
I am guessing that you need to clear the system immutable flag on chpass, passwd, etc.   Try:
```
# cd /home/alr48/cheri/output 
# chflags -Rf noschg rootfs
# rm -Rf rootfs
```
before doing the installworld to make sure you are starting with a clean directory.

RichardsonAlex, 2016-01-24:
I am compiling as a normal user so the error makes sense. Not sure if it has any effect on the created disk.img.
Launching the image in QEMU afterwards seems to work fine, but I haven't tried using chpass or passwd yet

commit 3c4b2bcc (2016-01-26): Put libcheri debug bits in /usr/lib/debug.

Should fix #107.
commit cd167196 (2016-11-22): MFV r307314:
6988 spa_sync() spends half its time in dmu_objset_do_userquota_updates

Using a benchmark which creates 2 million files in one TXG, I observe
that the thread running spa_sync() is on CPU almost the entire time we
are syncing, and therefore can be a performance bottleneck. About 50% of
the time in spa_sync() is in dmu_objset_do_userquota_updates().

The problem is that dmu_objset_do_userquota_updates() calls
zap_increment_int(DMU_USERUSED_OBJECT) once for every file that was
modified (or created). In this benchmark, all the files are owned by the
same user/group, so all 2 million  [... cut]
closed 2016-01-26 by brooksdavis