CTSRD-CHERI/cheribsd #108: MDROOT fails to load root after merge

davidchisnall, 2016-01-25:
I am no longer able to boot an MDROOT kernel after the merge.  Attempting to mount root fails with error 22 (`EINVAL`).  The `mountroot>` prompt shows `md0` exists, but it appears not to be mountable.  The root FS is around 18MB, so well under the size permitted (though the kernel now seems to be about 50MB, when the ones from before the merge were around 34MB).  The root fs image is built with the `makeroot` script:

```
$ file cheriroot.img
cheriroot.img: Unix Fast File system [v1] (big-endian), last mounted on , last written at Mon Jan 25 11:06:02 2016, clean flag 1, number of blocks 18032, number of data blocks 17935, number of cylinder groups 1, block size 8192, fragment size 1024, minimum percentage of free blocks 8, rotational delay 0ms, disk rotational speed 60rps, TIME optimization
```

brooksdavis, 2016-01-26:
I'm not seeing such large sizes.  There was a major change in the way images are embedded via the kernel config.  Which method are you using to specify which image to use?  If you're using `makeoptions    MFS_IMAGE=/foo/baz/baz/mdroot.img` it might be worth trying removing `MD_ROOT_SIZE` since that is only needed if you want to pre-reserve space.

davidchisnall, 2016-01-26:
Removing that line fixes it.

closed 2016-01-26 by davidchisnall