CTSRD-CHERI/cheribsd #110: New cheritest/cheriabitest failures: test_sandbox_divzero_*

staceyson, 2016-01-26:
With e92a423520a1fd661f4a9dff76e7f7635f0c66d6 using Qemu-CHERI test_sandbox_divzero_* now fails for both cheritest and cheriabitest:

```
root@:~ # cheritest test_sandbox_divzero_catch
TEST: test_sandbox_divzero_catch: Exercise sandboxed divide-by-zero exception; caught
FAIL: test_sandbox_divzero_catch: Expected signal 7, got 0
root@:~ # cheritest test_sandbox_divzero_nocatch
TEST: test_sandbox_divzero_nocatch: Exercise sandboxed divide-by-zero exception; uncaught
FAIL: test_sandbox_divzero_nocatch: invoke returned 0 (expected -1)
root@:~ # cheriabitest test_sandbox_divzero_catch
TEST: test_sandbox_divzero_catch: Exercise sandboxed divide-by-zero exception; caught
FAIL: test_sandbox_divzero_catch: Expected signal 7, got 0
root@:~ # cheriabitest test_sandbox_divzero_nocatch
TEST: test_sandbox_divzero_nocatch: Exercise sandboxed divide-by-zero exception; uncaught
FAIL: test_sandbox_divzero_nocatch: invoke returned 0 (expected -1)
```

brooksdavis, 2016-01-28:
428496cebe38ddfdefe7a057a6f1a06f7ab84bc1 includes a switch to inline assembly, but fails to correct the problem as the wrong signal is delivered.

brooksdavis, 2016-01-28:
@rwatson any opinion on the correctness of the current?

rwatson, 2016-02-19:
These errors continue to evolve with improvement to the compiler and ```cheritest```/```cheriabitest```. Currently, they are failing on CHERI in FPGA due to getting ```BREAK``` instead of ```TRAP``` exception, which is likely a hardware bug. It would be interesting to know if things go better on Qemu with recent changes (i.e., as of 3922ed275e82ab7cb21aa2292ea831f1ea34f880).

rwatson, 2016-05-27:
Changed to ```XFAIL``` in 02b6d25067a1ed2e17aa4c04f7c11963b7f92043: this appears to be due to a bug in the LLVM assembler, which is generating ```BREAK``` instructions, rather than conditional trap instructions, when the ```div``` pseudo-instruction is used.

jrtc27, 2022-01-21:
libcheri was removed in #1056

closed 2022-01-21 by jrtc27