CTSRD-CHERI/cheribsd #111: cap length violation in memcpy_c() running tar/pkg

staceyson, 2016-01-27:
Attempting to install some packages gives a cap length violation.  Looking at an instruction trace generated by qemu-cheri it happens in the `copy_caps` loop in `memcpy_c()`.  Part of the instruction trace is given below (`/rescue/tar tvzf ....`):

```
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Cap Memory Read [000000016140ff60] = v:0 tps:0200202502401025
    c:03c0e825dfb00010 b:dfb10018dfb20020 l:dfbc0028dfbe0030
    Write C05|v:0 s:1 p:01200812 b:dfb10018dfb20020 l:dfbc0028dfbe0030
             |o:240fe80cfffdfff0 t:2002025
0x00000001206f4c60:  bnvc       a1,a1,0x1206f4ce4
    Write a1 = 000000000000ff80
0x00000001206f4c64:  bne        a1,a2,0x1206f4c5c
0x00000001206f4c68:  csc        c5,a1,95(c3)
    Cap Memory Write [00000001610b9f60] = v:0 tps:0200202502401025
    c:03c0e825dfb00010 b:dfb10018dfb20020 l:dfbc0028dfbe0030
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Cap Memory Read [000000016140ff80] = v:0 tps:dfbf003803e00008
    c:67bd004000000000 b:67bdffa0ffbf0058 l:ffbe0050ffbc0048
    Write C05|v:0 s:0 p:01f00004 b:67bdffa0ffbf0058 l:ffbe0050ffbc0048
             |o:ffff009f0040ffa8 t:dfbf0038
0x00000001206f4c60:  bnvc       a1,a1,0x1206f4ce4
    Write a1 = 000000000000ffa0
0x00000001206f4c64:  bne        a1,a2,0x1206f4c5c
0x00000001206f4c68:  csc        c5,a1,95(c3)
    Cap Memory Write [00000001610b9f80] = v:0 tps:dfbf003803e00008    c:67bd004000000000 b:67bdffa0ffbf0058 l:ffbe0050ffbc0048
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Cap Memory Read [000000016140ffa0] = v:0 tps:ffb10040ffb00038
    c:03a0f02590ac0000 b:3c0100050039102d l:1180005f64100000
    Write C05|v:0 s:0 p:7fd8001c b:3c0100050039102d l:1180005f64100000
             |o:c79ff0209072efd3 t:ffb10040
0x00000001206f4c60:  bnvc       a1,a1,0x1206f4ce4
    Write a1 = 000000000000ffc0
0x00000001206f4c64:  bne        a1,a2,0x1206f4c5c
0x00000001206f4c68:  csc        c5,a1,95(c3)
    Cap Memory Write [00000001610b9fa0] = v:0 tps:ffb10040ffb00038
    c:03a0f02590ac0000 b:3c0100050039102d l:1180005f64100000
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Cap Memory Read [000000016140ffc0] = v:0 tps:645c0350240b0000
    c:2402005b2403002a b:2407003f2408005d l:2409005c00a05025
    Write C05|v:0 s:0 p:12058000 b:2407003f2408005d l:2409005c00a05025
             |o:fffb001bfffaffcd t:645c0350
0x00000001206f4c60:  bnvc       a1,a1,0x1206f4ce4
    Write a1 = 000000000000ffe0
0x00000001206f4c64:  bne        a1,a2,0x1206f4c5c
0x00000001206f4c68:  csc        c5,a1,95(c3)
    Cap Memory Write [00000001610b9fc0] = v:0 tps:645c0350240b0000
    c:2402005b2403002a b:2407003f2408005d l:2409005c00a05025
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Cap Memory Read [000000016140ffe0] = v:0 tps:000c0e0000016603
    c:2981005c14200008 b:000000001188000a l:0000000015890012
    Write C05|v:0 s:1 p:0000b301 b:000000001188000a l:0000000015890012
             |o:2981005c0297fffe t:c0e00
0x00000001206f4c60:  bnvc       a1,a1,0x1206f4ce4
    Write a1 = 0000000000010000
0x00000001206f4c64:  bne        a1,a2,0x1206f4c5c
0x00000001206f4c68:  csc        c5,a1,95(c3)
    Cap Memory Write [00000001610b9fe0] = v:0 tps:000c0e0000016603
    c:2981005c14200008 b:000000001188000a l:0000000015890012
0x00000001206f4c5c:  clc        c5,a1,80(c4)
    Write BadVAddr = 00000001206f4c5c
--- Exception #18: vector ffffffff80000180
    Write C31|v:1 s:0 p:7fff8017 b:0000000000000000 l:0000010000000000
             |o:00000001206f4c5c t:0
    Write Status = 00000000408084b3
    Write Cause = 0000000000000048
    Write EPC = 00000001206f4c5c
--- Kernel mode (ERL=0, KX=1)
```

`copy_caps` loop:
```
copy_caps:
        clc     $c5, $a1, 0($c4)
        daddi   $a1, $a1, CAP_SIZE
        bne     $a1, $a2, copy_caps
        csc     $c5, $a1, -CAP_SIZE($c3)
```

brooksdavis, 2016-01-27:
In the register dump I have it's obvious that something has gone wrong before entering the `copy_caps` loop.  The salient bits are:

```
$c03: v:1 s:0 p:7fff807d b:0000000160680f40 l:0000000000010000 o:0 t:0
$c04: v:1 s:0 p:7fff807d b:0000000160800000 l:0000000000010000 o:0 t:0
Dec 22 11:36:29 beri1 kernel: a0: 0x10000       a1: 0x10000     a2: 0   a3: 0x7fffffe510
```

Some how the loop was entered in a completely nonsensical state where even without the length violation the loop would never end as `$a1` would never reach `0`.

brooksdavis, 2016-01-27:
So the code `copy_caps` is:

```
 aligned_copy:
       addi    $at, $zero, (0xFFFF-CAP_ALIGN_MASK)
        and     $a2, $a2, $at        # a2 = number of capability-aligned bytes to copy
        dadd    $a2, $a2, $a1        # ...plus the number already copied.
```
I think this is an attempt to mask off the "remaining" bits at the end, but the practical effect for an '0x10000' byte, fully aligned copy is to mask off all the bits and run off the end.

closed 2016-01-27 by brooksdavis

brooksdavis, 2016-01-27:
After further thought, I think the code often copied all the bits for non-capability programs.  As long as  `$a2 & 0xffe0` wasn't zero the capability at a time copy would terminate early and copy the rest a byte at a time.

commit 8d7a5c91 (2016-01-27): When computing the number of capability aligned bytes to copy, only mask
off the bottom (misaligned) bits.

Fixes #111