CTSRD-CHERI/cheribsd #112: SD card image too large

tmarkettos, 2016-01-28:
freebsd-sdcard.img.xz is growing larger:
Jenkins #251 is 2306867200 bytes
Jenkins #264 is 2516582400 bytes

The Terasic 2GB SD card is 2021654528 bytes and the SD controller currently limits us to 2GiB (2147483648 bytes), which means it doesn't fit.

brooksdavis, 2016-01-28:
Will you be using the Altera controller on the DE5?  If so, I'll split the image into a full featured on and one that fits.  If not, I might just rename it and drop SDCard support from the documentation now that we've got USB on the DE4.

tmarkettos, 2016-01-28:
Yes, that's the plan.  I have the SD card working on the DE5 (I think) via my frankenadaptor but I don't have a suitable card image to write to it.  With an existing image it starts booting but falls over probably due to blocks off the end of the card.

brooksdavis, 2016-01-28:
I've committed a fix to the build system to get the sdcard image back under 2GB by removing debug symbols.

closed 2016-01-28 by brooksdavis